morrownr/8814au #4: Kernel 5.10.7 with TP-Link T9UH

jhu54, 2021-01-16:
Hi,

Just curious if anyone has tried this driver with the 5.10.7 kernel release? Am successfully using it with 5.10.6 but if I switch to the 10.7 kernel it no longer shows the device is present even though a USB query shows the device is present...

morrownr, 2021-01-16:
I'll take a look when able.

jhu54, 2021-01-16:
My testing:

ran the remove script on 10.6
booted into 10.7
ran the install script
this is the make.log from 10.7
[make.log](https://github.com/morrownr/8814au/files/5824320/make.log)

ran the remove script in 10.7
booted into 10.6
ran the install script
No issues.

From the log file, this was the error noted:
  MODPOST /var/lib/dkms/rtl8814au/5.8.5.1/build/Module.symvers
make[3]: *** No rule to make target 'scripts/module.lds', needed by '/var/lib/dkms/rtl8814au/5.8.5.1/build/8814au.ko'.  Stop.

Hope this helps

morrownr, 2021-01-17:
It helped greatly. I currently don't have 5.10 loaded on any of the systems I use for testing so this saves me some work. For what it is worth, I have a few systems available for testing but not enough to keep everything available all of the time.

Back to the issue: The problem is in here: linux-headers-5.10.7-051007-generic

A very very similar problem was reported on another driver here a couple of months ago and the problem was tracked down to a change in the header files. My recommendation is that you make a report to the kernel maintainer similar to the one above and include the make.log.

Where is the 5.10.7 kernel from? Ubuntu mainline?

jhu54, 2021-01-17:
Yes - ubuntu mainline.

I will report it to them. In the meantime, I am finding 5.10.6 or 5.9.16 work better then the default Ubuntu 5.4.x kernel on my z490 MB so it is no major issue. Found in particular the wifi performance on the newer kernel with your driver was better, I am quite a distance from the AC router and would fall back to 867mb/s all the time on the older kernel. Just moving to the 5.9.16 kernel (no other location or environment changes), I now get a solid 1300mb/s.

morrownr, 2021-01-17:
Thanks for making the report. I subscribe to the linux-wireless list so as to keep up with happenings in the kernel regarding wireless and it doesn't surprise me to hear you say things are getting better. Within the last year, I have seen many things start working and working well. A year ago, getting any of the Realtek USB WiFi adapter working with hostapd in AP mode in a full featured way was a pip dream. Now the driver I have up for the rtl8812au works so well with hostapd that it is a dream... 867Mb/s, 11AC, 80 channel width, no drops or errors. Totally solid. 

Something else that is not apparent to most folks that stop by here is the number of folks that use this site ( github.com/morrownr ). I just added the numbers and for the last 2 weeks this site has had 4,778 Views and 1,523 Clones. There are a lot of people using these drivers and the information but there are very few bugs reported. I try to keep the very latest code from Realtek online and available and test the additions and changes that I and others make. There is one driver that is bothering me and I wish I could get an update from Realtek. It is the driver you are using for the rtl8814au. I've done everything I know to do to help it at this point and it is a good driver for client mode but it is a really bad driver for AP mode. I need you and everyone that can help to contact the seller of your adapter or any and all sellers of adapters based on the rtl8814au and request that they contact Realtek to give them a modern version of the Linux source code. We need updated source code for the 8814au.

Regards,

Nick

jhu54, 2021-01-17:
Sure - would have been easy in a past life - worked in RFID and had the wireless engineers from Motorola, Qualcomm and Cisco on speed dial, but that was 15 years ago.

What I can see is the file /var/lib/dkms/rtl8814au/5.8.5.1/build/Module.symvers is empty when the make fails. Do you have a debug option in the install script I can enable? I can play around with it and see if anything else pops up...

Tested 5.10.8 and has the same issue. A lot of changes in the Module.symvers between the 5.10.6 and .8 - if you can attach a  /var/lib/dkms/rtl8814au/5.8.5.1/build/Module.symvers from a successful compile (gets deleted on success on my machine) I can try and trace this a little further for you.

jhu54, 2021-01-18:
NVM - module.lds is not being generated during the manual make of the module. Tried another driver for 8814au and it has the exact same issue at the exact same location. It won't load on 5.10.6 anyway. What ever is responsible for generating the module.lds file is not being executed. Also seems to be affecting the BCMWL drivers (dropped in an apple AC card to test) which is normally part of the core drivers set. When I get time, I will update the post on the kernel bugs...

morrownr, 2021-01-18:
I need to ponder the options for debugging.

While I am pondering, take a look at this:

https://www.reddit.com/r/linux_gaming/comments/kwazmj/5107_breaks_nvidia_dkms/

It appears there are additional casualties due to 5.10.7. I went on a search to find the info about the similar problem here with another driver and I was unsuccessful, however, if this is affecting a lot of people it won't take long for someone to figure out what the problem is.

jhu54, 2021-01-18:
Yep I can confirm that issue with the nvidia 460 drivers (I have both AMD and nvidia drivers installed - switch out cards to test different things regularly so easier to just install everything). Feels like they rushed the 10.7 and 10.8 out, not enough widespread testing before release. I posted the bcmwl issue as well - it would appear that it is all linked, so if they fix one, they will fix them all...

jhu54, 2021-01-18:
Thought this might interest you: 

[image]

[image]

So compiled and installed on a fresh download (an hour earlier) of 5.10.8 but same errors in 5.10.7. Also the nvidia 460 now compiled without errors. Device is visible on USB but alas no network connection. Progress made - the 5.10.8 was different to the previous download I took yesterday.

Not sure what I can do next - let me know

jhu54, 2021-01-18:
Sorted it out for 5.10.8

Removed the driver from 10.6 using the remove-driver script. Using mainline, removed both 10.7 and 10.8. Located the 8814au.ko file in the /lib/modules/.../net/wireless and manually renamed it to .old (was still there after the install script). Rebooted

Ran the install-driver script in 10.6, rebooted, install 10.8 through mainline SUCCESS - compiled.

Reboot into 10.8 and we have a working network...

closed 2021-01-18 by jhu54