morrownr/8814au #6: Aireplay-ng

skiocink, 2021-03-12:
Deauthentication does not work.
The adapter sets correctly the channel but ACKs are 0 / 0

morrownr, 2021-03-12:
I'm a little behind on investigating issues so if you are able to narrow down what I need to look at, that would help.

mnkor, 2021-03-13:
I'm experiencing similar but a lot of this is new to me as and i don't know enough to be able to troubleshoot. There are comments/issues on the other github pages for the 8814au driver which mention similar. I think its to do with not being about to do injection? Again, all new to this....

skiocink, 2021-03-13:
> I'm a little behind on investigating issues so if you are able to narrow down what I need to look at, that would help.

Big thanks to you for your hard work but I'm not able to help you. I'm a newbee
Your driver, at the moment, is the better i tested

morrownr, 2021-03-22:
Sorry about being slow to get back to this issue. I should have some time later this week to investigate. I'll need a step by step summary of what you are doing in order to try to duplicate the problem.

I must warn you though, if you are trying to do pen testing or run an ap, this driver is not great at those things. Realtek has not done a good job with this driver and it is getting dated. The only Realtek USB driver that has tested satisfactory at pen testing or running an ap is the 8812au driver. I would go that direction only if you already have the adapter with the 8812au chipset. If you need to spend new money then I recommend reading the following:

 https://github.com/morrownr/USB-WiFi

morrownr, 2021-04-24:
@mnkor  @skiocink 

I finally came to this issue on my to-do list this week. 

I have put some work into monitor mode before on this driver and now i remember why I gave up. This is not a good driver. It is dated and is not standards compliant. This driver is based on the latest source available from Realtek and it is from 2019... and was not good then. My recommendation for Linux used is not to buy adapters based on the Realtek 8814au chipset. Much better alternatives are available.

My advice for those doing pen testing is to get away from Realtek based adapters. Earlier this week I was doing some testing and updated this document:

https://github.com/morrownr/7612u/blob/main/Monitor_Mode.md

That document is about injection testing and deauth testing with adapters based on mt7612u and mt7610u chipsets. The drivers for both of those chipsets are in-kernel and are really good.

Here is the main site where that document is located here:

https://github.com/morrownr/7612u

If you need more information about adapters that do work well with Linux and pen testing, go here:

https://github.com/morrownr/USB-WiFi

Regards,

Nick

closed 2021-06-02 by morrownr