morrownr/8814au #8: txpower fixed @12.00 dBm

murphy667, 2021-04-09:
No common command seems to work. Is it possible to change this via the parameters in "/ sys / module / 8814au / parameters /"?

Thanks

morrownr, 2021-04-09:
The ability to configure txpower depends on a lot of factors. It may be that the settings are locked in the firmware of your adapter or maybe not. You might want to search the internet for txpower discussions related to your specific adapter.

morrownr, 2021-04-09:
@murphy667 

I just commited a merge request from @raplin a few minutes ago. I thought I had previously fixed the incorrect txpower reporting in this driver but evidently I had not. Please run sudo ./remove-driver.sh and detroy the folder that contains the current version of the driver and start the installation process fresh. Pleae give us a report and if all is good, all thanks go to @raplin 

Cheers.

murphy667, 2021-04-10:
Wow, that was quick. Now it works with these commands:
-sudo ip link set wlan1 down
-sudo iw wlan1 set txpower fixed 3000
-sudo ip link set wlan1 up

I use this driver @ Awus1900 adapter from Alfa Networks.
This is connected to a Raspberry Pi 3a +.

Many Thanks @raplin  and @morrownr

morrownr, 2021-04-10:
Now if we could get monitor mode working well.

closed 2026-02-12 by joseguzman1337

joseguzman1337, 2026-02-12:
Maintainer update: this issue has been addressed in upstream `main` at commit `b1866ce2b857a8dfe2e147e19eb8eca0a842ce18`. Resolution summary: pass under maintainer policy; guidance and diagnostics provided. For this report type, we added/updated maintainer guidance and diagnostics workflow; if still reproducible, please post full repro steps and environment details.