morrownr/8814au #9: Raspberry Os, Set Wifi Country. Network Manager (GUI) not working after driver install

murphy667, 2021-04-10:
After installing the driver on RaspberryPiOs (Buster latest release) the network manager (GUI) no longer works.
 It shows: "Set wifi country". 
I tried to set with: 
 -sudo raspi-config: Error: Could not communicate with wpa_supplicant. 

 -sudo ip link set wlan0 down, 
   sudo ip link set wlan1 down, 
   sudo iw reg set DE (or any other), 
   sudo ip link set wlan0 up,
   sudo ip link set wlan1 up: 
       No Error and -iw reg get shows: 
             globalcountry DE: DFS-ETSI
             (2400 - 2483 @ 40), (N / A, 20), (N / A)
             (5150 - 5250 @ 80), (N / A, 20), (N / A), NO-OUTDOOR, AUTO-BW
             (5250 - 5350 @ 80), (N / A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
             (5470 - 5725 @ 160), (N / A, 26), (0 ms), DFS
             (5725 - 5875 @ 80), (N / A, 13), (N / A)
             (57000 - 66000 @ 2160), (N / A, 40), (N / A)
         but GUINetworkmanager not working

 -setting the wifi country over GUI-Settings: Same

- killing processes:   
  PID Name
  381 wpa_supplicant
  408 dhcpcd
  476 wpa_supplicant
 3684 avahi-daemon
 3686 avahi-daemon
 and restarting these : Nope

-setting wifi connection with /etc/wpa_supplicant/wpa_supplicant.conf works fine.

I´m done, I've tried everything I know.... Maybe you can help me further

Thanks a lot

morrownr, 2021-04-10:
If you scroll down far enough in the README.md (front page) here in this repo there is a section called

Set regulatory domain to correct setting in OS

Try that.

Now, let me see. RasPiOS and Network Manager. When last i checked RasPiOS does not ship with Network Manager, does it? I don't have good visibility on what is going on here. Back up and start the story from the beginning.

morrownr, 2021-05-31:
Closing due to lack of follow up.

closed 2021-05-31 by morrownr