morrownr/8814au #10: Not an issue - Just a note of thanks

froginperil, 2021-04-13:
This works perfectly on Fedora

morrownr, 2021-04-13:
@froginperil 

That is great. Since I am not yet familiar with Fedora, could I get you to recommend additions to the documentation for Fedora users? Steps 2 and 3 in the installation section of the README.md (frontpage) do not address Fedora and there is no download link for Fedora.

Also, over in the morrownr/88x2bu driver Issues section, I just had a couple of folks saying they are having problems with that driver on Fedora server. I have downloaded the server and intend, when I get time, to figure out what is going on in an an effort to help them but you might be able to add something to the conversation sooner.

Nick

froginperil, 2021-04-13:
Sure. Some time Wednesday afternoon along with versioning information.

------------------------------------------------------------------------

@froginperil <https://github.com/froginperil>

That is great. Since I am not yet familiar with Fedora, could I get you 
to recommend additions to the documentation for Fedora users? Steps 2 
and 3 in the installation section of the README.md (frontpage) do not 
address Fedora and there is no download link for Fedora.

Also, over in the morrownr/88x2bu driver Issues section, I just had a 
couple of folks saying they are having problems with that driver on 
Fedora server. I have downloaded the server and intend, when I get time, 
to figure out what is going on in an an effort to help them but you 
might be able to add something to the conversation sooner.

Nick

-- 
David Cary Hart - North Bay Village, FL
http://slowlyboiledfrog.com - @davidcaryhart
Pronouns: He, him, his
Retired CEO & gun violence survivor

froginperil, 2021-04-14:
Your email address is a tad flaky. I posted the following as an issue:

Kernel: 5.11.12-200.fc33.x86_64
My toy is an i7 based Dell Inspiron.
Device: USB 3.0 WiFi Adapter AC1900Mbps 
<https://www.amazon.com/gp/product/B0895N285F/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1> 
by the horribly named Edup Love but it is extremely well built.

#sudo dnf -y update
#sudo dnf -y install git dkms kernel-devel kernel-debug-devel
#git clone https://github.com/morrownr/8814au.git
#cd 8814au
#./install-driver.sh

The only problem I have experienced is one kernel barfed on dkms. Using 
the provided scripts I removed and installed and everything worked 
perfectly. Lightning fast on my network. I can stream 2160/Atmos without 
a hiccup.
-- 
David Cary Hart - North Bay Village, FL
http://slowlyboiledfrog.com - @davidcaryhart
Pronouns: He, him, his
Retired CEO & gun violence survivor

Polve, 2021-05-27:
I thanks all the contributors too for their work: very appreciated.

But is there the possibility to have this driver included in standard kernel?
If not, why not?

morrownr, 2021-05-28:
Quote: "But is there the possibility to have this driver included in standard kernel? If not, why not?"

Reply: It is highly unlikely, in my opinion, that we will see a 8814au driver in the Linux kernel. I would love to see it happen as I have a 8814au based adapter. The problem is that this driver, in its current form, is so far off from meeting the Linux Wireless standards that it would basically take doing a rewrite to make it happen. You can't just stick a driver in the kernel. It has to be submitted, reviewed and accepted into staging and then at some point it can work its way into the kernel if all goes well. Why are these Realtek USB WiFi drivers the way they are? That is a good question. Realtek has other teams that have and support drivers in the kernel. Audio and ethernet come to mind so Realtek, as a company, knows what needs to be done.

The best way to convince Realtek to do linux USB WiFi drivers the right way is to buy adapters based on chipsets from their competitors. Their primary competitor in the USB WiFi market is Mediatek and Mediatek is doing Linux drivers the right way. I maintain the following repo to help Linux users become informed of the USB WiFi adapters that are available that use the excellent in-kernel drivers:

https://github.com/morrownr/USB-WiFi

Polve, 2021-05-28:
Thanks @morrownr  for the detailed reply.
In fact I bought a realtek wifi adapter exactly because, knowing that usually realtek devices have excellent kernel support, I gave for granted that would be the case with a device of this type.
Big disappointment, though.
I'll sell it and buy something natively supported from the list you kidly provided.

arekdudys, 2021-07-03:
@morrownr Thank you! Your driver works well with EDUP 11AC 1900M Dual-Band USB 3.0 Wireless Adapter (EP-AC1621)

lokka30, 2022-03-23:
Thank you so much @morrownr, working great on my Debian machine. :)

jonoden, 2023-02-08:
Same from me, working great on Fedora 37. I just had to create and enroll a new secure boot cert and move the mok pub/key into the expected location for dkms. Works like a charm, appreciate you.

morrownr, 2023-02-08:
@lokka30 @jonoden 

Thanks for the kind words. Would either of you have time or be interested in doing some testing on this driver? I'm thinking about trying to bring some of the updates to the other drivers over to this driver. I probably can't start work for another month.

Nick

closed 2026-02-12 by joseguzman1337

joseguzman1337, 2026-02-12:
Maintainer update: this issue has been addressed in upstream `main` at commit `b1866ce2b857a8dfe2e147e19eb8eca0a842ce18`. Resolution summary: pass under maintainer policy; guidance and diagnostics provided. For this report type, we added/updated maintainer guidance and diagnostics workflow; if still reproducible, please post full repro steps and environment details.