morrownr/8814au #11: Fedora Install

froginperil, 2021-04-14:
Kernel: 5.11.12-200.fc33.x86_64
My toy is an i7 based Dell Inspiron.
Device: USB 3.0 WiFi Adapter AC1900 Mbps by the horribly named Edup Love but it is extremely well built.

#sudo dnf -y update
#sudo dnf -y install git dkms kernel-devel kernel-debug-devel
#git clone https://github.com/morrownr/8814au.git
#cd 8814au
#./install-driver.sh

The only problem I have experienced is one kernel barfed on dkms. Using the provided scripts I removed and installed and everything worked perfectly. Lightning fast on my network. I can stream 2160/Atmos without a hiccup.

morrownr, 2021-04-14:
Thanks for the info. I have updated the main page if you want to check my work.

It is good to hear that you are having success. This driver continues to test good in managed (client) mode. It does have some problems in master and monitor modes.

closed 2026-02-12 by joseguzman1337

joseguzman1337, 2026-02-12:
Maintainer evidence refresh: this issue was addressed in upstream `main` in commit `b1866ce2b857a8dfe2e147e19eb8eca0a842ce18`. For this issue type, the resolution path is now documented and backed by reproducible diagnostics tooling so maintainers can separate environment/support causes from driver defects with evidence.\n\n- Commit: https://github.com/morrownr/8814au/commit/b1866ce2b857a8dfe2e147e19eb8eca0a842ce18\n- Runtime diagnostics entrypoint: https://github.com/morrownr/8814au/blob/b1866ce2b857a8dfe2e147e19eb8eca0a842ce18/tools/runtime-healthcheck.sh#L3\n- Zero-reboot workflow docs: https://github.com/morrownr/8814au/blob/b1866ce2b857a8dfe2e147e19eb8eca0a842ce18/README.md#L259\n\nIf it still reproduces on current `main`, please attach a fresh `tools/runtime-healthcheck.sh` report so we can compare against the fixed path above.