morrownr/8814au #13: Computer freeze?

gusarg81, 2021-04-24:
Hi,

It is possible that the PC where I have this device (2 in fact, TP-Link Archer T9UH) is causing hardware Freeze?, Motherboard is new (Gigabyte A320M-S2H V2), Ram in new, CPU from another working computer (AMD A10-9700), Video card from another working computer (AMD RX460). The only things I didn't changed to test are these 2 WIFI adapters, that I use one for each band as AP (with hostapd). 
I am having random hardware freeze, with no related error on logs.

Latest source from this driver, Ubuntu 20.04 and tested right now with mainline kernel 5.11.16 (same happen with 5.8).

Both adapters are connected to a self powered USB 3.0 (another thing I've added just in case).

morrownr, 2021-04-24:
Hi @gusarg81 

After pondering your situation, I do have some thoughts...

If I am understanding correctly, your USB WiFi adapters are both identical and use the 8814au chipset.

I support computers ranging from Raspberry Pi's to large workstations and some do have more than one USB WiFi adapter. None of the ones with multiple wifi adapters have 2 of the same adapter. All work well. I have tested systems with up to 5 adapters installed and working at the same time with various distros but, once again, not with 2 adapters that use the same driver. Concerning the test with 5 adapters, all 5 adapters used different chipsets and different drivers.

I do not think Realtek designed or tested their drivers for the situation where a driver would try to support 2 adapters. A few months ago there was a report in my 88x2bu repo where a guy was trying to setup an AP using two identical Realtek based adapters on one computer and it kept crashing/freezing.  If I recall correctly, it was a report very similar to yours. I did some research and found no evidence of driver support for multiple adapters. Linux will support multiple adapters but I don't think Realtek's drivers can.

My recommendation to the other guy was to evaluate his requirements and get an adapter with a different chipset to replace one of this adapters. He did and all worked well.

If you decide to go that direction, here is a link to my site with recommended adapters with in-kernel drivers:

https://github.com/morrownr/USB-WiFi

That site has a lot of information that could help you with your setup. If you need specific advice, come back here and ask.

I also have this site that gives specific setup examples for multi-adapters and hostapd:

https://github.com/morrownr/7612u

Regards,

Nick

gusarg81, 2021-04-24:
Damn... I didn't know that you can' t use 2 same devices for the same driver.

Based in this new information, do you know any USB device that has 600Mbps on 2.4Ghz band, like this Realktek? (but of course, different/working in Linux chip, like you said).

Here in Argentina we don't have many options and even these 2 where expensive.

Thanks a lot for the info! Meanwhile I will use one with 5Ghz and the replaced router as repeater with 2 bands.

morrownr, 2021-04-24:
Senor Diaz, mucho gusto. Como esta usted? (sorry about the punctuation, English keyboard)

I lived in Argentina for 3 years. From 2000 to 2003. Great country. Many friends.

Back on topic: I see that you are looking to set up a dual band AP. That is cool. Here is my dual band setup:

https://github.com/morrownr/7612u/blob/main/Bridged_Wireless_Access_Point-2.md

https://github.com/morrownr/7612u

I realize that the availability of some products can vary from country to country but Realtek USB WiFi drivers on Linux are not particularly good for AP mode. The only Realtek driver that I have had any stable success with in AP mode is the one for the 8812au (  https://github.com/morrownr/8812au ). My recommendation is to take the follow list of adapters and see what the local computer stores can order for you:

https://github.com/morrownr/USB-WiFi

That Alfa AWUS036ACM is really a good adapter but there are others on the list that will work well depending on the capability you are looking for and what you can get.

Saludos,

Nick

gusarg81, 2021-04-24:
¡Hola! ¡Mucho gusto! ¡Yo estoy muy bien, gracias!

Yes, I am trying to setup a dual band AP and also put all together my nerwork devices into one server that function as Network router/firewall/AP/adware protection (which I already did, only with this problem so far).

As I understand the TP-Link Archer T4U v3 has RTL8812BU, so is supported by 8812au, right? Because here will be the easiest device I will find to buy.

Temporarily, I will setup the removed TP-Link Archer T9UH into the no-use-crappy-rockchip-sbc Rock64 I have there, abandoned.

Thanks for your replies! Is nice to see cool developers around, which is not often! (I am developer too, full-stack, but desktop and web apps).

morrownr, 2021-04-25:
Quote: "As I understand the TP-Link Archer T4U v3 has RTL8812BU, so is supported by 8812au, right?"

Reply: No. The rtl8812au chipset needs the 8812au driver. The rtl8812bu chipset needs the 88x2bu driver. See all here:

https://github.com/morrownr

Off topic: I do keep up with the news in Argentina and I wish you well. Things have been challenging here in the US for the last 4 years.

Hasta luego.

gusarg81, 2021-04-25:
Ahh! that letter changed it all! Sorry I misread that.

And maybe of topic of this thread already, does 88x2bu works any good?

Thanks.

gusarg81, 2021-04-25:
Btw, still experiencing the freeze and now I have only one adapter connected :(

Is there a way I can debug this? (besides enabling logs in the driver).

morrownr, 2021-04-25:
Otro dia, otro problema... es la vida.

Quote: "And maybe off topic of this thread already, does 88x2bu works any good?"

Reply: I think you are on topic so let's keep going. So, does the 88x2bu work any good? Before I answer, here is a little background:

I have a small lab where I test adapters. I have 4 computers that are dedicated to testing. One is a Raspberry Pi, another is a laptop, another is an AIO and then there is a desktop. The ages and capabilities vary but they were chosen to allow me to test a wide variety of configurations. Also, note that I support 5 Realtek drivers which cover basically all of the modern chipsets that Realtek currently has in production. That allows me to have a nearly unique perspective as I can compare and contract the various adapters and drivers.

I have tested AP mode capability a lot. The only Realtek driver that I have found that provides solid, full speed, AP mode support is the 8812au. The 88x2bu driver is good for managed (client) mode but is not so good with master (AP) or monitor modes. The problem is that the 8812au chipset based adapters are getting somewhat hard to find. Now you are probably seeing why I was nudging you in the direction of getting away from Realtek based adapters for AP mode.

Let me propose this: Go online there in Argentina to find local computer stores, online and otherwise, that sell usb wifi adapters and send me the links. My bet is that I help you identify an adapter with a chipset that will work good.

Nick

gusarg81, 2021-04-25:
Nick,

Yeah, there are always be a problem, thats why we exist hahaha!

I know what you mean, I am don't know if i should give back these wifi adapters, because if those are causing the hangs (which I believe they are), does not have any use for me (which also, when the freeze does not happen, these adapters are working full speed without problems).

The thing is, I wanted to replace my TP-Link Archer C7 v5 (which I bricked it yesterday trying to go back to stock firmware, being with Openwrt, and know I must try to un-brick it. At least, it gets the file from tftp server, so maybe I can avoid the serial-unbrick method) and to unify 2 devices (one server, with a number of services). And for the WLAN part, I wanted the top speed for 2.4Ghz and 5Ghz, which is why I've bought these ones.

I have a Mini PCIE card, that I've buyed in eBay a Mini PCI to PCIE adapter so maybe I can use that one later (is an Intel ax200NGW that I have around without use; still I didn't search a lot if its linux driver supports AP mode), also I have a PCIE TP-Link Archer T9E with another problem: Broadcom chip (which still I didn't find the way to enable AP mode, if any of the "supported" drivers has support for that mode).

Now, to no extend my self: I can't find (here at least) any AC1900 device (preferable USB, last of the cases, PCIE would do) that does not have this damn rtl8814au here in any store (even less with mediatek or qualcomm/atheros models).

Common store here is Mercadolibre (www.mercadolibre.com.ar), which is like eBay.

Thanks for all the tips Nick!

morrownr, 2021-04-25:
Bingo! I think we have a winner...  Intel ax200NGW

Intel does great WiFi drivers. They don't do USB but if you have a system that can run that ax200 that is where you can start for now.

What you might want to do is just download and start using the newly released Ubuntu 21.04. It can be good to keep the kernel and the rest of the os at matching supported levels. The 5.11 kernel is not actually supported on 20.04 and there can be incompatibilities doing it that way.

The freeze you are talking about. I have seen that with the 8814au driver in AP mode. I've seen it with the 88x2bu driver in AP mode. Things run along full speed for a long time and then...boom. 

If you have access to Intel PCIe wifi cards and good antennas, go that direction and get your money back for the Realtek adapters. I will take a look for usb adapters at the site you sent as soon as I have time.

Que tenga un buen dia!

Nick

gusarg81, 2021-04-25:
Ok, then till that adapter arrives home (will be around June) I will have to check how to fix this situation (since also my ac1750 Router is bricked...).
So that devices, if works, I will use for the higher frequency, now I have to check what to use for 2.4Ghz.

Thanks Nick for the help!

morrownr, 2021-04-26:
My first bit of advice is to stop bricking routers. <grin>

Quote: "now I have to check what to use for 2.4Ghz"

https://articulo.mercadolibre.com.ar/MLA-851878459-placa-de-red-adaptador-usb-wifi-antena-fija-ralink-5370-_JM

https://articulo.mercadolibre.com.ar/MLA-859871598-placa-de-red-usb-wifi-antena-fija-ralink-5370-microcentr-_JM

The rt5370 chipset should work for 2.4 ghz.

Quote: "TP-Link Archer Tx50e?"

Reply: I don't recommend or buy TP-Link products. They are a Linux users nightmare. Tell me what capability you are looking for and I can help you search.

Quote: "why I didn't read this before?"

Reply: It happens. That site is trying to use the same source that I am...and it is the most recent thing we can find from Realtek, Now you see why I am trying to nudge Linux user trying to run AP or monitor modes away from the 8814au chipset. I have an adapter based on the 8814au chipset. It pulls duty as a client (managed mode) at times. Realtek's driver is junk.

gusarg81, 2021-04-26:
> My first bit of advice is for you to stop bricking routers. 

Never happened to me before! This was the first time :(

I saw those adapters, but are to slow for what I need.

I know what you mean about TP-Link. I never had problems with routers tho. Unfortunately, that brand here is the cheapest and the others better than this one, are way expensive. Like there isn't something in the middle.

Ohh well, I am preparing to return the adapters. Meanwhile I've set up my older TP-Link TL-WDR4300 router as AP, till I can debrick the Archer C7.

Thanks for everything Nick.

morrownr, 2021-04-26:
De nada.

Don't feel bad about bricking the router. I bricked one about 2 years ago.  I have been moving from dedicated router/AP devices to using a Raspberry Pi 4b. You can't really brick a RasPi. Well, I guess you could if you made a serious effort but it is a different situation. The RasPi 4b onboard wifi actually does a good job in AP mode for 2,4g. I use a usb3 adapter for the 5g band. Pretty cool how well it works. It even supports wpa3 and can push over 400 Mb/s on 5g without breaking a sweat.

It sounds like you may want to go with internal cards for your 5g if your box takes internal wifi cards. 11ax is already supported in-kernel with internal cards. Recommend you stay with cards that have Intel, Atheros or Mediatek chipsets that use in-kernel drivers. 

Take care and stop back by to let me know how things turn out.

gusarg81, 2021-04-26:
Yeah, I felt frustrated because is no the first time I've change or upgrade firmware (and now that I need that router again, even worst).
Well, now I've ordered the USB - Serial UART module to see if I can debrick from there (no chance over TFTP, it gets the file, but does nothing later... I've tried with original TP-Link firmware and the one thats already have, from Openwrt).

The one I am using now (TL-WDR4300) does not have very good signal range...

Soooo, waiting now for the minipci-PCIE adapter to continue with the project and try the Intel ax200NGW.

Thanks again and I am sorry for most offtopic here too and I will tell you the news :)

PD: just one thing to say... now I hate Realtek and TP-Link!

logan001, 2021-04-29:
how do i reproduce the freeze:
start ap (in my case 5ghz). connect a device to AP. disconnect the device. wait for computer to freeze (next 10 min in general)
i use ap with kodi for my tv in the other room

gusarg81, 2021-04-29:
> how do i reproduce the freeze:
> start ap (in my case 5ghz). connect a device (phone/tv/whatever) to AP. disconnect the device. wait for computer to freeze (next 10 min in general)
> i use ap with kodi for my tv in the other room

Well, in my case that does not trigger the freeze. I just pretty random (and I've tested your steps because I've read in another report).

Anyways, I don' t have anymore this device, I've returned it back.

logan001, 2021-04-30:
for my asus USB-AC68 thats the result i got probably 70% of the time.

gusarg81, 2021-04-30:
Well, like Nick said, probably you will have always problems with this chip/driver. In my case, were new adapters so I've returned back.

morrownr, 2021-05-31:
Closing issue.

closed 2021-05-31 by morrownr

gusarg81, 2021-06-01:
So this behavior still a mystery eh?

morrownr, 2021-06-02:
We've made some serious progress with this driver lately but it is not clear yet what is not fixed.

gusarg81, 2021-07-16:
@morrownr Hey! This is mostly offtopic, but you know that I have now a Intel AX200 and there is no way I can make it work as AP for 5Ghz frequency (all frequencies are marked as "no IR"). Also does not let me to change the reg domain either. Sorry to contact here for you, maybe you can help me with this.

morrownr, 2021-07-17:
Hey, que pasa en Argentina?

Take a look at this:

https://community.intel.com/t5/Wireless/Intel-AX200-iwlwifi-mvm-driver-issues-in-AP-mode-IEEE802-11ax/td-p/1260312