morrownr/8814au #14: Install error on Realtek WIFI Driver for 8814au

jtome7, 2021-04-26:
[root@localhost 8814au]# pwd
/root/GIT/rtl8814au-morrownr/8814au
[root@localhost 8814au]# ls
8814au.conf  edit-options.sh  install-driver.sh  Makefile  raspi32.sh  remove-driver.sh
core	     hal	      Kconfig		 os_dep    raspi64.sh  save-log.sh
dkms.conf    include	      LICENSE		 platform  README.md
[root@localhost 8814au]# ./install-driver.sh

install-driver.sh version 20210416
Start installation.
Copying source files to: /usr/src/rtl8814au-5.8.5.1
Copying 8814au.conf to: /etc/modprobe.d

Creating symlink /var/lib/dkms/rtl8814au/5.8.5.1/source ->
                 /usr/src/rtl8814au-5.8.5.1

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' -j8 KVER=4.18.0-240.22.1.el8_3.x86_64 KSRC=/lib/modules/4.18.0-240.22.1.el8_3.x86_64/build.....(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.18.0-240.22.1.el8_3.x86_64 (x86_64)
Consult /var/lib/dkms/rtl8814au/5.8.5.1/build/make.log for more information.
An error occurred. dkms build error = 10
Please report this error.

morrownr, 2021-04-26:
Good day.

Is there a reason your system is still using kernel 4.18? That kernel is no longer supported and has not been for some time.

I often have to install the operating system to try to duplicate bugs but I hesitate to do so when the operating system is no longer supported. Can I get you to upgrade to a supported kernel?

Regards.

jtome7, 2021-04-26:
Hello, Nick!

As for the kernel version, I just use whatever the CentOS Linux updates
provide and from my perspective, they seem to be stuck at 4.18 apparently.

I'll have to do some research to see if they have 5.x kernel available ...
just not readily familiar with things like that.

Appreciate your instructive reply ... I'll update you at such time I find
out how to get the 5.x kernel.  Thanks!

*Joel*

*************************** Joel Tome*

*Greensboro, NC *

*jtome7 @ gmail.com <http://gmail.com>****************************

morrownr, 2021-04-26:
Joel,

The following link will give you an idea of what the supported kernels are:

https://en.wikipedia.org/wiki/Linux_kernel_version_history

Upgrading could fix this issue. It might not.

I have no experience with Centos so I don't even know if the installation instructions were what is needed. I'm willing to learn.

Cheers,

Nick

morrownr, 2021-05-31:
Closing issue.

closed 2021-05-31 by morrownr