morrownr/8814au #15: Mesh Interface Creating Error (Operation not supported (-95)

wildpeartree, 2021-04-28:
Hi,

I'm trying to create mesh interface via using Alfa Awus1900 Wi-Fi adapter, but I got error shown as below;

[image]

My os and driver versions ; 
Os: Kali Linux 2020.4 , Kernel  5.10.0-kali5-amd4,  Alfa AWUS1900 Driver:  v5.8.5.1 

After rapid search, I also tried to manipulate kernel mesh configuration as described on this link; https://www.kali.org/docs/nethunter/nethunter-kernel-3-config-2/

but so far, nothing changed. Do you have anything you can suggest ?

Best,

morrownr, 2021-04-28:
Hi wildpeartree,

I can make suggestions but first, let me show you something:

I am going to run $ iw list (and copy the appropriate section from the output)

Realtek rtl8814au based adapter (the Alfa AWUS1900 uses the rtl8814au as you know):

	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * monitor

Mediatek mt7612u chipset based adapter (this is a Alfa AWUS036ACM - AC1200):

	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * AP/VLAN
		 * monitor
		 * mesh point
		 * P2P-client
		 * P2P-GO

Mediatek rt2870 chipset based adapter (this is a very old N150 adapter):

	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * AP/VLAN
		 * monitor
		 * mesh point

What the above shows is that the Realtek driver for the 8814au chipset does not support mesh. It also shows that the Mediatek usb wifi drivers for the entire spectrum of modern devices (N150-AC-1200) do support mesh.

This is not something that can be fixed in the community. This is something that Realtek has to do. I have had to assign my 8814au based adapter to duty on a system that only needs managed mode support because that is all that this driver can reliably do.

My suggestion is to read the following document:

https://github.com/morrownr/USB-WiFi

It is a long document but contains a lot of good information related to USB WiFi adapters and their use on Linux. Let me know if you have any questions.

Nick

wildpeartree, 2021-05-03:
Hi Nick,

First of all, I am very sorry for the late reply. 

You are absolutely right, my "iw list" output does not include mesh mode. As you said, this is a chipset issue.

But as a result of my research, mesh mode is usable, when you can enable to raw data support of chipset. (there are similar cases on ath10k - qca988x) 

I believe that 802.11ac based chipsets should support mesh mode interface. I'll spend a little more time on this case If I can handle this issue, I will let you know.

Best,

closed 2021-05-03 by wildpeartree

morrownr, 2021-05-04:
Hi wildpeartree,

No problem with the late reply. I reply late sometimes because I am busy. I understand.

This is not a chipset issue, it is a driver issue. Realtek does a terrible job with their Linux drivers and the 8814au is the worst.

This is the reason I maintain this site - https://github.com/morrownr/USB-WiFi

In my opinion, Linux users are much better off staying away from Realtek based adapters. At least until Realtek starts doing drivers the right way.

Regards,

Nick