morrownr/8814au #17: Packet injection not working for the ALFA AWUS1900

haroldkummer, 2021-04-29:
problem using USB 3 - 5.10.0-kali3-amd64 2021.1

morrownr, 2021-04-29:
Hello haroldkummer,

I feel your pain. I have an adapter based on the 8814au chipset and injection does work with this driver on it either. This driver is not a good driver. I and others have worked on it. This particular driver is very fragile. It is like driving a Model T Ford on a modern interstate highway. We have a hard enough time just keeping it up with the changes and updates in the kernel.

I maintain a document that address USB WiFi adapters on Linux:

https://github.com/morrownr/USB-WiFi

Near the end of the document there is a section about Realtek chipsets based adapters and the Realtek drivers. Here is a copy and paste from the section on the rtl8814au:

- WPA3 does not work
- no DFS channels support
- no support for interface combinations
- no mesh support
- no AP/VLAN support
- no P2P-client support
- no P2P-GO support
- master (AP) mode can crash your system
- monitor mode support is bad

That list just shows some of the major issues. I could write several more pages. What I have been doing is engaging with sellers and makers to make sure they are aware of how bad this driver is. What would do the most good is for you to contact Alfa and let them know that you are not satisfied with the Linux driver support for your AWUS1900. Write them weekly if you feel like it. Everyone that has an adapter based on this chipset should be writing or calling to complain about the terrible quality of the Linux support.

What needs to happen is Realtek needs to start work toward building a Linux Standards compliant driver that is included in the Linux kernel.

I have a Alfa AWUS036ACM. It is based on the mt7612u chipset and the driver is in the Linux kernel. It is such a pleasure to use. Monitor mode just works. 

https://github.com/morrownr/7612u/blob/main/Monitor_Mode.md

I have additional information on the ACM here...

https://github.com/morrownr/7612u

What I have done here is put my 8814au based adapter to only pulling managed (client) duty because that is about all it does well right now and I am making noise hoping for better support. This driver is so broken that we cannot make any real progress with it. What I recommend to Linux users looking to but a new usb wifi adapter is to look at adapters with Mediatek chipsets:

https://github.com/morrownr/USB-WiFi

Regards,

Nick

conqueror8888, 2021-05-03:
Hello Nick,

I have long been searching for a USB Wi-Fi adaptor that supports WPA3-Enterprise (128- or 192-bit). Would you have any idea about this?

Thanks in advance!

Vincent

https://github.com/morrownr/USB-WiFi<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fmorrownr%2FUSB-WiFi&data=04%7C01%7C%7C45dfd258fcf348b89e1608d90b390c2c%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637553161919197622%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=8VfZQX2jbo204ExrkH4%2BmyIqohBgZXWXUmWwdb0IzRg%3D&reserved=0>

https://github.com/morrownr/7612u/blob/main/Monitor_Mode.md<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fmorrownr%2F7612u%2Fblob%2Fmain%2FMonitor_Mode.md&data=04%7C01%7C%7C45dfd258fcf348b89e1608d90b390c2c%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637553161919197622%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=M9WZuuYo9IZYxD2Q99WNspGAiebyi74I%2BxYZa9oMPsM%3D&reserved=0>

https://github.com/morrownr/7612u<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fmorrownr%2F7612u&data=04%7C01%7C%7C45dfd258fcf348b89e1608d90b390c2c%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637553161919207580%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=09o4rNR30n6uekRo4Uewo3NrQy%2BcAoGHmT9xqkV%2BQt4%3D&reserved=0>

https://github.com/morrownr/USB-WiFi<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fmorrownr%2FUSB-WiFi&data=04%7C01%7C%7C45dfd258fcf348b89e1608d90b390c2c%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637553161919207580%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=jV7APYq4W85EjuIrfAIiNMlX%2F%2F%2Fz2J3kHGOesFp7DSY%3D&reserved=0>

morrownr, 2021-05-03:
Hello Vincent,

Quote: "I have long been searching for a USB Wi-Fi adaptor that supports WPA3-Enterprise (128- or 192-bit). Would you have any idea about this?"

Reply: I have only been testing WPA3-Personel but I suspect the Mediatek USB WiFi drivers in the kernel will support Enterprise as it is part of the documented WiFi standards and the in-kernel drivers are very standards compliant. I've actually been amazed during my recent testing. I tested adapters all the way back to one with a Mediatek rt2870 chipset (N150) that dates back to the very early days of 11n. It tested good for WPA3-P. 

Documentation in general seems to imply that the adapters must include support for WPA3 but my testing seems to indicate that WPA3 support depends only on the driver. If you have a suggestion for a good test setup, I can set up to test Enterprise, I will try to give it shot as I have time...and if I have the equipment.

What you could do is get an adapter based on the mt7612u chipset and test it. There is a lot of information and links to adapters at this site:

https://github.com/morrownr/USB-WiFi

Nick

morrownr, 2021-05-31:
@haroldkummer 

Recent work on this driver seems to have improved monitor and master modes considerably. Users are reporting injection and deauth are working. I did a test 2 days ago on 5 GHz to check injection and deauth and they are working well. That is not to say that everything is working perfectly but everything I tested works fine now.

Please test and report when able. If you find problems, please give me a checklist so that I can try to duplicate the problem.

Cheers

lurky1982, 2021-06-01:
Hi there, id just like to confirm and thank you i have tested and confirmed your repo  through git  then chmod +x install.sh
followed by ./install.sh ... successfully installed my awus 1900 ALFA AC adapter RTL8814AU chipset with injection working, 
in test and in real ive captured 4 different handshakes and can confirm it works on disk and VMWARE Kali .
thaks very much i must have tried install with injection working and failed 50 timmes on 30 + VM machines well done you 👍😁

morrownr, 2021-06-01:
Hi @lurky1982 

Appreciate the report and kind words. This driver is a pain in the ass but it seems we are finally winning some of the battles these days.

lurky1982, 2021-06-01:
yes indeed the driver for this particular adapter and kali has been a struggle... but now it is working it's  awesome😁
usually I wouldn't have persisted with b1 adapter this long but i have tested so many different brands and chipsets for quite few years now, and the ALFA AC  AWUS1900 is so far ahead of others for standard use on windows OS ; I was disappointed when couldn't use it in Wi-Fi pentesting on Kali  but seems now you have solved this 😁👍 thanks again

Sintetic01, 2021-09-12:
Hello, guys. I need help. I have adapter ALFA AWUS1900 (chipset 8814au )and driver 5.8.5.1 running with Kali Linux. Deauthentication is working Ok in 2.4 GHz but not working in 5 GHz band. Could you explain me how you fixed this problem. Do you have any patch for drivers?

morrownr, 2021-09-12:
@Sintetic01 

It would be best if you start a new bug as people may miss this old bug that is here for informational purposes only. It would also help if we knew exactly what hardward and version of Kali you are using as well as the exact commands you are using so that we can duplicate the issue.

To answer the question, all patches are already in the driver. I try not to do much exploratory work on this driver these days as this driver is very fragile and hard to maintain...but it is the best available.

What I generally recommend to Kali users is to check this site out:

https://github.com/morrownr/USB-WiFi

All adapters listed at this site work with Kali and are plug and play. Plus the drivers are Linux Standards compliant and are regularly updated. The ALFA AWUS036ACHM is a really good adapter to use with Kali as it has VERY long range.

Cheers

oom-, 2022-10-14:
Hello @morrownr, thank you for your amazing job.
I currently have a AWUS1900 and I'm really disappointed with it on kali, I don't know how many drivers I already tried but I their is always something wrong about it.

I read your paper here: https://github.com/morrownr/USB-WiFi/blob/main/home/The_Short_List.md

Which one would you recommend between the ALFA AWUS036ACM and ALFA AWUS036ACHM ?
I'm looking to buy something easy to use with monitoring, packet injection, and good range. 
Something that I know will not have issues with the driver in 3 months.

Thank you very much !

morrownr, 2022-10-14:
Hi @oom- 

Let me see here. I wish we had a better driver for the AWUS1900 but we don't. We have worked on this driver but the lack of documentation and the fact that this driver is very fragile has led to a decision just to maintain what we have and hope for an updated release that may never happen. Anyway, on to better things.

> Which one would you recommend between the ALFA AWUS036ACM and ALFA AWUS036ACHM ?

Based on what you are looking for, the answer is easy: ACHM. There are a couple of threads in the issues section over in the USB-WiFi repo where discussions and testing raged on for a lot of messages. In the end, it was obvious that the best overall adapter for Kali was the ACHM. It has the best range of any dual band adapter on the market. I have joked at times with folks that the ACHM could smoke their AC1200 adapters in throughput. Now, it won't smoke them if both adapters are 10 ft from the AP but when you start moving them away, there will be a point where the AC1200 adapters will lose signal and the ACHM is still going strong (he he). I don't recall anyone reporting monitor mode or packet injection bugs with the ACHM. It is reasonably priced and just a handy little adapter to have around.

With that said, indications are that Alfa is working on at least 2 new adapters based on the new mt7921au chipset. I have a Comfast CF-951AX that has the new chipset. Interestingly enough, the driver, mt7921u, made it into the mainline Linux kernel before adapters were available in July. I am testing and so far the chipset appears to be impressive and the driver is in good shape as well. Is everything perfect at this point? No, but it is mostly an issue with the distros catching up with the driver and hardware. Hey, this is a AX3000 chipset and not everything has caught up with WiFi 6e yet. There have been reports from folks using the Comfast CF-953AX that monitor mode is working very well and it even supports 80 MHz channel width with 5 GHz. I got the cheap little Comfast adapter just for testing and don't recommend it.

The Issues and Discussions sections of the USB-WiFi repo have additional info and you can ask questions:

https://github.com/morrownr/USB-WiFi

You can look at the Main Menu of the above repo and select item 2. I has more info on all sorts of adapters and chipsets.

Nick

oom-, 2022-10-14:
Wow @morrownr, thank you very much. I didn't expect a so nicely detailed and clear answer. 

I'm going to order the `AWUS036ACHM` as soon as my `AWUS1900` get refunded., I want some sure value right now but I will definitely keep an eye on the new arrivals.

Thank you very much again, have a great weekend.

closed 2026-02-12 by joseguzman1337