morrownr/8814au #18: (info)[stability] TRx configuration differs depending on USB2/3

abroenni, 2021-05-13:
Hi
so I've done some more digging into the code base and I have merged some commits into my [DEV Branch](https://github.com/abroenni/8814au/tree/dev). The code is ugly.

I first cherrypicked some features I saw on the aircrack-ng version of this driver as I thought more features is better. But while I was going through the debug logs, I noticed that the driver presents the physical device differently to the system depending if it is plugged in into an usb2 port or an usb3 port (This requires properly telling the driver to make use of USB3 with the module param).

This could only be very specific to my device. But I noticed that when in USB3 mode the driver configures the device as a 3T3R with 3 streams. When plugged into a USB2 mode it configures the device as a 2T4R device. This can be confirmed by loocking at the the system journal when the driver is loading, or by doing a `cat hal_info` under /proc/net/rtl8814au/{devicename}/.
I removed the decision block where it checks for USB2/USB3 in the code and now the driver configures the device as a 3T3R regardless of USB2 or 3. 

This then subsequently also activated 3 spatial streams in HT and VHT mode, which can be confirmed with `iw phyX info` This now allows to specify ht_capab=[..][RX-STBC3][..] and vht_capab=[..][RX-STBC-3][..]. I have uploaded some hostapd configs in my branch as well.

I build the driver on raspbian 32bit os and configured it for AC mode with 80Mhz bandwidth. hostpad would come up, but the moment I would use an AC device that can use 80Mhz, the driver driver would be killed by the system and reloaded. the journal would say that there is an over-current on the USB port. But it would not make the entire system hang. My setup is a rpi 3b+ using the built in wifi in station mode to connect  to my main wifi on 5Ghz.

I remembered from an earlier project, that the raspbian os has some stability issues when used in 5ghz mode and I had found that arch arm 64 bit is much better. So I setup arch arm 64bit for this as well and installed the driver (no problems at all). Once the same hostapd conf for AC was set up, I ran the test. And this time it would not crash the kernel module and hostapd stayed up.

I even went as far as setting up a second usb wifi device with an older ralink chip(driver rt2800usb) on the same rpi3b+. Now it is hosting 2 wifis: one for 2.4G (ralink) and one for 5G(8814a). Its been up for 24hours so far and clients can connect and disconnect no problem. But like someone else said about this driver, the crashes happen randomly, so I am only testing and sharing it here with no PR.

I might snap some iperf3 stats for the different configurations. 

cheers 

PS: don't activate Beamforming for AP mode in hostapd nor module config. It really cripples the throughput from what I have found.

morrownr, 2021-05-13:
Hi abroenni,

Quote: "The code is ugly."

Reply: That is a serious understatement. I've used meld a lot to compare code between the various drivers I have up here and I don't know how the Realtek dev keeps this stuff sorted out. This 8814au driver shows signs of a lack of love more so than the other drivers. In fact, this driver is getting hard to maintain. I spend most of my time in the code working on updates related to kernel changes and this driver is a real challenge to keep going. We need a cleaned up, modernized version.

I have measured the current on my adapter based on a 8814au chipset. When I put a good load on with iperf3, it will pull nearly 900 mA. The RasPi 3b, 3b+ and 4b allocate only 1200 mA for the entire usb subsystem and I don't think it handles spikes very well. We could use a powered hub except for the backfeeding issue with RasPis. I have seriously tried to use my 8814au adapter with a RasPi4b but have not been able to find a way to have stable operation in AP mode.

The most stable hostapd settings I have found are:

ht_capab=[HT40+][HT40-][SHORT-GI-20][SHORT-GI-40][MAX-AMSDU-7935]
vht_capab=[MAX-MPDU-11454][SHORT-GI-80][HTC-VHT]
# Note: [TX-STBC-2BY1] causes problems

I was checking out your dev branch. Let me know if you want me to test it and what you want it tested with.

If you can figure out how to stabilize the AP mode so that it is usable, you would be a hero to a lot of folks with this adapter.

morrownr, 2021-05-31:
@abroenni 

Have you looked at this anymore?

abroenni, 2021-06-01:
Since the AP fix was applied, the 3R3T:3 streams change doesn't cause any problems for me (the change is in my dev branch). I got a bit confused about the differences and similarities between spatial streams involving antenna count for TX-RX and STBC stream counts.

The problem I came across is, that the capability is not advertised in wifi beacons. So while the AP is running, when doing an `iw phyX info` on the AP host, it returns 3 -RX STBC streams for 2Ghz and 5ghz bands for HT (for VHT I can see the 3 streams, that is fine). But upon checking beacons transmitted, I cannot see the same stbc streams advertised. For me, it shows, that there are no RX-STBC streams. Meaning, that stations, which try to connect, would do so without ever using 3 stbc streams, even if they could.

I have a rough idea where to look for this in the code, but debugging this is very time consuming. I briefly looked into how to configure dbg in a qemu vm with the device attached to it to properly get debug messages and I can set breakpoints.  But I haven't done this before for a kernel wifi module.

This rabbit hole about the 3 streams may not make much of a difference, if I ever get it to work properly. It would just make the chip a little more user friendly. I think rx-stbc streams are about transmission redundancy, rather than max transmission speed.  

If I have something usable that seems merge-worthy, I will create a PR.

"[TX-STBC-2BY1] causes problems " Can you explain more what the problem with this is, or has this been resolved with the AP fix. I have been running an AP with this since the AP fix, and it seems stable. Debug output seems clean too.

closed 2026-02-12 by joseguzman1337

joseguzman1337, 2026-02-12:
Maintainer update: this issue has been addressed in upstream `main` at commit `b1866ce2b857a8dfe2e147e19eb8eca0a842ce18`. Resolution summary: pass under maintainer policy; guidance and diagnostics provided. For this report type, we added/updated maintainer guidance and diagnostics workflow; if still reproducible, please post full repro steps and environment details.

joseguzman1337, 2026-02-12:
Maintainer evidence refresh: this issue was addressed in upstream `main` in commit `b1866ce2b857a8dfe2e147e19eb8eca0a842ce18`. For this issue type, the resolution path is now documented and backed by reproducible diagnostics tooling so maintainers can separate environment/support causes from driver defects with evidence.\n\n- Commit: https://github.com/morrownr/8814au/commit/b1866ce2b857a8dfe2e147e19eb8eca0a842ce18\n- Runtime diagnostics entrypoint: https://github.com/morrownr/8814au/blob/b1866ce2b857a8dfe2e147e19eb8eca0a842ce18/tools/runtime-healthcheck.sh#L3\n- Zero-reboot workflow docs: https://github.com/morrownr/8814au/blob/b1866ce2b857a8dfe2e147e19eb8eca0a842ce18/README.md#L259\n\nIf it still reproduces on current `main`, please attach a fresh `tools/runtime-healthcheck.sh` report so we can compare against the fixed path above.