morrownr/8814au #20: Need AP Mode testers

morrownr, 2021-05-17:
A pull request was submitted and merged today by @MatteoPaier. He may had found the problem with AP Mode. Early results look promising.

I am requesting everyone that has an interest to test and report on AP Mode here in this issue.

Regards,

Nick

abroenni, 2021-05-17:
Great! Will start testing and report back. But might take till the weekend.

m00tpoint, 2021-05-18:
I can help with this effort. I just got done being disappointed when I made my 8814AU based USB NIC into an AP and only could get 72 mbits/sec from 2 feet away. 

What are the issues with AP mode on this chip that we should be looking to see resolved? I'm hoping speed is one of them .... :-)

MatteoPaier, 2021-05-18:
Have a look at #19.
My main goal was to resolve the kernel freeze that may happen after some client connections/disconnections, but some early tests with `iperf3` by @morrownr suggest that also the speed is positively affected.

morrownr, 2021-05-18:
I did some quick testing this morning after @MatteoPaier sent in the pull request. Things looked good. Since then I changed over our RasPi4b AP to use my 8814au based adapter. The users at this location have been pounding it. I've thrown some extra gas on the fire by running long running iperf3 tests. I've really been pounding it and it is totally solid here so far.

I'm on 5 GHz channel 36, channel width 80 and I am seeing iperf3 kick out throughput numbers over 550 Mb/s. My internet is 1 gb fiber and this beast of  a stick is not slowing things down much. For what it is worth, I don't have another AC1900 adapter so I am getting this speed with an AC1200 adapter as a client.

For anyone else that stops by to test: This driver has never been able to handle 11AC a full up settings without crashing until now. This is astounding. (now, who has the time and knowledge to fix monitor mode and WPA3?)

If anyone needs to some info to get hostapd going, let me know. The more testers, the better.

Oh, I forgot to mention. If you are testing with a RasPi, you need to know that 8814au based adapters can and will take over 900 mA when pushed hard. Given that the total power available to the USB subsystem in a Pi4b, Pi3b and pi3b+ is only 1200 mA, you have very little capacity for anything but this adapter.

Nick

m00tpoint, 2021-05-18:
Did a quick install -- thank you for the very very simple scripts! I was previously running the Aircrack driver. My adapter is an EDup (it has 4 antennas, it must be good!). The machine is a lilttle HP Elitedesk with an Intel I5 6600T running Mint 20.1. A quick trip to modprobe land and I now have the adapter on USB 3 (yay). I was getting  nominal speed of 72 Mbits in iwconfig and actual throughput to Comcast's speed test of around 45 Mbits. That is up about 15 mbits on both with this driver to 87 and 60ish. Besides the USB tweak, I also set rtw_vht_enable=2. The AP itself seems to think it has a link speed of 215ish mbits/sec, which is somewhat entertaining given actual results. 

So things look a bit better for me than they did, but still a far cry from what I'd consider a usable wifi router. (My Internet speed with wired connection tops out just under 500 mbits.)  I've done very little of my homework, so I probably have a lot of tweaking to do. 

Old men like me should not be up this late on a work night, time for bed! :-)

m00tpoint

morrownr, 2021-05-18:
Hi mOOtpoint,

I can relate to the old men thing. I'm not exactly young myself. Learned programming in college using FORTRAN and COBOL, long before PCs and Linux existed.

How about we look into some tweaks, My AP setup here:

RasPi4b
RasPiOS - the standard 32 bit version
CAT 6 ethernet cable plugged into the RasPi feeding internet
hostapd
generic 8814au usb adapter connected into a usb3 port (actually, I have multiple adapters and change out as I need to test which means I lines in my hostapd that I have to deactivate and activate depending on the adapter. I had long ago given up using the 8814au as an AP.)

You told me you are running Mint 20.1 so that means a 64 bit Ubuntu 20.4 based system running on a x64/amd64 based system. Got it.

What I need to know is what software is turning this system into an AP?
I also need to know how that hp system is connected to the internet?

I'm seeing consistent, stable speeds of around 550 Mb/s with my RasPiAP. Yes, with the new patch, things are so fast and stable that I smell smoke... at least with 5 GHz. Still need to test 2.4 Ghz.

morrownr, 2021-05-18:
Update:  I decided to try the AP Mode patch on the 88x2bu driver since it has suffered from the same crashing AP Mode problem. The patch seems to have worked on the 88x2bu driver for 5 GHz but not for 2.4 GHz. After a report, I tested 2.4 GHz with the 88x2bu driver and it was indeed crashing with 2.4 Ghz. It is stable with 5 GHz.

So, I decided to come back here to test 2.4 GHz. I am finding it to be stable here. 

/etc/modprobe.d/8814au.conf

The following gives rock solid stability with 2.4 GHz:

options 8814au rtw_drv_log_level=4 rtw_led_ctrl=1 rtw_vht_enable=1 rtw_power_mgnt=0 rtw_switch_usb_mode=0

The following are the settings that I normally use with 5 GHZ:

options 8814au rtw_drv_log_level=4 rtw_led_ctrl=1 rtw_vht_enable=2 rtw_power_mgnt=1 rtw_switch_usb_mode=1

The 5 GHz settings used for 2.4 GHz show a slight drop off and a retry about every 30 seconds so that is an abnormality that I am going to try to get sorted out and documented. Which of the 3 parameters is the problem? Hmmm...

I can also paste my hostapd.conf settings for 2.4 and 5 GHz if it would benefit anyone.

So, it appears so far that the patch has stabilized both bands in AP mode. I will be chasing down the problem with 88x2bu while testing continues.

m00tpoint, 2021-05-18:
> Hi mOOtpoint,

> What I need to know is what software is turning this system into an AP?
> I also need to know how that hp system is connected to the internet?

I'm running hostapd and the HP is cabled to the Comcast router I hope to replace. I get just under 500 mbits on Comcast speed tests from the HP.

> I'm seeing consistent, stable speeds of around 550 Mb/s with my RasPiAP. Yes, with the new patch, things are so fast and stable that I smell smoke... at least with 5 GHz. Still need to test 2.4 Ghz.
That's impressive!! A somewhat related question ... is dual band possible without another wifi interface?

morrownr, 2021-05-18:
Now I am curious. I've had some guides up for a while concerning how to set up AP mode with a RasPi but somebody requested one for Ubuntu the other day so I started looking at it. Here is where I am so far:

https://github.com/morrownr/7612u/blob/main/Bridged_Wireless_Access_Point-Ubuntu-1.md

Would you mind taking a look? I need someone to save this poor soul from kicking my dog. Too many moving networking parts in too many distros.

Nick

m00tpoint, 2021-05-18:
I'll be happy to later this evening

m00tpoint, 2021-05-19:
I grabbed what seemed like the appropriate lines from your guide to add to my hostapd.conf. I'm now seeing 433 nominal / 245 actual per Comcast speed test. At this point the older wireless NIC in the laptop is going to become the bottleneck. I had my stepson try with his newer (Windows) machine, and he's seeing 122 down and 23 up, not sure why the discrepancy. 

Yes!

Re: Your guide, I'll look through it more carefully here, but I immediately note there's no information about how to create the br0 interface. I'm using a routed WLAN, so I'll be of limited help, but on the other hand I fiddle with bridges for other reasons on a 3rd machine. I'll see what I might have in my notes.

More later, and many thanks, 
Denis

m00tpoint, 2021-05-19:
OK .... I went down and raided my work machine of its usb wifi adapter, which as it turns out is an 88x2bu. It spends its life at 2.4 GHz on my work windows laptop due to packet loss at 5 GHz from that piece of the house. 

With the 88x2bu on the laptop here, using your driver, and rtw_vht_enable at 1, I max out the internet connection on 5GHz to the 8814au AP. I'll have to fiddle a bit in the next day or two with iperf to see what happens above 450 mbits.

morrownr, 2021-05-19:
Quote: "I grabbed what seemed like the appropriate lines from your guide to add to my hostapd.conf. I'm now seeing 433 nominal / 245 actual per Comcast speed test."

Reply: Good. While testing over the last 2 days, I have been seeing 550 Mb/s using iperf3 between a client with a 8812bu (88x2bu driver) and a RasPi / 8814au based AP. You are welcome to post your hostapd.conf and 8814au.conf and I'll be glad to suggest tweaks.

Quote: "I don't believe deleting and linking of /etc/resolv.conf is necessary."

Reply: I'm going to remove that section. The biggest remaining issue seems to be making sure Network Manager and all associated things are shut down in the desktop version of the os and that Netplan and all associated things are shut down in the server version of the os. While this is a bidged setup, I also have a request for a routed setup and I plan to work on that when I finish this bridged setup so if you have incite on doing routed, that would be great.

Nick

m00tpoint, 2021-05-20:
Hi Nick,

With the antennas for both adapters a few inches apart, I'm getting iperf runs in the 490-510 mbit range between the two machines, with a nominal connection speed in iwconfig of 867 on the 88x2bu and 1.3 Gb on the 8814au. I'll post my current hostapd.conf and 8814au.conf below. 

I'll put together something about configuration in routed mode to help with the guide. 

Dennis

m00tpoint, 2021-05-20:
8x2bu.conf (client machine):
`options 88x2bu rtw_drv_log_level=0 rtw_led_ctrl=1 rtw_vht_enable=1 rtw_power_mgnt=1 rtw_switch_usb_mode=0`

8814au (ap machine):
`options 8814au rtw_drv_log_level=0 rtw_led_ctrl=1 rtw_vht_enable=2 rtw_power_mgnt=1 rtw_switch_usb_mode=0`

hostapd.conf
'interface=wlan0
#channel=6
#ieee80211n=1
#hw_mode=g
#ssid=mercury_2
#wpa=2
#wpa_passphrase=ygcd448c
#wpa_key_mgmt=WPA-PSK
#rsn_pairwise=CCMP
#auth_algs=1

interface=wlan0
ieee80211n=1
ieee80211ac=1
wmm_enabled=1

hw_mode=a
ssid=mercury_5
wpa=2
wpa_key_mgmt=WPA-PSK
wpa_passphrase=<redacted>
rsn_pairwise=CCMP
auth_algs=1

# From Nick

driver=nl80211
country_code=US
beacon_int=100
dtim_period=2
max_num_sta=32
macaddr_acl=0
rts_threshold=2347
fragm_threshold=2346

# rtl8812au - rtl8811au -  rtl8812bu - rtl8811cu - rtl8814au
# band 1 - 2g - 20 MHz channel width
#ht_capab=[SHORT-GI-20][MAX-AMSDU-7935]
# band 2 - 5g - 40 MHz channel width
ht_capab=[HT40+][HT40-][SHORT-GI-20][SHORT-GI-40][MAX-AMSDU-7935]
# Required for 80 MHz width channel operation on band 2 - 5g
vht_oper_chwidth=1
#
# Use the next line with channel 36  (36 + 6 = 42) band 2 - 5g
#vht_oper_centr_freq_seg0_idx=42
#channel=36
# Use the next line with channel 149 (149 + 6 = 155) band 2 - 5g
vht_oper_centr_freq_seg0_idx=155
channel=149

morrownr, 2021-05-20:
Recommendation:

In your 88x2bu,conf and 8814au.conf, I recommend one change: rtw_switch_usb_mode=1

That takes you to USB3 mode and should increase speed.

In your hostapd.conf file, I am not seeing 4 lines that I expected:

bridge=bro

I'm curious about your setup as it appears you are not bridging interfaces.

I'm also not seeing a vht_capab= line:

vht_capab=[MAX-MPDU-11454][SHORT-GI-80][HTC-VHT]

The above should work with the 8814au.

I also don't see:

ieee80211n=1
ieee80211ac=1

m00tpoint, 2021-05-22:
I'll do some testing later tonight; right now there's too much 5GHz noise in the house for valid test numbers. 

When I rebooted the AP machine today, after reboot Network Manager was managing both of its interfaces, even though both mac addresses were disabled in /etc/NetworkManager/NetworkManager.conf. (!) I ended up stopping and disabling it (sudo systemctl stop NetworkManager; sudo systemctl disable NetworkManager.) As soon as I did, the configs in /etc/network/network.d became active. I'd recommend that your guides tell people to do that to Network Manager, and not to rely on that conf file being respected. 

As far as my non-bridged setup goes, you need two things. First, "sysctl -w net.ipv4.ip_forward=1." This tells the kernel to route packets from other systems between the various interfaces -- it turns a host into a router. So when a packet arrives at 192.168.200.1 (my wlan0 IP), destined for the Internet, the system sends it out eno1 to 10.0.0.1 because that's where its default route points. If/when I want a wired LAN, I'll give that subnet a third IP address (192.168.201.1/24 or something), and traffic between the wired and wireless LAN's will route across the system, instead of bridging across and sharing the same subnet. It's also possible to create a guest subnet and SSID with yet a fourth network, and prevent it from accessing the local LAN but allowing it to access the Internet. Etc. etc. Then you just assign a static IP to wlan0 in /etc/network (192.168.200.1/24 in my case, set up DHCP or a static IP on the host machine, and you're off to the races. 

The other piece that's needed is to NAT that packet so that the packet's source IP is altered from 192.168.200.200 (my laptop's wireless IP) to 10.0.0.220 (the AP's eno1 interface). This is done with "sudo iptables -t nat -A POSTROUTING -o eno1 -j MASQUERADE," followed by "sudo iptables-save" so that it survives reboot. That NAT command translates as, "After you route the packet, if you're going to send it out eno1, use your eno1 interface IP as the source for the packet instead of whatever it currently has." This or something like it is of course needed in a bridged setup also, so that Internet traffic is natted to the ISP-facing interface. 

Dennis

m00tpoint, 2021-05-22:
OK, had some time tonight, but didn't make any progress. I have introduced some sort of typo into my current copy of hostapd.conf, and as sometimes unfortunately happens, I did not back up my working copy before altering it. Uncommented, here's one that "works" at about 1 mbit (comment lines removed). I'm hoping you can spot something that I'm missing.

interface=wlan0
driver=nl80211
ctrl_interface=/var/run/hostapd
ctrl_interface_group=0

ssid=mercury_5

country_code=US

ieee80211d=1
ieee80211h=1

hw_mode=a
channel=36

beacon_int=100
dtim_period=2
max_num_sta=32
macaddr_acl=0
rts_threshold=2347
fragm_threshold=2346

wpa_passphrase=ygcd448c
auth_algs=1
ignore_broadcast_ssid=0
wpa=2
rsn_pairwise=CCMP
wpa_key_mgmt=WPA-PSK

ieee80211n=1
wmm_enabled=1
ht_capab=[HT40+][HT40-][SHORT-GI-20][SHORT-GI-40][MAX-AMSDU-7935]

ieee80211ac=1
vht_capab=[MAX-MPDU-11454][SHORT-GI-80][HTC-VHT]

vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=42

morrownr, 2021-05-22:
Dennnis,

That looks good. For maximum performance you need the following in  /etc/modprobe.d/8814au.conf but you likely have them set or you would not be seeing the numbers that you are:

rtw_vht_enable=2 rtw_switch_usb_mode=1

To test or use 2.4 GHz, you would need some changes along these lines:
```
hw_mode=g
channel=6

#ieee80211ac=1
#vht_capab=[MAX-MPDU-11454][SHORT-GI-80][HTC-VHT]

#vht_oper_chwidth=1
#vht_oper_centr_freq_seg0_idx=42
```

If you have time to test 2.4 GHz, it would be appreciated.

I understand what you are saying about a routed setup. I'm going to play with that as I have time. You do realize that you can create a separate network/guest access with a bridged setup? Lotsa stuff to play with.

Nick

m00tpoint, 2021-05-22:
Yes I have the usb and vht_enable set on the 8814. It's doing quite well I'd say. I'll fiddle with 2.4Ghz and let you know how it goes

m00tpoint, 2021-05-22:
On 2.4 GHz I'm seeing about 50 mbits, during the middle of the day with a fair amount of RFI. I'd call that successful.

abroenni, 2021-05-23:
Hi, so here are my tests so far. Sorry for being late to the testing party ;-)

|Device| OS| Kernel| USB2/3|			
|---------|------------------------|----|---|
|Raspberrypi 3b+| Arch Arm aarch64| 5.11.4-1| USB2|
				
|Driver(/proc/net/rtl8814au)| TRx Path| max_tx_cnt | tx_nss_num | rx_nss_num| 
|------------------------------------|------------|------------------|-----------------|-----------------|
|     hal_trx_mode                | 2T4R     |2                  |2                   | 3                 |

| hostapd.conf |  setting|
| ----                 | ----------|
|ht_capab | [HT40+][HT40-][SHORT-GI-20][SHORT-GI-40][MAX-AMSDU-7935] |
|vht_capab| [MAX-MPDU-11454][SHORT-GI-80][HTC-VHT]|

| 8814au.conf| setting|
|-------              |-------  | 
|    options       | rtw_drv_log_level=0 rtw_led_ctrl=0 rtw_vht_enable=2 rtw_power_mgnt=0 rtw_switch_usb_mode=2|

| Mbits/sec   |              | N-2G (20Mhz) | N-2G (40Mhz)* | N-5G (40Mhz) | AC-5G (80Mhz) |
|-------------|--------------|--------------|---------------|--------------|---------------|
|Linux laptop | receive      |  75.2        |               |              |               |			
|             | transmit     |  52.6        |               |              |               |			
|mobile phone (older)  | receive      | 101.0        |               | 272          | 287           |
|     | transmit     | 107.0        |               | 158          | 163           |
|mobile phone (newer) | receive      | 107.0        |               |              | 290           |
|	  | transmit     | 106.0        |               |              | 245           |
|older macbook pro | receive |  84.9        |               | 203          |	           |
|	          | transmit     |  54.0        |               | 45           |               |	
					
*requires patched hostapd to force 40Mhz in 2Ghz band					

Measured how:
Run the 8814au-AP. Start iperf 3 in servermode on 8814au-AP: `iperf3 -s -B {IP of 8814au device}`
Connect with client to the 8814au wifi. On client run iperf3 for 
receive  = `iperf3 -c {IP of 8814au device} -t 30`
transmit = `iperf3 -c {IP of 8814au device} -t 30 -R`

So "receive" and "transmit" are seen from the AP.

While I was testing, I noticed that I don't have that many devices capable of AC mode, especially capable of 3 streams. So these values are probably lower than max, however they represent real device values. I'll test some USB3 setups as well.

Arch arm aarch64 linux on the Pi is also stable with the fix.  
Oh and I run my rpi3b+ using routes and not using masquerading at all. I could help with that, I guess.

morrownr, 2021-05-23:
@m00tpoint 

Quote: "On 2.4 GHz I'm seeing about 50 mbits, during the middle of the day with a fair amount of RFI. I'd call that successful."

Reply: I'm not seeing any problem with 2,4 here either. I had applied the patch also to the 88x2bu driver as it had similar problems to what we were seeing here with AP mode. With the 88x2bu driver, if using a USB3 port and rtw_switch_usb_mode=1, 2.4 GHz would show the same data package stall we had been seeing but 5 GHz worked good. During testing I discovered that if I use rtw_switch_usb_mode=2 with 2.4 GHz, everything worked fine.

I think we are close to being able to declare that AP mode is working and stable.

morrownr, 2021-05-23:
@abroenni 

Good report. Those numbers indicate solid USB2 AP mode operations. There are going to be a lot of folks happy about AP mode working.

Have you tried rtw_power_mgnt=1 or rtw_power_mgnt=2 to see what the results are. This 8814au chipset uses a lot of power and those settings can bring down power usage a little which might be worthwhile if there are no adverse effects.

m00tpoint, 2021-05-24:
It certainly is on my machine and in my testing. Glad I’m able to help a bit with improving the driver situation for these Linux-challenged chips. 😊

abroenni, 2021-05-25:
@morrownr You asked earlier about "who has the time and knowledge to fix monitor mode and WPA3?". Maybe consider creating two "Feature" - issues for those ideas? (just asking, cuz its not my git here ;-) )

morrownr, 2021-05-25:
A guy in another repo helped me chase the WPA3 issue down the rabbit hole. We aren't going to be able fix it. Among other things, it involves areas where these Realtek drivers are not Linux Standards compliant. We would end up breaking things that are working on a regular basis so WPA3 is going to depend on Realtek for a fix. Will they fix it? I have no idea. 

Monitor mode is something we stand a better chance to fix and I have put in some time on it and it works better now that is did when we got the code from Realtek but it is something that will take a lot of work. Much of the time I use toward this site is involved in basic maintenance like keeping the drivers in sync with new kernels. My time will be limited over the next 2 months but I welcome any and all to work on monitor mode. I will start an issue for folks to use.

morrownr, 2021-05-25:
I am declaring victory. AP mode is working with this driver on 8814au based adapters. I have beat it to death and I need to reconfigure my hardware to work on another project. My hat is off to @MatteoPaier for finding the fix. Thank you.

Nick

joseguzman1337, 2026-02-12:
Maintainer evidence refresh: this issue was addressed in upstream `main` in commit `b1866ce2b857a8dfe2e147e19eb8eca0a842ce18`. For this issue type, the resolution path is now documented and backed by reproducible diagnostics tooling so maintainers can separate environment/support causes from driver defects with evidence.\n\n- Commit: https://github.com/morrownr/8814au/commit/b1866ce2b857a8dfe2e147e19eb8eca0a842ce18\n- Runtime diagnostics entrypoint: https://github.com/morrownr/8814au/blob/b1866ce2b857a8dfe2e147e19eb8eca0a842ce18/tools/runtime-healthcheck.sh#L3\n- Zero-reboot workflow docs: https://github.com/morrownr/8814au/blob/b1866ce2b857a8dfe2e147e19eb8eca0a842ce18/README.md#L259\n\nIf it still reproduces on current `main`, please attach a fresh `tools/runtime-healthcheck.sh` report so we can compare against the fixed path above.