morrownr/8814au #21: More clarification for install process

Xendergo, 2021-05-22:
I'd suggest mentioning that you'd need to install the correct version of the kernel headers, because (at least on manjaro) typing in `pacman -S linux-headers` gives you a bunch of options for versions to install.

As a complete newbie to linux, I had no idea what the choice even meant, and I picked the default, which was wrong, though I figured this out eventually after a bunch of research

morrownr, 2021-05-22:
Good morning Xendergo,

You are right. Let me propose something. Can I get you to write up an appropriate note that can be included in the right places in the README? My experience with Manjaro is mostly just what some Manjaro users here have taught me so that I can test with it. I keep a partition on one system for testing purposes but I always have to go to my notes when working with it. I am a Debian/RasPiOS/Ubuntu/Linux Mint kinda guy and I prefer somebody that actually knows what they are doing write this up.

Nick

Xendergo, 2021-05-22:
I can try, but keep in mind I'm really bad at writing documentation...

morrownr, 2021-05-23:
@Xendergo 

I merged your pull request. Thanks.

Quote: "Note: if it asks you to choose a provider, make sure to choose the one that corresponds to your version of the linux kernel (for example, linux510-headers for version 5.10) if you have the incorrect version installed, you'll have to uninstall it and reinstall the correct version."

What matters is if this is helpful to Manjaro users but those of us that are not that familiar with Manjaro might not fully understand some things as the note is currently written. So, maybe we can work on clarifying some things:

"Note: If (it) asks..."

What is it? Should we use the word Manjaro?

"...choose a (provider)..."

The term provider is not clear to me. If you can clarify what is going on at that point, maybe we can clarify the wording.

If we can get to the point where inexperience Manjaro users can't mess this up, we will have met our goal. There is no doubt that this part of Manjaro is a little confusing.

Xendergo, 2021-05-24:
I would probably use the word "pacman" since it would be the package manager asking you to choose a provider, not the OS itself

"Choose a provider" is just the wording used by pacman, I don't know what other wording we would use.

I'll have to be away from my pc for a while, so it might be tricky to test stuff or send screenshots. However, at this point, what's happening is that you types in the command, and pacman loads something, and it asks you to choose a provider, and it gives you a bunch of options, and you type in a number to choose one, and then you can go and install it. I'm guessing `linux-headers` is a [virtual package](https://wiki.archlinux.org/title/Pacman#Virtual_packages)

morrownr, 2021-05-31:
Okay, I've cleaned the wording up as best I can given that I don't have time to go play with Manjaro right now. If you think we can improve the wording at some point, let me know. The wording does not have to make sense to the general public, just to Arch and Manjaro users.

Xendergo, 2021-06-01:
Ok, cool!

dragon99919, 2021-08-30:
Actually, thinking because it is still open...
Is there any supported way to install the driver on multiple kernels, like we can install in Manjaro for example?
Should I install the driver and then, when installing another kernels with headers, DKMS should handle the building procedure by itself?

morrownr, 2021-08-30:
To a large degree, Manjaro is still a mystery to me. Due to a lack of time to do more exploring, my experience is mostly with Debian based systems.

Ubuntu (Debian based) and all of the systems built on the Ubuntu base, what you are describing is how it is done. When new kernels come in, DKMS compiles a new driver for the new kernel and it will compile the driver for old kernels if any of the old kernels are missing a driver and it is will delete the driver for all kernels when uninstalling.

dragon99919, 2021-08-30:
Well, it seems that surely something indeed goes wrong here... Here's the output of the kernel installation (I was sitting on 5.10 LTS and tried with 5.12):
`The following packages will be installed:
linux512
linux512-headers
linux512-nvidia

Starting
resolving dependencies...
looking for conflicting packages...
Packages (3) linux512-5.12.19-1  linux512-headers-5.12.19-1  linux512-nvidia-470.63.01-1
Total Download Size:   157.06 MiB
Total Installed Size:  264.69 MiB
:: Proceed with installation? [Y/n]
:: Retrieving packages...
linux512-5.12.19-1-x86_64 downloading...
 linux512-headers-5.12.19-1-x86_64 downloading...
 linux512-nvidia-470.63.01-1-x86_64 downloading...
checking keyring...
checking package integrity...
loading package files...
checking for file conflicts...
checking available disk space...
:: Processing package changes...
installing linux512...
>>> NOTE, 5.12.19 was the last maintenance release by Greg Kroah-Hartman.
This kernel series is now marked EOL (end of life).
    
    It is recommend to move on to linux510 for LTS series or to
    linux513 for the current stable.
Use followed cmd to do that: 'sudo mhwd-kernel -i linux510'
    or for current stable:       'sudo mhwd-kernel -i linux513'
Optional dependencies for linux512
    crda: to set the correct wireless channels of your country [installed]
installing linux512-headers...
installing linux512-nvidia...
In order to use nvidia module, reboot the system.
:: Running post-transaction hooks...
(1/7) Arming ConditionNeedsUpdate...
(2/7) Updating module dependencies...
(3/7) Install DKMS modules
==> dkms install --no-depmod -m rtl8814au -v 5.8.5.1 -k 5.12.19-1-MANJARO
Error! Bad return status for module build on kernel: 5.12.19-1-MANJARO (x86_64)
Consult /var/lib/dkms/rtl8814au/5.8.5.1/build/make.log for more information.
==> Warning, `dkms install --no-depmod -m rtl8814au -v 5.8.5.1 -k 5.12.19-1-MANJARO' returned 10
==> depmod 5.12.19-1-MANJARO
(4/7) Updating linux initcpios...
==> Building image from preset: /etc/mkinitcpio.d/linux512.preset: 'default'
-> -k /boot/vmlinuz-5.12-x86_64 -c /etc/mkinitcpio.conf -g /boot/initramfs-5.12-x86_64.img
==> Starting build: 5.12.19-1-MANJARO
-> Running build hook: [base]
-> Running build hook: [udev]
-> Running build hook: [autodetect]
-> Running build hook: [modconf]
-> Running build hook: [block]
-> Running build hook: [keyboard]
-> Running build hook: [keymap]
-> Running build hook: [filesystems]
-> Running build hook: [fsck]
==> Generating module dependencies
==> Creating gzip-compressed initcpio image: /boot/initramfs-5.12-x86_64.img
==> Image generation successful
==> Building image from preset: /etc/mkinitcpio.d/linux512.preset: 'fallback'
-> -k /boot/vmlinuz-5.12-x86_64 -c /etc/mkinitcpio.conf -g /boot/initramfs-5.12-x86_64-fallback.img -S autodetect
==> Starting build: 5.12.19-1-MANJARO
-> Running build hook: [base]
-> Running build hook: [udev]
-> Running build hook: [modconf]
-> Running build hook: [block]
-> Running build hook: [keyboard]
-> Running build hook: [keymap]
-> Running build hook: [filesystems]
-> Running build hook: [fsck]
==> Generating module dependencies
==> Creating gzip-compressed initcpio image: /boot/initramfs-5.12-x86_64-fallback.img
==> Image generation successful
(5/7) Updating Kernel initcpios for Nvidia-DRM...
(6/7) Refreshing PackageKit...
(7/7) Updating Grub-Bootmenu
Generating grub configuration file ...
Found theme: /usr/share/grub/themes/manjaro/theme.txt
Found linux image: /boot/vmlinuz-5.12-x86_64
Found initrd image: /boot/intel-ucode.img /boot/initramfs-5.12-x86_64.img
Found initrd fallback image: /boot/initramfs-5.12-x86_64-fallback.img
Found linux image: /boot/vmlinuz-5.10-x86_64
Found initrd image: /boot/intel-ucode.img /boot/initramfs-5.10-x86_64.img
Found initrd fallback image: /boot/initramfs-5.10-x86_64-fallback.img
Warning: os-prober will be executed to detect other bootable partitions.
Its output will be used to detect bootable binaries on them and create new boot entries.
Adding boot menu entry for UEFI Firmware Settings ...
Detecting snapshots ...
Root partition isn't a btrfs filesystem.
This script only supports snapshots of the btrfs filesystem.
If you think an error has occurred , please file a bug report at " https://github.com/Antynea/grub-btrfs "
Nothing to do. Abort.
Found memtest86+ image: /boot/memtest86+/memtest.bin
done

Done ...`

And that's the build.log file:
`DKMS make.log for rtl8814au-5.8.5.1 for kernel 5.12.19-1-MANJARO (x86_64)
Mon Aug 30 21:43:02 CEST 2021
make: uname: No such file or directory
make: pwd: No such file or directory
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.12.19-1-MANJARO/build M=  modules
make: pwd: No such file or directory
make: make: No such file or directory
make: *** [Makefile:2383: modules] Error 127`

morrownr, 2021-08-31:
Just had to find a bug didn't you? I am behind on working issues right now so let me suggest this:

Post a new bug with a title like "Need help with Manjaro" and show the same info you posted here. There are some Manjaro users that watch this repo but they may not see this issue hidden down here like it is.

dragon99919, 2021-08-31:
Sure, will do!

joseguzman1337, 2026-02-12:
Maintainer update: this issue has been addressed in upstream `main` at commit `b1866ce2b857a8dfe2e147e19eb8eca0a842ce18`. Resolution summary: pass under maintainer policy; guidance and diagnostics provided. For this report type, we added/updated maintainer guidance and diagnostics workflow; if still reproducible, please post full repro steps and environment details.