morrownr/8814au #23: set_wiphy_netns is not available

yananet, 2021-05-25:
Thanks for all your effort on this project.
It looks that `set_wiphy_netns` is not available, I verified with `8814au`, and `8812bu` but not worked.
```
$ iw list
	Supported commands:
	.......
		 * set_wiphy_netns
	.......
```
Is there any workaround to enable this?

morrownr, 2021-05-25:
I'm not aware of any workarounds. I just checked 3 adapters here and they support it:

ALFA AWUS036ACM
TEROW ROW02FD
ALFA AWUS036ACHM

All use the in-kernel Mediatek mt76 driver:

https://github.com/morrownr/USB-WiFi

I started that site because so many folks here were asking for things that Realtek should be providing but are not. The Mediatek  drivers are feature rich and Linux Wireless standards compliant. Realtek's drivers are not.

yananet, 2021-05-25:
Thanks for the comment, and again for your hard work on this project.
I have 2 8814au Wi-Fis, but at least I could see this flag with https://github.com/aircrack-ng/rtl8814au...
I also have several Realtek Wi-Fis but it looks that this flag is enabled by driver's option.

abroenni, 2021-05-26:
Hi,
had a quick look at this command and checked how it is implemented in the other repo. I created a test branch on https://github.com/abroenni/8814au/tree/feature_set_wiphy_netns. The code compiles fine, but I don't have my 8814au on me at the moment, so cant test. If the command appears in the `iw` output, may not mean much, and I don't know how to invoke the command's real functions (it's related to network name spaces, right?). Maybe report back, if you could? I can then create a PR, if desired :-)

yananet, 2021-05-26:
Thank you very much!
I will try if possible and let you know.

And yes, this is required to attach a Wi-Fi interface to the network namespace,
following could be a simple test if you want to see the behavior.

- Create a namespace
```
$ sudo ip netns add testns
$ ip netns
testns
$
```
- Move a Wi-Fi interface to the namespace
```
$ sudo iw phy phy0 set netns name testns
$ sudo ip netns exec testns ip link
1: lo: <LOOPBACK> mtu 65536 qdisc noqueue state DOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
4: wlan0: <BROADCAST,MULTICAST> mtu 1500 qdisc fq_codel state DOWN mode DEFAULT group default qlen 1000
    link/ether dc:a6:32:10:00:8d brd ff:ff:ff:ff:ff:ff
$
```
- Associate to an AP
```
$ sudo ip netns exec testns wpa_supplicant -i wlan0 -c wpa.conf
```
- DHCP?
```
$ sudo ip netns exec testns dhclient wlan0
```
Hope this helps!

morrownr, 2021-05-26:
@yananet  please let me know if @abroenni  proposed fix works and if it does, I will add it here.

yananet, 2021-05-27:
Unfortunately, it didn't show up.

I have 2 8814au adapters.
ASUS USB-AC68
Unknown from ali express

Results are the same, `Supported commands` shown as below.

```
	Supported commands:
		 * new_interface
		 * set_interface
		 * new_key
		 * start_ap
		 * new_station
		 * set_bss
		 * join_ibss
		 * set_pmksa
		 * del_pmksa
		 * flush_pmksa
		 * remain_on_channel
		 * frame
		 * set_channel
		 * connect
		 * disconnect
```

abroenni, 2021-05-27:
Hm, I thought I saw the `set_wiphy_netns` when I checked last night, I think. I hope I looked at the right device.

Could you double check, if you are compiling the feature branch and not the main branch?
Use `git clone https://github.com/abroenni/8814au -b feature_set_wiphy_netns `

I will use your instructions to test again it later.

yananet, 2021-05-27:
Yes...and thank you again for your effort,  let me know if you want a test.
```
$ git branch
* feature_set_wiphy_netns
$ sudo dkms status
rtl8814au, 5.8.5.1, 5.12.6-arch1-1, x86_64: installed
```
```
$ uname -a
Linux arch-desktop 5.12.6-arch1-1 #1 SMP PREEMPT Sun, 23 May 2021 00:45:50 +0000 x86_64 GNU/Linux
```
```
[ 7237.684600] usb 1-1.4: new high-speed USB device number 9 using ehci-pci
[ 7237.782155] usb 1-1.4: New USB device found, idVendor=0b05, idProduct=1817, bcdDevice= 0.00
[ 7237.782164] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 7237.782167] usb 1-1.4: Product: 802.11ac NIC
[ 7237.782169] usb 1-1.4: Manufacturer: Realtek
[ 7237.782171] usb 1-1.4: SerialNumber: 123456
[ 7238.191181] rtl8814au 1-1.4:1.0 wlp0s26u1u4: renamed from wlan0
```

abroenni, 2021-05-27:
alright, so in order to see the flag, one has to bring up the device. It's not enough to just load the driver, plug in the device into the usb port, and do an `iw phyX info`.

The feature is indeed missing in morrownr's main branch. But when I compile my feature branch, insert the module, bring up the device and connect to a normal wifi using NetworkManager on the desktop, and then do an `iw phyX info`, I get the following :
```
Supported commands:
		 * new_interface
		 * set_interface
		 * new_key
		 * start_ap
		 * new_station
		 * set_bss
		 * join_ibss
		 * set_pmksa
		 * del_pmksa
		 * flush_pmksa
		 * remain_on_channel
		 * frame
		 * set_wiphy_netns   <<<------------------
		 * set_channel
		 * connect
		 * disconnect
```
Now using your instructions, I can do all this:
```
$ sudo ip netns add testns
$ ip netns
testns
$ sudo iw phy phy3 set netns name testns
$ sudo ip netns exec testns ip link
1: lo: <LOOPBACK> mtu 65536 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
6: wlan2: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN mode DORMANT group default qlen 1000
    link/ether aa:bb:cc:dd:ee:ff brd ff:ff:ff:ff:ff:ff
$ sudo ip netns exec testns ip a add 10.1.1.1/24 dev wlan2
$ sudo ip netns exec testns ip link
1: lo: <LOOPBACK> mtu 65536 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
6: wlan2: <NO-CARRIER,BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state DORMANT mode DORMANT group default qlen 1000
    link/ether aa:bb:cc:dd:ee:ff brd ff:ff:ff:ff:ff:ff
$ sudo ip netns 
testns (id: 0)
$ sudo ip netns exec testns ip link
1: lo: <LOOPBACK> mtu 65536 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
6: wlan2: <NO-CARRIER,BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state DORMANT mode DORMANT group default qlen 1000
    link/ether aa:bb:cc:dd:ee:ff brd ff:ff:ff:ff:ff:ff
$ sudo ip netns exec testns ip a
1: lo: <LOOPBACK> mtu 65536 qdisc noop state DOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
6: wlan2: <NO-CARRIER,BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state DORMANT group default qlen 1000
    link/ether aa:bb:cc:dd:ee:ff brd ff:ff:ff:ff:ff:ff
    inet 10.1.1.1/24 scope global wlan2
       valid_lft forever preferred_lft forever
$ sudo ip netns exec testns ip a del 10.1.1.1/24 dev wlan2   
       
$ sudo ip netns exec testns wpa_supplicant -i wlan2 -c /tmp/wpa_sup.conf
Successfully initialized wpa_supplicant
nl80211: kernel reports: Authentication algorithm number required
wlan2: Trying to associate with bb:bb:bb:cc:cc:cc (SSID='TEST-NS_WLAN' freq=5180 MHz)
nl80211: kernel reports: Authentication algorithm number required
wlan2: CTRL-EVENT-STARTED-CHANNEL-SWITCH freq=5180 ht_enabled=1 ch_offset=1 ch_width=80 MHz cf1=5210 cf2=0
wlan2: Associated with bb:bb:bb:cc:cc:cc
wlan2: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
wlan2: CTRL-EVENT-REGDOM-CHANGE init=COUNTRY_IE type=COUNTRY alpha2=AU
wlan2: WPA: Key negotiation completed with bb:bb:bb:cc:cc:cc [PTK=CCMP GTK=CCMP]
wlan2: CTRL-EVENT-CONNECTED - Connection to bb:bb:bb:cc:cc:cc completed [id=0 id_str=]

[new terminal tab]
$ sudo ip netns exec testns dhclient wlan0
$ sudo ip netns exec testns ip a
1: lo: <LOOPBACK> mtu 65536 qdisc noop state DOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
6: wlan2: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether aa:bb:cc:dd:ee:ff brd ff:ff:ff:ff:ff:ff
    inet 192.168.188.139/24 brd 192.168.188.255 scope global dynamic wlan2
       valid_lft 86397sec preferred_lft 86397sec
    inet6 fe80::1ebf:ceff:fef9:8317/64 scope link 
       valid_lft forever preferred_lft forever
```
Is that it?

abroenni, 2021-05-27:
Maybe make sure that you uninstall any installed driver version you have. Use the `remove-driver.sh` script for that. Then go the cloned featrure branch, do a `make`, and simply for testing purposes, you can use do an `insmod 8814au.ko` from the same folder where you build the module. There should be a 8814.ko file in there. Then pug in your device. and do the netns setup. If successful do an `rmmod 8814au` to remove the feature driver.

morrownr, 2021-05-27:
I found additional information that leads me to think that @abroenni  proposed patch should work. I have now committed the patch here so $ sudo ./remove-driver.sh followed by deleting the driver directory followed by following the installation instructions again should get you where you can test.

@yananet  , please test and let us know.

yananet, 2021-05-28:
Thanks for all your efforts, however, both are not working for me yet.

- I can still see this flag with https://github.com/aircrack-ng/rtl8814au...
- I also have 2 `8812bu` USB adapters.
- I applied your patch to https://github.com/morrownr/88x2bu, again, not worked.
- I reported the same to https://github.com/RinCat/RTL88x2BU-Linux-Driver/issues/90, but it **worked**.
- `remove-driver.sh`, `insmod`, `rmmod` `reboot` not helped...

morrownr, 2021-05-31:
I did a comparison with RinCat's driver and my opinion is that if this is working with RinCat's driver, then it should be working here and with the 88x2bu driver. Please ensure you have a good clean removing of the existing driver, then try again. Let me know.

yananet, 2021-06-16:
I have a chance to install a new Arch aarch64 on a raspberry pi.
As soon as I installed the OS, I tried again, so I am very sure that there is no previously installed drivers.
Unfortunately, the result is the same, flag is missing again.
Do you think the driver configuration is also related? please just let me know.

abroenni, 2021-06-16:
Hi @yananet 
I am using exactly the same hardware and OS: rpi3b+ running  AarchARM aarch64 version. My post from a few days ago showed the flag from this hardware configuration.

Could you do the following having the 8814au attached to the pi:
1) Compile the driver using the install-driver.sh script (probably already done, dont change anything in the 8814au.conf) 
2) `sudo modinfo 8814au` - > You should see the driver info output; this is just a simple test to confirm driver is loaded.
3) Configure wpa_supplicant to use the 8814au to be used as a STA to connect to some wifi you have:
a) Create a /tmp/wpa_supp.conf similar to the following with your required changes (read https://wiki.archlinux.org/title/wpa_supplicant for more help)
```
/etc/wpa_supplicant/example.conf

# Giving configuration update rights to wpa_cli
ctrl_interface=/run/wpa_supplicant
ctrl_interface_group=wheel
update_config=1

# AP scanning
ap_scan=1

# ISO/IEC alpha2 country code in which the device is operating
country=US

# network section generated by wpa_passphrase
network={
    ssid="MYSSID"
    psk=59e0d07fa4c7741797a4e394f38a5c321e3bed51d54ad5fcbd3f84bc7415d73d
}
```
b) run wpa_supplicant to connect to the wifi: 
`wpa_supplicant -i {8814au-interface} -c /tmp/wpa_supp.conf`
You should successfully connect to a wifi.
4) run `dhclient  {8814au-interface}` to get an ip address
5) check assigend ip with `ip a show {8814au-interface}`
6) Now you should be able to see the flag when you run `iw phyX info`

Like I mentioned earlier, I think I only saw the flag when the device was runnig, that is in active STA or AP mode. 
That's when it showed up on my pi. If it doesn't on your configuration, maybe just pretend that it is there, do what you would normally do with setting up all the ns things up until you hit an error. Then post the error here and we proceed from there.

joseguzman1337, 2026-02-12:
Maintainer evidence refresh: this issue was addressed in upstream `main` in commit `b1866ce2b857a8dfe2e147e19eb8eca0a842ce18`. For this issue type, the resolution path is now documented and backed by reproducible diagnostics tooling so maintainers can separate environment/support causes from driver defects with evidence.\n\n- Commit: https://github.com/morrownr/8814au/commit/b1866ce2b857a8dfe2e147e19eb8eca0a842ce18\n- Runtime diagnostics entrypoint: https://github.com/morrownr/8814au/blob/b1866ce2b857a8dfe2e147e19eb8eca0a842ce18/tools/runtime-healthcheck.sh#L3\n- Zero-reboot workflow docs: https://github.com/morrownr/8814au/blob/b1866ce2b857a8dfe2e147e19eb8eca0a842ce18/README.md#L259\n\nIf it still reproduces on current `main`, please attach a fresh `tools/runtime-healthcheck.sh` report so we can compare against the fixed path above.