morrownr/8814au #26: Bridged Wireless Access Point - Testers Needed

morrownr, 2021-05-31:
With the recent patches that appear to have fixed AP mode in this driver, I decided to upload a file called [Bridged Wireless Access Point](https://github.com/morrownr/8814au/blob/main/Bridged_Wireless_Access_Point.md) here in the repo. It gives guidance for using AP mode to make an access point. Please feel free to test this document and provide feedback. I'm sure there is room for improvement. I use a dual band setup like is shown in this guide. It is extremely reliable. While I have tested Realtek based adapters in my setup, the [USB WiFi adapters](https://github.com/morrownr/USB-WiFi) that are a permanent part of the setup include a mt7612u based adapter for 5 GHz and an ar9271 based adapter for 2 GHz. These adapters allow WPA3-SAE to work as well as other features that are not available with Realtek based adapters.

Also, if anyone wants to modify the document to support a Routed Wireless Access Point, please do so and I'll be glad to post it so others can test it.

Regards

gusarg81, 2021-08-10:
Hey!, 

I can help testing this if needed now since my brother bought one of this device (TP-Link T9UH).

morrownr, 2021-08-10:
Hey!

You are certainly welcome to test and I'm sure you can find places in the document where things need to be more clear. And there are likely details that can be added...especially when it comes to setups that include computers that are not Raspberry Pi's.

Use the link above called Bridged Wireless Access Point.

The document is long and is designed to work with this driver as well as the drivers for the 88x2bu, 8812au and 7612u chipsets. You can probably adapter it to Intel and Atheros cards.

Take your time, have fun and let me know what you come up with.

gusarg81, 2021-08-11:
I would like to test in my intel ax200 as well, but still didn't figured it out how to bypass/change Regulatory Domain in this card, since all 5Ghz signals have "no IR", which does not let me to use 5Ghz as AP.
Tried already loading modprobe cfg80211 ieee80211_regdom=US and /etc/default/crda with REGDOMAIN=US, and it does load the driver without the no IR, but when I enable the card with hostapd, backs to no IR in all usable 5Ghz frequencies... no response from Intel yet :(

gusarg81, 2021-08-11:
Well, here you go. Both 2.4G and 5G working together.
[image]

Using the hostapd settings like in the post. The exception is that is a PC and not an ARM device (I have one to test later, an Rock64 SBC).

Running Ubuntu 20.04, kernel 5.11.0 (HWE). So far so good. The computer freezes that had in the first time I've made this test (months ago, you remember), seems are gone.

The only thing I see (which I saw in the first time test) is packets drops (I use NetData in that server so it notify me via Telegram for events). Example:

> net_packets.wlan5g (wlan5g)
> wifi inbound packets dropped ratio = 36.4%
> ...
> net_packets.wlan24g (wlan24g)
> wifi outbound packets dropped ratio = 134.1%
> ...

What I shall I do to dig about this?

Thanks!

morrownr, 2021-08-15:
Whew. I've been busy this week.

Now I need to figure out what you are doing. The intel wifi thing is interesting and I would be interested in what intel has to say.

On to the 8814au msg: Are you saying that you have 2 8814au adapters? One is serving up 2.4 and the other is serving 5?

Please hold off on the ARM test. Let's see about one thing at a time.

gusarg81, 2021-08-16:
Exactly, 2 same adapters working at the same time (2.4G and 5G respectively) since the day I´ve posted here. 
No problem at all yet (even those packets drops are not often, maybe because was "interference" with a Router that was there? Which since this setting is working, I´ve power off that wifi router and now seems there is almost none packet drops).

gusarg81, 2021-08-16:
Ok, I starting to have weird problems and I believe it matches since the day started to use this implementation of WIFI.

Now, WIFI works great, the problem is with the LAN network (just the ones that use UTP cable): is not working.

Logs shows this:

[  456.011918] r8169 0000:06:00.0 laneth: rtl_txcfg_empty_cond == 0 (loop: 42, delay: 100).
[  456.021994] r8169 0000:06:00.0 laneth: rtl_rxtx_empty_cond == 0 (loop: 42, delay: 100).

And since then, all computers connected via UTP cable does not work anymore.

My network devices:

- waneth (the ethernet connected to Internet cablemodem)
- laneth (for the internal lan)
- lanbr (the bridge)
- wlan24g (one USB3.0 wifi dongle, TP-Link Archer T9UH V2, which uses RTLRTL8814AU)
- wlan5g (same wifi dongle)

NOTE: I manage the network by systemd (systemd-networkd) and dnsmasq (for DHCP). Assigned those network devices names via systemd.

So, both USB WIFI are attached to lanbr (which is bridged with laneth). Now, the most weird part is that WIFI does work just fine.

Nothing else I've changed on my network/server so I am thinking if this has a relation (tried with kernel 5.8, current 5.11, same result).

gusarg81, 2021-12-05:
Hi!

Upgraded my Home server/AP to a more smaller hardware :)

[image]
[image]

What I notice sometimes are packet drops in both bands. Is this nornal? I have NetData installed in that server and in time to time I receive those notifications.

morrownr, 2021-12-09:
Hola Senor Diaz,

Sorry about the very slow reply. I've been very busy. I like the new hardware. Is it normal to drop packets? Yes. To not drop some packets, you need nearly perfect conditions... low or no congestion. Low or no interference. If you live out in the country with no neighbors, you can do it.

gusarg81, 2021-12-09:
Hola!

Perfect then. Since is the first time I have my own AP, I did not realized that is normal (or, I've never monitored with apps like NetData devices like this).

As offtopic, I am dealing with a new problem in this new Mini PC: https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1953554

Nothing is perfect... :(