morrownr/8814au #27: Place the morrownr/8814au in the Kali Repos.

3ncrypted1, 2021-06-03:
Morrownr, Have you spoken to KimoCoder on the Aircrack-NG/rtl8814au Github? We don't have a working 8814au driver in the Kali repos. Kimocoder had forked it on the Aircrack-ng github, but injection's not working. Don't know if Kimocoder has the time to work on the Aircrack driver or is working on something concerning WPA3. As you know, 2.4/5Ghz injection is working on Morrownr/8814au driver & is stable.

morrownr, 2021-06-04:
Quote: "Have you spoken to KimoCoder on the Aircrack-NG/rtl8814au Github?"

Reply: No. KimoCoder did contact me with a message asking where I got the source for the 8821au driver. I told him my source and never heard back from him.

Quote: "As you know, 2.4/5Ghz injection is working on Morrownr/8814au driver & is stable."

Reply: What you may not know is that my tests are showing that all of the drivers here are testing good and are now operating in a stable manner for managed, master and monitor modes. The 5 Realtek drivers here cover 9 chipsets. I get a lot of help. The repo I am most proud of is actually USB-WiFi. I believe that if we, Linux users, buy adapters based on chipsets from a company that is doing USB WiFi driver development the right way, a lot of good things will happen... and most Linux users will be happier.

Quote: "Place the morrownr/8814au in the Kali Repos."

Reply: I am receptive to ideas as my workload here is not sustainable in the long run. I could use partners. Meshing the goals would be an issue to work out. My goals included making USB WiFi easier to use for all Linux users...whether it be managed, master, monitor or whatever the use case or distro may be. Good documentation, ease of use, stability and standardization are things that I think benefit the entire Linux community. Let me know if there is any interest.

abroenni, 2021-06-06:
Just FYI: I was following the discussion for the [rtl8814au-dkms-git](https://aur.archlinux.org/packages/rtl8814au-dkms-git/) package in Arch's AUR repo. On 6 June, this AUR package was switched from aircrack-ng/8814au to this repo here as its upstream. So there could be a lot more issue/request reports coming in soon. Cheers.

morrownr, 2021-06-06:
@abroenni 

Thanks for the heads up. You are hired. You job is help run this repo. Your pay is the same as mine.

Most folks that stop by do not realize that I maintain 7 usb wifi related repos and it is a lot of work:

88x2bu
8812au
8814au
8821au
8821cu
7612u
USB-WiFi

FYI: Total hits over the last 2 weeks = 8,600.

Cheers

jcmdln, 2021-06-06:
I apologize for any ignorance on my part as I've recently stumbled upon @morrownr's 8814au repo, but rather than expending effort on per-distribution dkms packages why not focus efforts on helping with getting these drivers into upstream Linux?

morrownr, 2021-06-06:
@jcmdln 

Your suggestion is welcome. You, like most Linux users, including me, would like to see usb wifi adapter support done the Linux way for Realtek chipset based adapters.

Let me address some things.

Quote: "rather than expending effort on per-distribution dkms packages"

 I established https://github.com/morrownr in an effort to help Linux users that want to use usb wifi adapters. There are 7 repos here covering a wide variety of common drivers and information regarding usb wifi adapters. I work on these repos. I do not do per-distribution packaging of any kind. I don't have the time but the source is there for anyone wishing to do whatever they want with it. So, to put it another way, I am not expending any effort whatsoever on distro specific packages.

Quote: "why not focus efforts on helping with getting these drivers into upstream Linux?"

These Realtek drivers will never be in Linux. The out-of-kernel Realtek drivers you see here are not based on Linux Wireless standards. The changes that would need to be made are beyond the capabilities of us here in the community. Those that make the decisions for what is included in the Linux kernel will not accept non-standards compliant drivers into the kernel. The standards required have been available for years but Realtek has made the choice to do their own thing. Even if I or someone else worked a miracle and got this 8814au driver into the kernel, how is it going to be supported long term? Without Realtek stepping up to support their product, it won't work.

My opinion is that the only way to fix this situation is for the Linux community as a whole to start buying usb wifi adapters that use chipsets from a company that is doing Linux driver support the right way-- Mediatek.  See the following repo and pass it on to anyone who might benefit:

https://github.com/morrownr/USB-WiFi

abroenni, 2021-06-07:
@morrownr  "You are hired. You job is help run this repo. Your pay is the same as mine." 
I can hardly confine my overwhelming emotions about this promotion :smile: 
I will do my best :+1: