morrownr/8814au #30: Connecting to 2Ghz Huawei AX3 yields 5Ghz actual connection and vice versa

nikty, 2021-06-10:
Strange problems here when using ASUS USB-AC68 to connect to Huawei AX3 router.
When connected to 2Ghz network, the speed of connection is actualy that of 5Ghz connection (tested with iperf).
Other devices (like phones) and also the same USB adapter plugged into Windows 10 host connect just fine, the result is correct speed for both frequencies.

The Linux is:
<pre>
Linux debian 4.19.0-16-amd64 #1 SMP Debian 4.19.181-1 (2021-03-19) x86_64 GNU/Linux
</pre>

Connection shows 2Ghz:
<pre>
wlxb06ebfa94f43  IEEE 802.11bgn  ESSID:"TEST_Huawei_AX3_2Ghz"  Nickname:"WIFI@RTL8814AU"
          Mode:Managed  Frequency:2.437 GHz  Access Point: B0:CC:FE:77:85:00   
          Bit Rate:400 Mb/s   Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:****-****-****-****-****-****-****-****   Security mode:open
          Power Management:off
          Link Quality=100/100  Signal level=-46 dBm  Noise level=0 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0
</pre>

But iperf is way too fast:
<pre>
$ iperf3 -c 192.168.3.113
Connecting to host 192.168.3.113, port 5201
[  5] local 192.168.3.3 port 44140 connected to 192.168.3.113 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  17.3 MBytes   145 Mbits/sec    0    901 KBytes       
[  5]   1.00-2.00   sec  23.8 MBytes   199 Mbits/sec    0   1.10 MBytes       
[  5]   2.00-3.00   sec  31.2 MBytes   262 Mbits/sec    0   1.31 MBytes       
[  5]   3.00-4.00   sec  32.5 MBytes   273 Mbits/sec    0   1.47 MBytes       
[  5]   4.00-5.00   sec  30.0 MBytes   252 Mbits/sec    0   2.05 MBytes       
[  5]   5.00-6.00   sec  31.2 MBytes   262 Mbits/sec    0   2.32 MBytes       
[  5]   6.00-7.00   sec  30.0 MBytes   252 Mbits/sec    0   2.32 MBytes       
[  5]   7.00-8.00   sec  31.2 MBytes   262 Mbits/sec    0   2.32 MBytes       
[  5]   8.00-9.00   sec  32.5 MBytes   273 Mbits/sec    0   2.32 MBytes       
[  5]   9.00-10.00  sec  31.2 MBytes   262 Mbits/sec    0   2.32 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   291 MBytes   244 Mbits/sec    0             sender
[  5]   0.00-10.01  sec   289 MBytes   242 Mbits/sec                  receiver

iperf Done.
</pre>

The driver is latest from git:
<pre>
filename:       /lib/modules/4.19.0-16-amd64/kernel/drivers/net/wireless/8814au.ko
version:        v5.8.5.1_35583.20191029
author:         Realtek Semiconductor Corp.
description:    Realtek Wireless Lan Driver
license:        GPL
srcversion:     D77CF5AF84235DF5728F810
alias:          usb:v0846p9054d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v20F4p809Bd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v20F4p809Ad*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p0106d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0E66p0026d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp8813d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v7392pA833d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v7392pA834d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v056Ep400Dd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v056Ep400Bd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0B05p1853d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0B05p1852d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0B05p1817d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p331Ad*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp8813d*dc*dsc*dp*ic*isc*ip*in*
depends:        usbcore,cfg80211
[... 87 lines cut ...]
parm:           rtw_rxgain_offset_5gh:uint
parm:           rtw_rxgain_offset_5gm:default RF Gain 5GM Offset value:0 (uint)
parm:           rtw_pll_ref_clk_sel:force pll_ref_clk_sel, 0xF:use autoload value (uint)
parm:           rtw_tx_pwr_by_rate:0:Disable, 1:Enable, 2: Depend on efuse (int)
parm:           rtw_tx_pwr_lmt_enable:0:Disable, 1:Enable, 2: Depend on efuse (int)
parm:           rtw_target_tx_pwr_2g_a:2.4G target tx power (unit:dBm) of RF path A for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_target_tx_pwr_2g_b:2.4G target tx power (unit:dBm) of RF path B for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_target_tx_pwr_2g_c:2.4G target tx power (unit:dBm) of RF path C for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_target_tx_pwr_2g_d:2.4G target tx power (unit:dBm) of RF path D for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_target_tx_pwr_5g_a:5G target tx power (unit:dBm) of RF path A for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_target_tx_pwr_5g_b:5G target tx power (unit:dBm) of RF path B for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_target_tx_pwr_5g_c:5G target tx power (unit:dBm) of RF path C for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_target_tx_pwr_5g_d:5G target tx power (unit:dBm) of RF path D for each rate section, should match the real calibrate power, -1: undefined (array of int)
parm:           rtw_tsf_update_pause_factor:num of bcn intervals to stay TSF update pause status (int)
parm:           rtw_tsf_update_restore_factor:num of bcn intervals to stay TSF update restore status (int)
parm:           rtw_phy_file_path:The path of phy parameter (charp)
parm:           rtw_load_phy_file:PHY File Bit Map (int)
parm:           rtw_decrypt_phy_file:Enable Decrypt PHY File (int)
parm:           rtw_phydm_ability:uint
parm:           rtw_halrf_ability:uint
parm:           rtw_en_napi:int
parm:           rtw_en_gro:int
parm:           rtw_iqk_fw_offload:int
parm:           rtw_ch_switch_offload:int
</pre>

morrownr, 2021-06-11:
@nikty 

It is good to meet you. This is interesting. This is the first time I have seen someone report that their network connection is too fast.

We need to confirm what is going on. Try turning the 5GHz off in the wifi router so that the 2GHZ radio is the only one on. Then do a test. Then do the same with only the 5GHz radio on. Let's see what we get.

FYI: I may be slow to respond over the next few weeks as I am busy.

nikty, 2021-06-11:
### 5GHz off

I turned off 5GHz and restarted router both from web interface and by unplugging it from the wall.
The settings are as follows.
Iperf:
<pre>
$ iperf3 -c 192.168.3.113 -t 60
Connecting to host 192.168.3.113, port 5201
[  5] local 192.168.3.3 port 55078 connected to 192.168.3.113 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   9.00-10.00  sec  30.0 MBytes   252 Mbits/sec    0   2.89 MBytes       
[  5]  19.00-20.00  sec  32.5 MBytes   273 Mbits/sec    0   3.05 MBytes       
[  5]  29.00-30.00  sec  31.2 MBytes   262 Mbits/sec    0   3.05 MBytes       
[  5]  39.00-40.00  sec  30.0 MBytes   252 Mbits/sec    0   3.05 MBytes       
[  5]  49.00-50.00  sec  33.8 MBytes   283 Mbits/sec    0   3.05 MBytes       
[  5]  59.00-60.00  sec  32.5 MBytes   273 Mbits/sec    0   3.05 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-60.00  sec  1.86 GBytes   266 Mbits/sec    0             sender
[  5]   0.00-60.05  sec  1.85 GBytes   265 Mbits/sec                  receiver

iperf Done.
</pre>
[image]
[image]

### 2GHz off

[image]
[image]

With settings as above the results I get are like this.

<pre>
$ iperf3 -c 192.168.3.113 -t 60
Connecting to host 192.168.3.113, port 5201
[  5] local 192.168.3.3 port 55164 connected to 192.168.3.113 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   9.00-10.00  sec  6.40 MBytes  53.7 Mbits/sec    0    317 KBytes       
[  5]  19.00-20.00  sec  5.00 MBytes  41.9 Mbits/sec    0   1.07 MBytes       
[  5]  29.00-30.00  sec  6.25 MBytes  52.4 Mbits/sec    0   1.07 MBytes       
[  5]  39.00-40.00  sec  6.25 MBytes  52.4 Mbits/sec    0   1.07 MBytes       
[  5]  49.00-50.00  sec  7.50 MBytes  62.9 Mbits/sec    0   1.07 MBytes       
[  5]  59.00-60.00  sec  7.50 MBytes  62.9 Mbits/sec    0   1.07 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-60.00  sec   404 MBytes  56.5 Mbits/sec   12             sender
[  5]   0.00-60.11  sec   401 MBytes  55.9 Mbits/sec                  receiver

iperf Done.
</pre>

Here Bit rate shows 400 Mb/s, but more often I get 174 Mb/s.
<pre>
$ sudo iwconfig
enp2s0    no wireless extensions.

wlxb06ebfa94f43  IEEE 802.11AC  ESSID:"TEST_Huawei_AX3_5G"  Nickname:"WIFI@RTL8814AU"
          Mode:Managed  Frequency:5.66 GHz  Access Point: B0:CC:FE:77:85:04   
          Bit Rate:400 Mb/s   Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:****-****-****-****-****-****-****-****   Security mode:open
          Power Management:off
          Link Quality=95/100  Signal level=-34 dBm  Noise level=0 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

lo        no wireless extensions.
</pre>

<pre>
$ sudo iwconfig
enp2s0    no wireless extensions.

wlxb06ebfa94f43  IEEE 802.11AC  ESSID:"TEST_Huawei_AX3_5G"  Nickname:"WIFI@RTL8814AU"
          Mode:Managed  Frequency:5.7 GHz  Access Point: B0:CC:FE:77:85:04   
          Bit Rate:174 Mb/s   Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:****-****-****-****-****-****-****-****   Security mode:open
          Power Management:off
          Link Quality=94/100  Signal level=-24 dBm  Noise level=0 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

lo        no wireless extensions.
</pre>

nikty, 2021-06-11:
I was wrong about speed of connection with Windows 10 host.

The tests above give for Win10/Driver 1030.38.0712.2019

* 2 GHz only - 5 GHz off
Bit rate: 400 Mb/s; Iperf3 - 134 Mbit/s

* 5 Ghz only - 2 GHz off
Bit rate: 400 Mb/s; Iperf3 - 101 Mbit/s

5 Ghz is certainly slower than I would expect.

morrownr, 2021-06-14:
Let's see what the following shows:

 $ grep [[:alnum:]] /sys/module/8814au/parameters/*

nikty, 2021-06-15:
$ grep [[:alnum:]] /sys/module/8814au/parameters/*
/sys/module/8814au/parameters/if2name:wlan%d
/sys/module/8814au/parameters/ifname:wlan%d
/sys/module/8814au/parameters/rtw_adaptivity_en:0
/sys/module/8814au/parameters/rtw_adaptivity_mode:0
/sys/module/8814au/parameters/rtw_adaptivity_th_edcca_hl_diff:0
/sys/module/8814au/parameters/rtw_adaptivity_th_l2h_ini:0
/sys/module/8814au/parameters/rtw_ampdu_enable:1
/sys/module/8814au/parameters/rtw_amplifier_type_2g:0
/sys/module/8814au/parameters/rtw_amplifier_type_5g:0
/sys/module/8814au/parameters/rtw_antdiv_cfg:2
/sys/module/8814au/parameters/rtw_antdiv_type:0
/sys/module/8814au/parameters/rtw_beamform_cap:10
/sys/module/8814au/parameters/rtw_busy_thresh:40
/sys/module/8814au/parameters/rtw_bw_mode:33
/sys/module/8814au/parameters/rtw_channel:1
/sys/module/8814au/parameters/rtw_channel_plan:255
/sys/module/8814au/parameters/rtw_check_hw_status:0
/sys/module/8814au/parameters/rtw_chip_version:0
/sys/module/8814au/parameters/rtw_ch_switch_offload:0
/sys/module/8814au/parameters/rtw_decrypt_phy_file:0
/sys/module/8814au/parameters/rtw_drv_ant_band_switch:1
/sys/module/8814au/parameters/rtw_drv_log_level:0
/sys/module/8814au/parameters/rtw_dynamic_agg_enable:1
/sys/module/8814au/parameters/rtw_en_gro:1
/sys/module/8814au/parameters/rtw_en_napi:1
/sys/module/8814au/parameters/rtw_enusbss:0
/sys/module/8814au/parameters/rtw_FileMaskEfuse:0
/sys/module/8814au/parameters/rtw_full_ch_in_p2p_handshake:0
/sys/module/8814au/parameters/rtw_GLNA_type:0
/sys/module/8814au/parameters/rtw_halrf_ability:4294967295
/sys/module/8814au/parameters/rtw_hiq_filter:1
/sys/module/8814au/parameters/rtw_ht_enable:1
/sys/module/8814au/parameters/rtw_hwpdn_mode:2
/sys/module/8814au/parameters/rtw_hwpwrp_detect:0
/sys/module/8814au/parameters/rtw_hw_wps_pbc:1
/sys/module/8814au/parameters/rtw_initmac:(null)
/sys/module/8814au/parameters/rtw_ips_mode:1
/sys/module/8814au/parameters/rtw_iqk_fw_offload:0
/sys/module/8814au/parameters/rtw_lbkmode:0
/sys/module/8814au/parameters/rtw_led_ctrl:1
/sys/module/8814au/parameters/rtw_load_phy_file:68
/sys/module/8814au/parameters/rtw_low_power:0
/sys/module/8814au/parameters/rtw_lowrate_two_xmit:1
/sys/module/8814au/parameters/rtw_lps_chk_by_tp:1
/sys/module/8814au/parameters/rtw_lps_level:0
/sys/module/8814au/parameters/rtw_max_bss_cnt:0
/sys/module/8814au/parameters/rtw_max_roaming_times:2
/sys/module/8814au/parameters/rtw_mc2u_disable:0
/sys/module/8814au/parameters/rtw_mp_mode:0
/sys/module/8814au/parameters/rtw_network_mode:0
/sys/module/8814au/parameters/rtw_notch_filter:0
/sys/module/8814au/parameters/rtw_OffEfuseMask:0
/sys/module/8814au/parameters/rtw_pci_dynamic_aspm_linkctrl:0
/sys/module/8814au/parameters/rtw_phydm_ability:4294967295
/sys/module/8814au/parameters/rtw_phy_file_path:/lib/firmware/
/sys/module/8814au/parameters/rtw_pll_ref_clk_sel:15
/sys/module/8814au/parameters/rtw_power_mgnt:2
/sys/module/8814au/parameters/rtw_powertracking_type:64
/sys/module/8814au/parameters/rtw_pwrtrim_enable:2
/sys/module/8814au/parameters/rtw_qos_opt_enable:0
/sys/module/8814au/parameters/rtw_RFE_type:64
/sys/module/8814au/parameters/rtw_rfintfs:2
/sys/module/8814au/parameters/rtw_rf_path:8
/sys/module/8814au/parameters/rtw_rx_ampdu_amsdu:0
/sys/module/8814au/parameters/rtw_rxgain_offset_2g:0
/sys/module/8814au/parameters/rtw_rxgain_offset_5gh:0
/sys/module/8814au/parameters/rtw_rxgain_offset_5gl:0
/sys/module/8814au/parameters/rtw_rxgain_offset_5gm:0
/sys/module/8814au/parameters/rtw_rx_nss:0
/sys/module/8814au/parameters/rtw_rx_stbc:1
/sys/module/8814au/parameters/rtw_single_ant_path:0
/sys/module/8814au/parameters/rtw_smart_ps:2
/sys/module/8814au/parameters/rtw_switch_usb_mode:0
/sys/module/8814au/parameters/rtw_tsf_update_pause_factor:200
/sys/module/8814au/parameters/rtw_tsf_update_restore_factor:5
/sys/module/8814au/parameters/rtw_tx_aclt_flags:0
/sys/module/8814au/parameters/rtw_tx_ampdu_amsdu:2
/sys/module/8814au/parameters/rtw_TxBBSwing_2G:255
/sys/module/8814au/parameters/rtw_TxBBSwing_5G:255
/sys/module/8814au/parameters/rtw_tx_bw_mode:33
/sys/module/8814au/parameters/rtw_tx_nss:0
/sys/module/8814au/parameters/rtw_tx_pwr_by_rate:0
/sys/module/8814au/parameters/rtw_tx_pwr_lmt_enable:0
/sys/module/8814au/parameters/rtw_usb_rxagg_mode:2
/sys/module/8814au/parameters/rtw_vcs_type:1
/sys/module/8814au/parameters/rtw_vht_enable:1
/sys/module/8814au/parameters/rtw_vht_rx_mcs_map:43690
/sys/module/8814au/parameters/rtw_vrtl_carrier_sense:2
/sys/module/8814au/parameters/rtw_wifi_spec:0
/sys/module/8814au/parameters/rtw_wireless_mode:95
/sys/module/8814au/parameters/rtw_wmm_enable:1
/sys/module/8814au/parameters/rtw_wowlan_sta_mix_mode:0

morrownr, 2021-06-15:
Okay, I see something that would explain a lot of what you are seeing.

rtw_switch_usb_mode:0

This setting needs to be 1. In the driver directory run $ sudo ./edit-options.sh

You were in usb 2 mode. These Realtek drivers are like a standard transmission. You can pretty much max out 2.4 GHz wifi with usb 2 but 5 GHz wifi gets limited seriously.

Let me know what you see now.

nikty, 2021-06-16:
### rtw_switch_usb_mode=0

$ grep \\w /sys/module/8814au/parameters/* | grep rtw_switch_usb_mode     
/sys/module/8814au/parameters/rtw_switch_usb_mode:0

<pre>
$ iperf3 -c 192.168.3.113                                                
Connecting to host 192.168.3.113, port 5201
[  5] local 192.168.3.6 port 45378 connected to 192.168.3.113 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  12.1 MBytes   102 Mbits/sec    0    481 KBytes       
[  5]   1.00-2.00   sec  11.9 MBytes  99.7 Mbits/sec    0    669 KBytes       
[  5]   2.00-3.00   sec  11.2 MBytes  94.4 Mbits/sec    0    752 KBytes       
[  5]   3.00-4.00   sec  11.2 MBytes  94.4 Mbits/sec    0    851 KBytes       
[  5]   4.00-5.00   sec  12.5 MBytes   105 Mbits/sec    0    851 KBytes       
[  5]   5.00-6.00   sec  11.2 MBytes  94.4 Mbits/sec    0    902 KBytes       
[  5]   6.00-7.00   sec  11.2 MBytes  94.4 Mbits/sec    0    902 KBytes       
[  5]   7.00-8.00   sec  12.5 MBytes   105 Mbits/sec    0   1.05 MBytes       
[  5]   8.00-9.00   sec  12.5 MBytes   105 Mbits/sec    0   1.22 MBytes       
[  5]   9.00-10.00  sec  13.8 MBytes   115 Mbits/sec    0   1.23 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   120 MBytes   101 Mbits/sec    0             sender
[  5]   0.00-10.04  sec   117 MBytes  97.4 Mbits/sec                  receiver

iperf Done.
</pre>

### rtw_switch_usb_mode=1

$ sudo modprobe -r 8814au    
$ sudo modprobe 8814au rtw_switch_usb_mode=1

<pre>
$ iperf3 -c 192.168.3.113 
Connecting to host 192.168.3.113, port 5201
[  5] local 192.168.3.6 port 45382 connected to 192.168.3.113 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  12.0 MBytes   101 Mbits/sec    0    411 KBytes       
[  5]   1.00-2.00   sec  11.7 MBytes  98.5 Mbits/sec    0    505 KBytes       
[  5]   2.00-3.00   sec  13.0 MBytes   109 Mbits/sec    0    536 KBytes       
[  5]   3.00-4.00   sec  10.8 MBytes  90.2 Mbits/sec    0    536 KBytes       
[  5]   4.00-5.00   sec  12.4 MBytes   104 Mbits/sec    0    608 KBytes       
[  5]   5.00-6.00   sec  12.5 MBytes   105 Mbits/sec    0    906 KBytes       
[  5]   6.00-7.00   sec  12.5 MBytes   105 Mbits/sec    0   1.04 MBytes       
[  5]   7.00-8.00   sec  13.8 MBytes   115 Mbits/sec    0   1.11 MBytes       
[  5]   8.00-9.00   sec  12.5 MBytes   105 Mbits/sec    0   1.17 MBytes       
[  5]   9.00-10.00  sec  13.8 MBytes   115 Mbits/sec    0   1.23 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   125 MBytes   105 Mbits/sec    0             sender
[  5]   0.00-10.05  sec   122 MBytes   102 Mbits/sec                  receiver

iperf Done.
</pre>

### rtw_switch_usb_mode=2

$ sudo modprobe -r 8814au    
$ sudo modprobe 8814au rtw_switch_usb_mode=2
$ grep \\w /sys/module/8814au/parameters/* | grep rtw_switch_usb_mode
/sys/module/8814au/parameters/rtw_switch_usb_mode:2

<pre>
$ iperf3 -c 192.168.3.113                   
Connecting to host 192.168.3.113, port 5201
[  5] local 192.168.3.6 port 45436 connected to 192.168.3.113 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  10.6 MBytes  89.2 Mbits/sec    0    399 KBytes       
[  5]   1.00-2.00   sec  12.2 MBytes   102 Mbits/sec    0    563 KBytes       
[  5]   2.00-3.00   sec  9.57 MBytes  80.3 Mbits/sec    0    592 KBytes       
[  5]   3.00-4.00   sec  12.4 MBytes   104 Mbits/sec    0    660 KBytes       
[  5]   4.00-5.00   sec  11.2 MBytes  94.4 Mbits/sec    0    660 KBytes       
[  5]   5.00-6.00   sec  11.2 MBytes  94.4 Mbits/sec    0    836 KBytes       
[  5]   6.00-7.00   sec  11.2 MBytes  94.4 Mbits/sec    0    998 KBytes       
[  5]   7.00-8.00   sec  12.5 MBytes   105 Mbits/sec    0   1.03 MBytes       
[  5]   8.00-9.00   sec  11.2 MBytes  94.4 Mbits/sec    0   1.09 MBytes       
[  5]   9.00-10.00  sec  12.5 MBytes   105 Mbits/sec    0   1.09 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   115 MBytes  96.2 Mbits/sec    0             sender
[  5]   0.00-10.04  sec   112 MBytes  93.5 Mbits/sec                  receiver

iperf Done.
</pre>

nikty, 2021-06-16:
I also have adapter with 8812au chipset, tests like above give 118 / 142 / 163 for modes 0/1/2.

nikty, 2021-06-16:
The results seemed random, I moved away from the router (~3-4 m) and did this test:
* no .conf file in /etc/modprobe.d
* for every combination of rtw_switch_usb_mode=0,1,2 rtw_beamform_cap=0,10 iperf runs twice for 1 min

Repeated for 8812au also (see below). 
8814au doesn't even hit 100Mbit/s.

### 8814au

<pre>
rmmod 8814au
insmod /lib/modules/4.19.0-16-amd64/kernel/drivers/net/wireless/8814au.ko rtw_switch_usb_mode=0 rtw_beamform_cap=0
/sys/module/8814au/parameters/rtw_beamform_cap:0
/sys/module/8814au/parameters/rtw_switch_usb_mode:0
Waiting for connection
Connected
Connecting to host 192.168.3.113, port 5201
[  5] local 192.168.3.3 port 60122 connected to 192.168.3.113 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-5.00   sec  35.5 MBytes  59.6 Mbits/sec   11    226 KBytes       
[  5]   5.00-10.00  sec  36.8 MBytes  61.7 Mbits/sec    0    325 KBytes       
[  5]  10.00-15.00  sec  30.0 MBytes  50.4 Mbits/sec    0    897 KBytes       
[  5]  15.00-20.00  sec  26.2 MBytes  44.0 Mbits/sec    0    897 KBytes       
[  5]  20.00-25.00  sec  32.5 MBytes  54.5 Mbits/sec    0    897 KBytes       
[  5]  25.00-30.00  sec  23.8 MBytes  39.8 Mbits/sec    0    897 KBytes       
[  5]  30.00-35.00  sec  28.8 MBytes  48.2 Mbits/sec    0    897 KBytes       
[  5]  35.00-40.00  sec  31.2 MBytes  52.4 Mbits/sec    0    897 KBytes       
[  5]  40.00-45.00  sec  28.8 MBytes  48.2 Mbits/sec    0    897 KBytes       
[  5]  45.00-50.00  sec  28.8 MBytes  48.2 Mbits/sec    0    897 KBytes       
[  5]  50.00-55.00  sec  30.0 MBytes  50.3 Mbits/sec    0    897 KBytes       
[  5]  55.00-60.00  sec  27.5 MBytes  46.1 Mbits/sec    0    897 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-60.00  sec   360 MBytes  50.3 Mbits/sec   11             sender
[  5]   0.00-60.10  sec   357 MBytes  49.9 Mbits/sec                  receiver

iperf Done.
Connecting to host 192.168.3.113, port 5201
[  5] local 192.168.3.3 port 60128 connected to 192.168.3.113 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-5.00   sec  29.1 MBytes  48.8 Mbits/sec    0    600 KBytes       
[  5]   5.00-10.00  sec  29.9 MBytes  50.2 Mbits/sec    0    704 KBytes       
[  5]  10.00-15.00  sec  31.2 MBytes  52.4 Mbits/sec    0    856 KBytes       
[  5]  15.00-20.00  sec  32.5 MBytes  54.5 Mbits/sec    0   1.28 MBytes       
[  5]  20.00-25.00  sec  27.5 MBytes  46.1 Mbits/sec    0   1.28 MBytes       
[  5]  25.00-30.00  sec  27.5 MBytes  46.1 Mbits/sec    0   1.28 MBytes       
[  5]  30.00-35.00  sec  31.2 MBytes  52.4 Mbits/sec    0   1.28 MBytes       
[  5]  35.00-40.00  sec  31.2 MBytes  52.4 Mbits/sec    0   1.28 MBytes       
[  5]  40.00-45.00  sec  30.0 MBytes  50.3 Mbits/sec    0   1.28 MBytes       
[  5]  45.00-50.00  sec  31.2 MBytes  52.4 Mbits/sec    0   1.28 MBytes       
[  5]  50.00-55.00  sec  31.2 MBytes  52.4 Mbits/sec    0   1.28 MBytes       
[  5]  55.00-60.00  sec  33.8 MBytes  56.6 Mbits/sec    0   1.28 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-60.00  sec   366 MBytes  51.2 Mbits/sec    0             sender
[  5]   0.00-60.06  sec   363 MBytes  50.7 Mbits/sec                  receiver

iperf Done.
rmmod 8814au
insmod /lib/modules/4.19.0-16-amd64/kernel/drivers/net/wireless/8814au.ko rtw_switch_usb_mode=0 rtw_beamform_cap=10
/sys/module/8814au/parameters/rtw_beamform_cap:10
/sys/module/8814au/parameters/rtw_switch_usb_mode:0
Waiting for connection
Connected
Connecting to host 192.168.3.113, port 5201
[  5] local 192.168.3.3 port 60140 connected to 192.168.3.113 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-5.00   sec  32.7 MBytes  54.8 Mbits/sec   10    219 KBytes       
[  5]   5.00-10.00  sec  33.2 MBytes  55.7 Mbits/sec    0    313 KBytes       
[  5]  10.00-15.00  sec  35.4 MBytes  59.5 Mbits/sec    0    690 KBytes       
[... 497 lines cut ...]
[  5]   0.00-60.05  sec  1.09 GBytes   156 Mbits/sec                  receiver

iperf Done.
Connecting to host 192.168.3.113, port 5201
[  5] local 192.168.3.13 port 53318 connected to 192.168.3.113 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-5.00   sec   101 MBytes   170 Mbits/sec    0   1.46 MBytes       
[  5]   5.00-10.00  sec   101 MBytes   170 Mbits/sec    0   1.70 MBytes       
[  5]  10.00-15.00  sec   100 MBytes   168 Mbits/sec    0   1.79 MBytes       
[  5]  15.00-20.00  sec   100 MBytes   168 Mbits/sec    0   2.11 MBytes       
[  5]  20.00-25.00  sec   101 MBytes   170 Mbits/sec    0   2.11 MBytes       
[  5]  25.00-30.00  sec  98.8 MBytes   166 Mbits/sec    0   2.11 MBytes       
[  5]  30.00-35.00  sec  98.8 MBytes   166 Mbits/sec    0   2.11 MBytes       
[  5]  35.00-40.00  sec  98.8 MBytes   166 Mbits/sec    0   2.11 MBytes       
[  5]  40.00-45.00  sec  96.2 MBytes   161 Mbits/sec    0   2.11 MBytes       
[  5]  45.00-50.00  sec  98.8 MBytes   166 Mbits/sec    0   2.11 MBytes       
[  5]  50.00-55.00  sec   100 MBytes   168 Mbits/sec    0   2.11 MBytes       
[  5]  55.00-60.00  sec  93.8 MBytes   157 Mbits/sec    0   2.11 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-60.00  sec  1.16 GBytes   166 Mbits/sec    0             sender
[  5]   0.00-60.05  sec  1.16 GBytes   166 Mbits/sec                  receiver

iperf Done.
</pre>

nikty, 2021-06-17:
@morrownr Many thanks for your support and comments!

I'm going to stop investigating this for now, but if anything interesting comes up, I'll update the ticket.

abroenni, 2021-06-18:
Hi @nikty 
I been following this conversation and just want to add the following. I was gonna say that it could be that the antennas of the ASUS USB-AC68 may not be optimised for the 5Ghz spectrum. Yes, far fetched - I am aware. Unfortunately you cannot replace them, as it seems when looking at the device. So I googled for other iperf3 tests using this device and I found this [asus-usb-ac68-wifi-test](asus-usb-ac68-wifi-test). Unfortunately only windows driver tests, but they reflect hardware capabilities nonetheless. 
You are using your device in STA mode (client connecting to an access point) plugged into a usb2 port. In you testing, as far as I can tell, you have been running your iperf3 in client mode on on the computer where the AC68 is plugged into. 
When running iperf3 the way you did, you have been testing the uplink (data flowing AC68 --> iperf3sever). This is TX.
`iperf3 -c 192.168.3.113 -t 60 -R` would test the downlink (AC68 <--- iperf3server). This is RX. The device in USB2 mode uses more receive channels/antennas (Not quite sure.). Downlink speeds should be higher.
From iper3 man page:
```
Normally,  the test data is sent from the client to the server, and measures the upload speed of the client.
Measuring the download speed from the server can be done by specifying the -R flag on the client.
This causes data to be sent from the  server  to the client.

              iperf3 -c iperf3.example.com -p 5202 -R

```

So maybe try testing in both directions.

I ran the 4 tests on a 1 core cpu laptop using a 8814au plugged into USB2 simulating a 1G file upload and download on 2.4Gz and 5Ghz with default 8814au.conf:
```
============
5Ghz======
============
[DOWNLINK][8814au receiving data]
./iperf3 -c 192.168.188.87 -R -n 1G
Connecting to host 192.168.188.87, port 5201
Reverse mode, remote host 192.168.188.87 is sending
[  5] local 192.168.188.139 port 45314 connected to 192.168.188.87 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  18.0 MBytes   151 Mbits/sec                  
[  5]   1.00-2.00   sec  24.6 MBytes   206 Mbits/sec                  
[  5]   2.00-3.00   sec  23.8 MBytes   200 Mbits/sec                  
...
...             
[  5]  33.00-34.00  sec  27.3 MBytes   229 Mbits/sec                  
[  5]  34.00-35.00  sec  24.5 MBytes   206 Mbits/sec                  
[  5]  35.00-36.00  sec  27.3 MBytes   229 Mbits/sec                  
[  5]  36.00-36.33  sec  8.15 MBytes   209 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-36.38  sec  1.00 GBytes   237 Mbits/sec   10             sender
[  5]   0.00-36.33  sec  1.00 GBytes   236 Mbits/sec                  receiver

iperf Done.

[UPLINK] [8814au sending data]
./iperf3 -c 192.168.188.87 -n 1G
Connecting to host 192.168.188.87, port 5201
[  5] local 192.168.188.139 port 45318 connected to 192.168.188.87 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  12.7 MBytes   106 Mbits/sec    0    423 KBytes       
[  5]   1.00-2.00   sec  10.9 MBytes  91.2 Mbits/sec    0    672 KBytes       
[  5]   2.00-3.00   sec  16.0 MBytes   134 Mbits/sec    0    790 KBytes       
[  5]   3.00-4.00   sec  9.82 MBytes  82.4 Mbits/sec    0    790 KBytes       
[  5]   4.00-5.00   sec  15.7 MBytes   132 Mbits/sec    0    830 KBytes       
[  5]   5.00-6.00   sec  9.38 MBytes  78.7 Mbits/sec    0    830 KBytes       
[  5]   6.00-7.00   sec  25.2 MBytes   212 Mbits/sec    0    888 KBytes       
[  5]   7.00-8.00   sec  23.2 MBytes   194 Mbits/sec    0    888 KBytes       
[  5]   8.00-9.00   sec  11.2 MBytes  94.4 Mbits/sec    0    888 KBytes       
[  5]   9.00-10.00  sec  22.6 MBytes   189 Mbits/sec    0    888 KBytes       
[  5]  10.00-11.00  sec  18.3 MBytes   153 Mbits/sec    0    888 KBytes       
[  5]  11.00-12.00  sec  21.1 MBytes   177 Mbits/sec    0    888 KBytes       
[  5]  12.00-13.00  sec  26.0 MBytes   218 Mbits/sec    0    888 KBytes       
[  5]  13.00-14.00  sec  20.8 MBytes   174 Mbits/sec    0    888 KBytes       
[  5]  14.00-15.00  sec  5.72 MBytes  48.0 Mbits/sec    0    888 KBytes       
[  5]  15.00-16.00  sec  0.00 Bytes  0.00 bits/sec    0    888 KBytes       
[  5]  16.00-17.00  sec   954 KBytes  7.82 Mbits/sec    0    888 KBytes       
[  5]  17.00-18.00  sec  0.00 Bytes  0.00 bits/sec    0    888 KBytes       
[  5]  18.00-19.00  sec  1.86 MBytes  15.7 Mbits/sec    0    888 KBytes       
[  5]  19.00-20.00  sec  18.4 MBytes   155 Mbits/sec    0   1.30 MBytes       
....
[... 56 lines cut ...]
[  5]   0.00-1.00   sec  16.5 MBytes   138 Mbits/sec    0    646 KBytes       
[  5]   1.00-2.00   sec  15.5 MBytes   130 Mbits/sec    0    682 KBytes       
[  5]   2.00-3.00   sec  15.5 MBytes   130 Mbits/sec    0    755 KBytes       
[  5]   3.00-4.00   sec  15.4 MBytes   129 Mbits/sec    0    755 KBytes       
[  5]   4.00-5.00   sec  15.3 MBytes   128 Mbits/sec    0    755 KBytes       
[  5]   5.00-6.00   sec  15.9 MBytes   133 Mbits/sec    0    795 KBytes       
[  5]   6.00-7.00   sec  15.8 MBytes   133 Mbits/sec    0    795 KBytes       
...
[  5]  60.00-61.00  sec  16.2 MBytes   136 Mbits/sec    0   2.00 MBytes       
[  5]  61.00-62.00  sec  15.0 MBytes   126 Mbits/sec    0   2.00 MBytes       
[  5]  62.00-63.00  sec  16.2 MBytes   136 Mbits/sec    0   2.00 MBytes       
[... cut ...]
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-65.30  sec  1.00 GBytes   132 Mbits/sec    0             sender
[  5]   0.00-65.35  sec  1021 MBytes   131 Mbits/sec                  receiver

iperf Done.

```
In 5Gz, the downlink maxes out the USB2.
I also noticed on 5Ghz, the data sending (TRx) of the my device is a lot more flackier/unstable on lager files, but 2.ghz seems to fine. But that could be my antennas (I can change them).
hope this helps.

nikty, 2021-06-18:
Hi @abroenni

Your comment is quite insightful!

> So maybe try testing in both directions. Or is that what you did, because you sent each iperf block twice?

No, I only did TX test, because its instability and speed initially triggered me. RX does indeed has higher speeds.

closed 2022-06-13 by morrownr