morrownr/8814au #31: install driver problem on raspberry with kali + awus1900

emmgr23, 2021-06-28:
hi, im a newbie in this domain, but im trying my best. i dont know how, but i succeed to install the driver 3 month ago for the awus1900. it's working fine in usb 3 activated, but i wanted to install this update to have the ap working.
and to be honest im block to step 3, i got an error :

root@kali:~# sudo apt-get install -y linux-headers-$(uname -r) build-essential dkms git libelf-dev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package linux-headers-5.4.83-Re4son-v7l
E: Couldn't find any package by glob 'linux-headers-5.4.83-Re4son-v7l'
E: Couldn't find any package by regex 'linux-headers-5.4.83-Re4son-v7l'

somebody can help me please :)

morrownr, 2021-06-28:
My suggestion, if you want something that works, is to use the 32 bit Raspberry Pi OS. If you really want to use the 64 bit ARM version of Kali, recommend you seek help in the Kali forums as you are more likely to find the help needed there for now. When I am able, which could be a few more weeks, I can try to take a look.

emmgr23, 2021-06-29:
im using the 32bit of kali , 
my problem is im getting internet with awus1900 usb adaptor on my rpi , when i do a debit test directly on the rpi, im getting around 100mb.

i created a hotspot with the wifi 5ghz of the raspberry , and when i connect on one of my device to this hotspot, and do a speed test, i barely got 10 to 20mb.
i tried to search everywhere a solution but couldn't find anything, and i dont understand why ?!
[image]

morrownr, 2021-06-29:
A guide that may be of help is right here in this repo:

https://github.com/morrownr/8814au/blob/main/Bridged_Wireless_Access_Point.md

I simply do not have the time to be of much help right now but if you are in no hurry, I will have time soon.

emmgr23, 2021-06-29:
thanks you very much , no im not in hurry i can wait, im searching for at least 2 months , and i'll keep searching 🙏🏽

morrownr, 2021-08-15:
I should have a little extra time at some point this coming week. What is your status?

closed 2022-06-13 by morrownr