morrownr/8814au #35: Device won't start

gusarg81, 2021-08-10:
Hola! Remember me?:P

So, my brother has this devil device, but using as client (not AP). When I've installed the driver, seems the device can't start. Here is the log:

> Aug 10 08:44:53 MaceWindu kernel: [27434.977590] RTW: cfg80211_rtw_scan(wlx7cc2c60e6976)
> Aug 10 08:44:53 MaceWindu kernel: [27434.977594] RTW: _rtw_pwr_wakeup call ips_leave....
> Aug 10 08:44:53 MaceWindu kernel: [27434.977595] RTW: ==>ips_leave cnts:439
> Aug 10 08:44:53 MaceWindu kernel: [27434.977596] RTW: ===>  rtw_ips_pwr_up..............
> Aug 10 08:44:53 MaceWindu kernel: [27434.977600] RTW: ===> ips_netdrv_open.........
> Aug 10 08:44:53 MaceWindu kernel: [27434.977695] RTW:  power-on :REG_SYS_CLKR 0x09=0xb0. REG_CR 0x100=0xea.
> Aug 10 08:44:53 MaceWindu kernel: [27434.977696] RTW:  MAC has not been powered on yet.
> Aug 10 08:44:53 MaceWindu kernel: [27434.985569] RTW: ===>_InitQueueReservedPage_8814AUsb()
> Aug 10 08:44:53 MaceWindu kernel: [27434.986009] RTW: <===_InitQueueReservedPage_8814AUsb()
> Aug 10 08:44:53 MaceWindu kernel: [27434.986170] RTW: FirmwareDownload8814A fw source from Header
> Aug 10 08:44:53 MaceWindu kernel: [27434.986171] RTW: FirmwareDownload8814A fw:NIC, size: 68320
> Aug 10 08:44:53 MaceWindu kernel: [27434.986172] RTW: FirmwareDownload8814A: fw_ver=33 fw_subver=6 sig=0x8814
> Aug 10 08:44:53 MaceWindu kernel: [27434.986690] RTW: HalROMDownloadFWRSVDPage8814A(): There is an Adapter is sending beacon.
> Aug 10 08:44:53 MaceWindu kernel: [27434.986811] RTW: dmem_pkt_size = 5792, iram_pkt_size = 62464,FWHeaderSize = 64, Len = 68320!!
> Aug 10 08:44:53 MaceWindu kernel: [27434.986811] RTW: MaxRsvdPageBufSize 1488,  Total len 68320
> Aug 10 08:44:53 MaceWindu kernel: [27434.988457] RTW: Check sum OK
> Aug 10 08:44:53 MaceWindu kernel: [27434.988497] RTW: dmem check sum tmp 33
> Aug 10 08:44:53 MaceWindu kernel: [27435.006392] RTW: Check sum OK
> Aug 10 08:44:53 MaceWindu kernel: [27435.006432] RTW: imem check sum tmp 105
> Aug 10 08:44:53 MaceWindu kernel: [27435.056585] RTW: FWDL success. write_fw:0, 68ms
> Aug 10 08:44:53 MaceWindu kernel: [27435.056613] RTW: rtl8814au_hal_init: Download Firmware Success!!
> Aug 10 08:44:53 MaceWindu kernel: [27435.374970] RTW: ==>PHY_SwitchWirelessBand8814() 2.4G
> Aug 10 08:44:53 MaceWindu kernel: [27435.376292] RTW: <==PHY_SwitchWirelessBand8814():Switch Band OK.
> Aug 10 08:44:53 MaceWindu kernel: [27435.388878] RTW: txpath=0x3, rxpath=0xf
> Aug 10 08:44:53 MaceWindu kernel: [27435.388880] RTW: txpath_1ss:0x1, num:1
> Aug 10 08:44:53 MaceWindu kernel: [27435.388881] RTW: txpath_2ss:0x3, num:2
> Aug 10 08:44:53 MaceWindu kernel: [27435.392814] RTW: rtl8814au_hal_init in 412ms
> Aug 10 08:44:53 MaceWindu kernel: [27435.392819] RTW: rtw_hal_set_macaddr_port wlx7cc2c60e6976- hw port(0) mac_addr =7c:c2:c6:0e:69:76
> Aug 10 08:44:53 MaceWindu kernel: [27435.393295] RTW: rtw_hal_get_macaddr_port wlx7cc2c60e6976- hw port(0) mac_addr =7c:c2:c6:0e:69:76
> Aug 10 08:44:53 MaceWindu kernel: [27435.393297] RTW: wlx7cc2c60e6976- hw port(0) mac_addr =7c:c2:c6:0e:69:76
> Aug 10 08:44:53 MaceWindu kernel: [27435.393415] RTW: [HW_VAR_ENABLE_RX_BAR] 0x6A2=0x520
> Aug 10 08:44:53 MaceWindu kernel: [27435.404919] RTW: rtw_hal_set_macaddr_port wlx7cc2c60e6976- hw port(0) mac_addr =7c:c2:c6:0e:69:76
> Aug 10 08:44:53 MaceWindu kernel: [27435.405277] RTW: hw_var_set_opmode()-3339 mode = 2
> Aug 10 08:44:53 MaceWindu kernel: [27435.405524] RTW: Registry kfree default force disable.
> Aug 10 08:44:53 MaceWindu kernel: [27435.405548] RTW: <===  rtw_ips_pwr_up.............. in 428ms
> Aug 10 08:44:53 MaceWindu kernel: [27435.405549] RTW: nolinked power save leave
> Aug 10 08:44:53 MaceWindu kernel: [27435.405577] RTW: ==> ips_leave.....LED(0x0c228282)...
> Aug 10 08:44:53 MaceWindu kernel: [27435.406116] RTW: SetHwReg8814A:(HW_VAR_CHECK_TXBUF)TXBUF Empty(1) in 0 ms
> Aug 10 08:44:53 MaceWindu kernel: [27435.406242] RTW: wlx7cc2c60e6976 sleep m0=0x00000002, ori reg_0x4d4=0x00000000
> Aug 10 08:44:54 MaceWindu kernel: [27436.593589] RTW: ==>PHY_SwitchWirelessBand8814() 5G
> Aug 10 08:44:54 MaceWindu kernel: [27436.595614] RTW: <==PHY_SwitchWirelessBand8814():Switch Band OK.
> Aug 10 08:44:57 MaceWindu kernel: [27438.865736] RTW: ==>PHY_SwitchWirelessBand8814() 2.4G
> Aug 10 08:44:57 MaceWindu kernel: [27438.867990] RTW: <==PHY_SwitchWirelessBand8814():Switch Band OK.
> Aug 10 08:44:57 MaceWindu kernel: [27438.888828] RTW: wlx7cc2c60e6976 wakeup m0=0x00000002, ori reg_0x4d4=0x00000002
> Aug 10 08:44:57 MaceWindu kernel: [27438.888953] RTW: survey done event(44) band:0 for wlx7cc2c60e6976
> Aug 10 08:44:57 MaceWindu kernel: [27438.889019] RTW: rtw_indicate_scan_done(wlx7cc2c60e6976)
> Aug 10 08:44:57 MaceWindu kernel: [27438.899365] RTW: ==>rtw_ps_processor .fw_state(8)
> Aug 10 08:44:57 MaceWindu kernel: [27438.899367] RTW: ==>ips_enter cnts:440
> Aug 10 08:44:57 MaceWindu kernel: [27438.899367] RTW: nolinked power save enter
> Aug 10 08:44:57 MaceWindu kernel: [27438.899368] RTW: ===> rtw_ips_pwr_down...................
> Aug 10 08:44:57 MaceWindu kernel: [27438.899369] RTW: ====> rtw_ips_dev_unload...
> Aug 10 08:44:57 MaceWindu kernel: [27438.909875] RTW: usb_read_port_cancel
> Aug 10 08:44:57 MaceWindu kernel: [27438.909951] RTW: usb_read_port_complete() RX Warning! bDriverStopped(False) OR bSurpriseRemoved(False)
> Aug 10 08:44:57 MaceWindu kernel: [27438.910067] RTW: usb_read_port_complete() RX Warning! bDriverStopped(False) OR bSurpriseRemoved(False)
> Aug 10 08:44:57 MaceWindu kernel: [27438.910107] RTW: usb_read_port_complete() RX Warning! bDriverStopped(False) OR bSurpriseRemoved(False)
> Aug 10 08:44:57 MaceWindu kernel: [27438.910147] RTW: usb_read_port_complete() RX Warning! bDriverStopped(False) OR bSurpriseRemoved(False)
> Aug 10 08:44:57 MaceWindu kernel: [27438.910188] RTW: usb_read_port_complete() RX Warning! bDriverStopped(False) OR bSurpriseRemoved(False)
> Aug 10 08:44:57 MaceWindu kernel: [27438.910226] RTW: usb_read_port_complete() RX Warning! bDriverStopped(False) OR bSurpriseRemoved(False)
> Aug 10 08:44:57 MaceWindu kernel: [27438.910267] RTW: usb_read_port_complete() RX Warning! bDriverStopped(False) OR bSurpriseRemoved(False)
> Aug 10 08:44:57 MaceWindu kernel: [27438.910307] RTW: usb_read_port_complete() RX Warning! bDriverStopped(False) OR bSurpriseRemoved(False)
> Aug 10 08:44:57 MaceWindu kernel: [27438.910311] RTW: usb_write_port_cancel
> Aug 10 08:44:57 MaceWindu kernel: [27438.910321] RTW: ==> rtl8814au_hal_deinit
> Aug 10 08:44:57 MaceWindu kernel: [27438.910548] RTW: CardDisableRTL8814AU
> Aug 10 08:44:57 MaceWindu kernel: [27438.913290] RTW: <=== rtw_ips_pwr_down..................... in 12ms

Is a TP-Link T9UH with of course RTL8814AU. Installed from the latest repo.

Ubuntu 20.04 (in fact,KDE Neon 20.04 which is the same thing).
Kernel: 5.4.0-80-generic

Any ideas? Thanks!

gusarg81, 2021-08-10:
NOTE: strange, tested in my notebook and I don't have this problem (same OS, but kernel 5.11.0-25, don' t know if related).

closed 2021-08-10 by gusarg81

morrownr, 2021-08-10:
Yes, I remember you. Como estas?

You open and close issues faster than I can get to them.

This 8814au driver is not a very good driver. My advice is to not worry too much about the logs unless it is not doing what you want it to do.