morrownr/8814au #37: crash when switch to monitor mode (blackarch 5.13.10-arch1-1 )

paleknight894, 2021-08-17:
after i execute airmon-ng start ,every thing about network crash (every network interface) ,but i can still edit file ,
log:
[  681.046805] RTW: rtw_wx_get_rts, rts_thresh=2347
[  681.046816] RTW: rtw_wx_get_frag, frag_len=2346
[  681.325120] RTW: rtw_ndev_notifier_call(wlp0s20f0u1) state:10
[  681.325250] RTW: netdev_close(wlp0s20f0u1) , bup=1
[  681.325259] RTW: -871x_drv - drv_close, bup=1
[  681.328266] RTW: rtw_ndev_notifier_call(wlp0s20f0u1) state:2
[  681.357292] RTW: cfg80211_rtw_add_virtual_intf(phy2) name:wlp0s20f0u1mon, type:6

morrownr, 2021-08-17:
I can probably offer an alternate way to do airmon-ng start that will work but I need to know the basics of what your setup is:

- distro?
- 32/64 bit?
- ARM, x86?

Some general advice while I wait on your answers: If you are going to be working with aircrack-ng or other pen testing/security apps, my recommendation is to get an adapter that uses drivers that are Linux Wireless Standards compliant. Here is a list of such adapters, links to those adapters and a lot of additional information.

https://github.com/morrownr/USB-WiFi

The specific adapter to buy depends on many factors so if you have questions, please ask.

I give this advice based on personal experience. I have an adapter based on a 8814au. I use it to test this driver and do specific duty but I advise Linux users to not spend new money on adapters based on 8814au because Realtek's support is subpar. The driver here in this repo is the most recent version of the driver for the 8814au and those of us that work on it have put in a lot of hours trying keep it going and make it better but there is only so much we can do. It is not a good driver. You can put lipstick on a pig but it is still a pig.

Cheers

paleknight894, 2021-08-21:
the kernel is from kernel.org linux mainline (amd64)
distro is Arch，but i think it is not important
i just found this driver work fine with linux 5.10 
must something changed from 5.10 to 5.13

paleknight894, 2021-08-21:
also is there any wifi6 usb card 
I can just found AX200(but it is a PCIE card)

paleknight894, 2021-08-21:
> the kernel is from kernel.org linux mainline (amd64)
> distro is Arch，but i think it is not important
> i just found this driver work fine with linux 5.10
> must something changed from 5.10 to 5.13

also i found in 5.10 interface rename failed(failed rename wlan0 to wlan0mon but it switched to monitor mode successful)

morrownr, 2021-08-21:
Quote: 'also is there any wifi6 usb card? I can just found AX200(but it is a PCIE card)"

Reply: I read that DLink is planning to release a wifi 6 usb adapter at some point this year. I have no interest in it for 2 reasons:

1) It is made by DLink and their Linux support is terrible.
2) It contains a Realtek chipset. Realtek's Linux support is bad.

Your best bet is to get a wifi 6 device in card format. They are available in formats for desktop systems and laptops. The chipsets to look for are from Intel, Mediatek and Atheros. I monitor the linux-wireless mailing list where all of the work on wireless drivers takes place, The devs from Intel, Mediatek and Atheros are very active and do a good job. Not so with Realtek.

morrownr, 2021-08-21:
Quote: "also i found in 5.10 interface rename failed(failed rename wlan0 to wlan0mon but it switched to monitor mode successful)"

Reply: Now you are beginning to see why I recommend usb adapters (and cards) that are Linux Wireless standards compliant. What you are trying to do can sort of be done but not the way the book says it should be done. If you are trying to learn pen testing and security analysis, please save your own sanity and go over to the following site and get yourself an adapter that is Linux  Wireless standards compliant:

https://github.com/morrownr/USB-WiFi

On the other hand, if you insist, I can walk you through what you can and cannot do with interface renaming with Realtek drivers.

closed 2022-07-03 by morrownr