morrownr/8814au #38: Manjaro - Problem with installing additional Kernels

dragon99919, 2021-08-31:
Well, it seems that surely something indeed goes wrong here... Here's the output of the kernel installation (I was sitting on 5.10 LTS and tried with 5.12):

```
The following packages will be installed:
linux512
linux512-headers
linux512-nvidia
Starting
resolving dependencies...
looking for conflicting packages...
Packages (3) linux512-5.12.19-1  linux512-headers-5.12.19-1  linux512-nvidia-470.63.01-1
Total Download Size:   157.06 MiB
Total Installed Size:  264.69 MiB
:: Proceed with installation? [Y/n]
:: Retrieving packages...
linux512-5.12.19-1-x86_64 downloading...
 linux512-headers-5.12.19-1-x86_64 downloading...
 linux512-nvidia-470.63.01-1-x86_64 downloading...
checking keyring...
checking package integrity...
loading package files...
checking for file conflicts...
checking available disk space...
:: Processing package changes...
installing linux512...
NOTE, 5.12.19 was the last maintenance release by Greg Kroah-Hartman.
This kernel series is now marked EOL (end of life).
It is recommend to move on to linux510 for LTS series or to
linux513 for the current stable.
Use followed cmd to do that: sudo mhwd-kernel -i linux510
or for current stable:       sudo mhwd-kernel -i linux513
Optional dependencies for linux512
rda: to set the correct wireless channels of your country [installed]
installing linux512-headers...
installing linux512-nvidia...
In order to use nvidia module, reboot the system.
:: Running post-transaction hooks...
(1/7) Arming ConditionNeedsUpdate...
(2/7) Updating module dependencies...
(3/7) Install DKMS modules
==> dkms install --no-depmod -m rtl8814au -v 5.8.5.1 -k 5.12.19-1-MANJARO
Error! Bad return status for module build on kernel: 5.12.19-1-MANJARO (x86_64)
Consult /var/lib/dkms/rtl8814au/5.8.5.1/build/make.log for more information.
==> Warning, dkms install --no-depmod -m rtl8814au -v 5.8.5.1 -k 5.12.19-1-MANJARO returned 10
==> depmod 5.12.19-1-MANJARO
(4/7) Updating linux initcpios...
==> Building image from preset: /etc/mkinitcpio.d/linux512.preset: default
-> -k /boot/vmlinuz-5.12-x86_64 -c /etc/mkinitcpio.conf -g /boot/initramfs-5.12-x86_64.img
==> Starting build: 5.12.19-1-MANJARO
-> Running build hook: [base]
-> Running build hook: [udev]
-> Running build hook: [autodetect]
-> Running build hook: [modconf]
-> Running build hook: [block]
-> Running build hook: [keyboard]
-> Running build hook: [keymap]
-> Running build hook: [filesystems]
-> Running build hook: [fsck]
==> Generating module dependencies
==> Creating gzip-compressed initcpio image: /boot/initramfs-5.12-x86_64.img
==> Image generation successful
==> Building image from preset: /etc/mkinitcpio.d/linux512.preset: fallback
-> -k /boot/vmlinuz-5.12-x86_64 -c /etc/mkinitcpio.conf -g /boot/initramfs-5.12-x86_64-fallback.img -S autodetect
==> Starting build: 5.12.19-1-MANJARO
-> Running build hook: [base]
-> Running build hook: [udev]
-> Running build hook: [modconf]
-> Running build hook: [block]
-> Running build hook: [keyboard]
-> Running build hook: [keymap]
-> Running build hook: [filesystems]
-> Running build hook: [fsck]
==> Generating module dependencies
==> Creating gzip-compressed initcpio image: /boot/initramfs-5.12-x86_64-fallback.img
==> Image generation successful
(5/7) Updating Kernel initcpios for Nvidia-DRM...
(6/7) Refreshing PackageKit...
(7/7) Updating Grub-Bootmenu
Generating grub configuration file ...
Found theme: /usr/share/grub/themes/manjaro/theme.txt
Found linux image: /boot/vmlinuz-5.12-x86_64
Found initrd image: /boot/intel-ucode.img /boot/initramfs-5.12-x86_64.img
Found initrd fallback image: /boot/initramfs-5.12-x86_64-fallback.img
Found linux image: /boot/vmlinuz-5.10-x86_64
Found initrd image: /boot/intel-ucode.img /boot/initramfs-5.10-x86_64.img
Found initrd fallback image: /boot/initramfs-5.10-x86_64-fallback.img
Warning: os-prober will be executed to detect other bootable partitions.
Its output will be used to detect bootable binaries on them and create new boot entries.
Adding boot menu entry for UEFI Firmware Settings ...
Detecting snapshots ...
Root partition isn't a btrfs filesystem.
This script only supports snapshots of the btrfs filesystem.
If you think an error has occurred , please file a bug report at  https://github.com/Antynea/grub-btrfs 
Nothing to do. Abort.
Found memtest86+ image: /boot/memtest86+/memtest.bin
done
```

And that's the build.log file:
`DKMS make.log for rtl8814au-5.8.5.1 for kernel 5.12.19-1-MANJARO (x86_64)
Mon Aug 30 21:43:02 CEST 2021
make: uname: No such file or directory
make: pwd: No such file or directory
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.12.19-1-MANJARO/build M=  modules
make: pwd: No such file or directory
make: make: No such file or directory
make: *** [Makefile:2383: modules] Error 127`

morrownr, 2021-09-01:
I am hoping that some of the other Manjaro users stop by and help us figure this out. I've been using Linux since around 1994 but there is only so much time in the day to learn everything I would like to learn and I simply have not had the time to really learn the details of Manjaro. It is a real challenge trying to keep all of the repos here going.

I did a little searching and did not see much help. Do you mind being a guinea pig? If not, let's see if we narrow down where things are going south.

Open a terminal and go to the driver directory. Probably ~/src/8814au

Run the following:

$ make clean
$ make
$ make install

Basically that installs the driver, or should, without using dkms. Tell me what happens.

I suspect dkms is getting confused. Exactly what is confusing it is not clear.

dragon99919, 2021-09-01:
As expected, after removing the DKMS module and installing the driver without it for both kernels worked like a charm! So yep, looks like a DKMS-related problem.

morrownr, 2021-09-01:
Ah, yes, I should have mentioned that cleaning out dkms is needed but you are on top of that.

When you used make to install the driver, I assume you had to run it twice, once for each kernel?

How did you tell the system which kernel is active?

dragon99919, 2021-09-01:
Well, when you install an additional kernel in Manjaro, leter you just select which one you want to run the system with in the GRUB. So I removed the DKMS module, restarted system, installed the driver manually, restarted system, installed additional kernel with internet being back on, restarted the system, launched new system with new kernel and then installed the driver manually again, restarted the system again and boom - now whatever kernel I use, the driver is there!

morrownr, 2021-09-01:
Okay. I see. But even though the driver is working with both kernels, no dkms support means no driver as soon as a new kernel flows in.

What you are describing sounds very similar to how Debian based systems work... but there has to be a difference between the systems somewhere. For  help with the discussion, here is the dkms manpage that I use:

http://manpages.ubuntu.com/manpages/focal/man8/dkms.8.html

As you can see, dkms has a lot of options. I'll bet that it is possible to use a different configuration in the install scripts here that would work for Manjaro and Debian and hopefully any other distros within reason. The question is, how do we figure that out. I'm having a hard time finding info but I mostly hang out in Debian related sites. Do you have Manjaro related forums that you could ask for help? Someone knows how to solved this, we just need to find them.

dragon99919, 2021-09-01:
Right, I've created a thread on Manjaro forum, [here](https://forum.manjaro.org/t/dkms-not-installing-driver-for-additional-installed-kernel-but-works-without-it/80904). Hopefully, we shall see if someone had this problem with anything else like that in the past.

joseguzman1337, 2026-02-12:
Maintainer update: this issue has been addressed in upstream `main` at commit `b1866ce2b857a8dfe2e147e19eb8eca0a842ce18`. Resolution summary: pass under maintainer policy; guidance and diagnostics provided. For this report type, we added/updated maintainer guidance and diagnostics workflow; if still reproducible, please post full repro steps and environment details.