morrownr/8814au #39: some minor issue with install script

logan001, 2021-09-09:
```
The driver was installed successfully.
Do you want to edit the driver options file now? [y/N] o
Do you want to reboot now? [y/N] n
```
seems that o equals n :)

morrownr, 2021-09-14:
The logic used is that anything other than "y" or "Y" tells the script to not edit the options file. I'm not seeing a problem. 

What do you think the behavior should be?

logan001, 2021-09-18:
repeat the question maybe?

morrownr, 2021-09-18:
How do you think the installation script should handle the issue you reported?

[y/N] indicated there are two answers to the question, yes or no. The capital N also indicates that no is the default reply so if you select Enter or any key other than y or Y, no is the answer you are sending to the script. This logic pretty much follows long standing tradition.

logan001, 2021-09-19:
simple if its neither y/Y/n/N just repeat the question
is it that hard to repeat the question?

morrownr, 2021-09-30:
I am very busy at this time. Please work on the script and submit a PR. I'll take a look.

closed 2022-07-03 by morrownr