morrownr/8814au #40: Offline installation

etterli, 2021-09-16:
I have a computer without a built in wlan adapter and have no access to LAN and can therefore not install any package.

To get a internet connection, it should be possible to compile the driver on an other linux machine and then install the driver manually offline. Compiling the driver works fine with make (ko and mod files are generated).

However, how do I now install the driver manually on the other machine? I tried to "replicate" the install-driver.sh script but this uses dkms and probably other packages under the hood which are not installed on my target machine.

My approach would be to load this kernel module temporarily and install all necessary packages and then install this driver the intended way:

- build driver
- copy *.ko file to other machine
- add signature due to secure boot
- load the driver with insmod
- install the required packages
- unload the driver
- install the driver the intended way (install-driver.sh)

What's your opinion on this approach? Will this work or have I missed anything?

morrownr, 2021-09-17:
Hello etterli,

I have looked at and tested several ideas regarding this subject. I'll give you my solution right up front and then we can discuss problems with the alternatives.

My solution is presented in the following repo:

 https://github.com/morrownr/USB-WiFi

Yes, my solution is to purchase an adapter that uses an in-kernel driver. They are widely available and some can be had for low prices. That doesn't mean that you have to replace your 8814au based adapter if you don't want to do so. What it means is that you have internet access to use the existing installation process. You can run both adapters on the same machine if you have a use case or you can use the new adapter as a spare.

When I look at costs, I look at everything. How much the total cost of anything amounts to more than the money you pay. It also includes the time and aggravation involved so spending a little to avoid the aggravation can make it a good deal.

The biggest problem with your approach is that you would have to repeat the process regularly.... every time your distro releases a new kernel for security and bug fix issues. 

Regards

etterli, 2021-09-19:
Hi morrownr

Thanks for your answer as well as your excellent summary and explanations regarding the USB WiFi topic! Repeating the process regularly is clearly not that nice.. So I will probably buy a well supported adapter provided by your list :)

For anyone else, as I wanted to experiment a little with kernel modules I've tried and managed to build and install the driver successfully the way I described above. Hint: I've just had to resolve some unknown symbols errors by first adding the module to the /lib/modules/'uname -r'/modules.dep and running depmod.

closed 2021-09-19 by etterli