morrownr/8814au #49: channel 44 not available anymore for AP

logan001, 2021-11-09:
somehow it seems that latest git (d36c0874716b0776ac6c7dcd6110598ef0f6dd53) does not allow channel for ap anymore.
i've roll back a few commit (ab1d7f21b4a68ed470caf5c211c66d651c1d2fff) and i can use channel 44 for AP.

morrownr, 2021-11-09:
Hmmm... let me see if I can figure out what is going on.

morrownr, 2021-11-10:
I've tested since last night and am not duplicating the problem. Maybe it was a configuration problem. If you see it again, send some conf files and let's see what we can see.

I merged some things this morning but it was all documentation related things. No code was changed.

Regards

logan001, 2021-11-10:
this is the hostap conf
```
interface=wlan1
country_code=RO
driver=nl80211
ssid=###########
hw_mode=a
channel=44
ieee80211n=1
ieee80211ac=1
wmm_enabled=1
ieee80211d=0
ieee80211h=0

vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=50

ht_capab=[RX LDPC][HT40+][HT40-][MAX-AMSDU-7935][DSSS_CCK-40][RX-STBC3][SHORT-GI-40][RX-STBC][RXLDPC][DSSS_CCK-40]
vht_capab=[Max-MPDU-11454][SHORT-GI-80][RX-STBC][RX-LDPC][TX-STBC][TX-STBC-2BY1][SU-BEAMFORMEE][HTC-VHT]

wpa=2
wpa_passphrase=##############
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP

```

and this is the modprobe conf
`options 8814au rtw_drv_log_level=0 rtw_led_ctrl=1 rtw_vht_enable=2 rtw_power_mgnt=0 rtw_switch_usb_mode=1 rtw_beamform_cap=3`

i forgot to mention that the adapter is usb-ac68
anyway this same hostap conf fails to create ap cause the channel 44 is not available but if i "downgrade" the driver it will work.

morrownr, 2021-11-10:
Something things to try:

> this is the hostap conf
> 
> ht_capab=[RX LDPC][HT40+][HT40-][MAX-AMSDU-7935][DSSS_CCK-40][RX-STBC3][SHORT-GI-40][RX-STBC][RXLDPC][DSSS_CCK-40]

ht_capab=[LDPC][HT40+][HT40-][SHORT-GI-20][SHORT-GI-40][MAX-AMSDU-7935][DSSS_CCK-40]

> vht_capab=[Max-MPDU-11454][SHORT-GI-80][RX-STBC][RX-LDPC][TX-STBC][TX-STBC-2BY1][SU-BEAMFORMEE][HTC-VHT]

vht_capab=[MAX-MPDU-11454][RXLDPC][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-1][HTC-VHT][MAX-A-MPDU-LEN-EXP7]

> and this is the modprobe conf `options 8814au rtw_drv_log_level=0 rtw_led_ctrl=1 rtw_vht_enable=2 rtw_power_mgnt=0 rtw_switch_usb_mode=1 rtw_beamform_cap=3`

options 8814au rtw_drv_log_level=0 rtw_led_ctrl=0 rtw_vht_enable=2 rtw_power_mgnt=1 rtw_switch_usb_mode=1

> i forgot to mention that the adapter is usb-ac68 anyway this same hostap conf fails to create ap cause the channel 44 is not available but if i "downgrade" the driver it will work.

I'm going to continue looking as I have time. If you have anything to add that might help, please do.

One thing I know about this old driver is that it can be cranky sometimes. We need new, modernized code for this chipset.

Regards

morrownr, 2021-11-10:
opps... update:

vht_capab=[MAX-MPDU-11454][RXLDPC][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-123][HTC-VHT][MAX-A-MPDU-LEN-EXP7]

morrownr, 2021-11-10:
Be careful with the last message above. [RX-STBC-123] is not working here. Use the previous if you have problems. Let me mess my own stuff up and not yours. I'll be back after I see what is going on.

logan001, 2021-12-27:
ok. it seems that i was right. partially.

for some reasons on some reboots channel 44 is not allowed (AP mode). i have to reboot to make it work. replugging the dongle will not work. i really have no clue why this is happening

morrownr, 2021-12-27:
Hi @logan001 

Something I just noticed that I had overlooked before is that your setup should not work in theory. I say that but I don't know the regular and DFS channels in your country so correct me if they are different than the US.

I can tell with the following settings that you want an 80 MHz channel width:

vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=50

rtw_vht_enable=2

Setting 44(50), without DFS support, should not work because channel 50 is a DFS channel. Your only real options for 80 MHz channel width while using regular channels is 36(42) and 149(155). DFS would be an option but I do not think this driver supports DFS channels. The more recent Realtek drivers that I have up do support DFS channels in AP mode but Realtek has not released an update for this driver since 2019 and it lacks many features of the newer drivers. I am keeping a watch out for a new version but nothing so far.

I may have some time later today to reconfigure my access point setup so that I can see what is going on here. I can run the code in the driver to check for issues but first I need to see what is going on.

Regards

morrownr, 2021-12-28:
@logan001 

I have a report for you.

After re configuring to use an adapter powered by an 8814au chipset in my RasPi4B access point, I set 44(50). I was able to connect to the AP and use the connection. I then used a scanner to verify which channels were actually in use. Since hostapd knows that it can't use 44(50), it had overridden my settings and used channel 36(38). So, it was centered on 38. It was an 80 MHz wide channel so it took up the frequencies from channel 30 to channel 46. My normal setting for an 80 MHz width channel in this part of the spectrum is 36(42) which provides a center channel of 42 and takes up the frequencies from channel 32 to 48.

Have you verified what channels your adapter is actually broadcasting on? hostapd will try to make things work, maybe not always successfully. With this driver, in 5 GHz, you really only have 2 options if you want 80 GHz channel width: 36(42) and 149(155).

logan001, 2021-12-29:
well i disable those lines in hostdap conf. and the result is still `IEEE 802.11 Configured channel (44) not found from the channel list of current mode (2) IEEE 802.11a`
so i guess channel 50 is not the problem. i just can't use channel 44. also the channel 50 is not shown on iw list.

logan001, 2021-12-29:
and it seems you are right. when my hostapd works it actually does not work on channel 44. it seems the app on the phon is shows channel 36(38). nice

logan001, 2021-12-29:
i was not capable of setting ch 36 with 42. the AP will give channel 36 is available.

morrownr, 2021-12-29:
> well i disable those lines in hostdap conf

I am confused. Which two lines did you disable?

> and the result is still `IEEE 802.11 Configured channel (44) not found from the channel list of current mode (2) IEEE 802.11a` so i guess channel 50 is not the problem. i just can't use channel 44. also the channel 50 is not shown on iw list.

It may seem like you have a lot of options but you really do not:

Option 1:

channel=36
vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=42

Option2:

channel=149
vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=155

> edit: all routers around me are using channel 36 for 5ghz.

If all routers around you are using channel 36 then use option 2 as shown above.

logan001, 2021-12-29:
the lines for 80Hz 

vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=[channel+6]

as for option 2...well:

```
* 5720 MHz [144] (disabled)
* 5745 MHz [149] (disabled)
* 5765 MHz [153] (disabled)
* 5785 MHz [157] (disabled)
* 5805 MHz [161] (disabled)
* 5825 MHz [165] (disabled)
* 5845 MHz [169] (disabled)
* 5865 MHz [173] (disabled)
* 5885 MHz [177] (disabled)

```

morrownr, 2021-12-30:
I did some searching. For country_code=RO, the allowable 5 GHz channels are:

36 40 44 48 52 56 60 64 100 104 108 112 116 132 136 140

The information that I found did not specific which, if any, are DFS channels. How about you research that?

This driver is not capable of supporting DFS channels. I wish it could but it can't. There are drivers available that can support DFS channels but they are available for adapters with different chipsets:

https://github.com/morrownr/8812au-20210629
https://github.com/morrownr/8821au-20210708

I do not know why Realtek support for the 8814au is so poor but it is.

Regards

morrownr, 2022-01-18:
I'm closing this issue as it appears there is not much that can be done.

Regards

closed 2022-01-18 by morrownr