morrownr/8814au #52: Unable to install

Arjdroid, 2021-11-17:
I was on a previous old version of the driver on Arch Linux but after updating to kernel 5.15 today my WiFi was no longer working.

So I ran the remove driver script, rebooted, cloned the newest version of the GitHub (connected to the internet via my phone's usb tethering).

When I tried running the driver install script with sudo, it said that a DKMs tree for rtl88q4au-5.8.5.1 already exists...

I tried rebooting again, I tried running the remove script (which says success) and rebooting and then trying again, I've also tried running the remove script without rebooting and then trying to install but it won't let me. It seems the remove script is broken...

morrownr, 2021-11-17:
Hi @Arjdroid 

While I keep a partitian with Manjaro on it for basic testing, I learned not to install an upgraded kernel. I live in the Debian world and am not familiar with the technical knowledge of the Arch package management system. I have worked with a couple of Arch/Manjaro users to see if there is a good way to modify the install and remove scripts here but we could not find a good way to do it... we could do it but it would break the installation on other distros. 

This repo is the upstream for the AUR 8814au driver:

https://aur.archlinux.org/packages/rtl8814au-dkms-git/

Maybe you get the driver there and it will not cause any problems. I think the driver is repackaged for Arch/Manjaro. Sorry I can't be more helpful but I don't have the time to be an expert in all areas.

Nick

Arjdroid, 2021-11-17:
No problem! I appreciate your great work on keeping this driver up to date. For now I've just switched to the linux-lts kernel package for the time being to get my work done and since linux-lts is on 5.10 it's working great right now. When I have more time I will test out the AUR package. Thank you for your quick response.

Arjdroid, 2021-11-21:
I have installed the aur package have you have told and it seems to work pretty well on my other Arch Linux system (with similar WiFi adapter). This time I tried being more careful so I decided to un-install the 8814au using the remove-driver script _before_ running pacman -Syu to update my kernel and everything. After rebooting, I used USB tethering on my phone to connect the PC to the internet and then ran my updates as normal and then I installed the rtl8814au-dkms-git AUR package and it is working great on kernel 5.15 right now. Thank you for helping out fix the issue.

closed 2021-11-21 by Arjdroid