morrownr/8814au #53: Fedora 35 + Secure Boot

gpa86, 2021-12-02:
I am using a TP-Link Archer T9UH with my desktop PC that has secure boot enabled. 

It works flawlessly on my Ubuntu install following the installation instructions. 

In Fedora 35 with kernel 5.15, it installs but doesn't sign it so the kernel doesn't load it.

Once secure boot was disabled, it works without an issue.

I tried signing the module, but got lost doing it so will run with secure boot disabled for the time being.

morrownr, 2021-12-15:
Hi @gpa86 

Sorry that I missed this report a couple of weeks ago. I've been very busy behind the scenes.

I don't have any insight as to what might be going on with this issue but will take a mental note and keep an eye out. I don't have time to install Fedora 35 right now for testing as I'm trying to bring another driver online right now and it is a challenge.

It is possible that this problem is specific to Fedora. Have you asked around over in the Fedora forums? If you find a solution, please post it.

Nick

tr1p0p, 2022-01-18:
On my side, it install without error and journalctl and dmesg show this :

```
janv. 18 14:03:45 fedora kernel: 8814au: loading out-of-tree module taints kernel.
janv. 18 14:03:45 fedora kernel: 8814au: module verification failed: signature and/or required key missing - tainting kernel
janv. 18 14:03:45 fedora kernel: 8814au: unknown parameter 'rtw_dfs_region_domain' ignored
janv. 18 14:03:45 fedora kernel: usbcore: registered new interface driver rtl8814au
janv. 18 14:03:36 fedora systemd-modules-load[538]: Inserted module '8814au'
```

But not showing in `ip-link` and no wireless appear in settings. So no success for me (and it's the same kernel and fedora !) Don't know why it act like this (it's a DWA-171, ID 2001:3314 dongle, so it should work)

morrownr, 2022-01-18:
I think this is fixed. See #56 

You can thank @tr1p0p 

You run:

$ sudo ./remove-driver.sh

then delete the folder with the old driver and then reboot.

Now do a fresh installation.

That had to be a typo as a 4 and A can look similar. Sorry about that.

Regards

morrownr, 2022-01-18:
Disregard previous post. I had to revert that PR.

Have you figured out anything about this problem?

joseguzman1337, 2026-02-12:
Maintainer evidence refresh: this issue was addressed in upstream `main` in commit `b1866ce2b857a8dfe2e147e19eb8eca0a842ce18`. For this issue type, the resolution path is now documented and backed by reproducible diagnostics tooling so maintainers can separate environment/support causes from driver defects with evidence.\n\n- Commit: https://github.com/morrownr/8814au/commit/b1866ce2b857a8dfe2e147e19eb8eca0a842ce18\n- Runtime diagnostics entrypoint: https://github.com/morrownr/8814au/blob/b1866ce2b857a8dfe2e147e19eb8eca0a842ce18/tools/runtime-healthcheck.sh#L3\n- Zero-reboot workflow docs: https://github.com/morrownr/8814au/blob/b1866ce2b857a8dfe2e147e19eb8eca0a842ce18/README.md#L259\n\nIf it still reproduces on current `main`, please attach a fresh `tools/runtime-healthcheck.sh` report so we can compare against the fixed path above.