morrownr/8814au #55: unknown parameter 'rtw_dfs_region_domain' ignored

tr1p0p, 2022-01-18:
Hi !

Trying to install this on last fedora 35 on linux kernel 5.15.14, with a ID know by device. Everything should be fine, except it doesn't.

So, install don't show error, but module not loaded automatically at boot. Trying to make it, and `journalctl | grep 8814au ` show this :

```
janv. 18 11:55:59 fedora kernel: 8814au: loading out-of-tree module taints kernel.
janv. 18 11:55:59 fedora kernel: 8814au: module verification failed: signature and/or required key missing - tainting kernel
janv. 18 11:55:59 fedora kernel: 8814au: unknown parameter 'rtw_dfs_region_domain' ignored
janv. 18 11:55:59 fedora kernel: usbcore: registered new interface driver rtl8814au
janv. 18 11:55:52 fedora systemd-modules-load[536]: Inserted module '8814au'
```
So i wonder why rtw_dfs_region_domain doesn't work anymore (got no mention on this anywhere) and also want to testify that this driver don't work for me on fedora 35 (no secure boot, no CSM).

Think maybe it's related to this line ?

morrownr, 2022-01-18:
Hi @tr1p0p 

Thanks for the report. `rtw_dfs_region_domain` was an oversight on my part as this driver does not support DFS channels in AP mode. I have now removed it and will merge the change soon. You can simply remove the entry and reboot. The easiest way to get the right file is:

$ sudo ./edit-options.sh

To the real problem: You are now the second person on Fedora 35 reporting the same thing. Let's see if we can figure out what is going on. Please post the results of the following command:

$ sudo uname -a; mokutil --sb-state; lsusb; rfkill list all; dkms status; iw dev

tr1p0p, 2022-01-18:
Ok i will do that immediately.

Also, the output :
```
Linux fedora 5.15.14-200.fc35.x86_64 #1 SMP Tue Jan 11 16:49:27 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
This system doesn't support Secure Boot
Bus 002 Device 006: ID 046d:c328 Logitech, Inc. Corded Keyboard K280e
Bus 002 Device 005: ID 046d:c05a Logitech, Inc. M90/M100 Optical Mouse
Bus 002 Device 004: ID 054c:09c2 Sony Corp. Storage Media
Bus 002 Device 003: ID 046d:082b Logitech, Inc. Webcam C170
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 2001:3314 D-Link Corp. DWA-171 AC600 DB Wireless Adapter(rev.A1) [Realtek RTL8811AU]
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 004: ID 18d1:4ee7 Google Inc. Nexus/Pixel Device (charging + debug)
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
0: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
rtl8814au/5.8.5.1, 5.15.14-200.fc35.x86_64, x86_64: installed
phy#0
	Interface wlp6s0u2
		ifindex 3
		wdev 0x1
		addr 72:43:b9:d6:c3:ba
		type managed
		txpower 20.00 dBm
```

The scan don't show any wifi in gnome settings for now

morrownr, 2022-01-18:
> ID 2001:3314 D-Link Corp. DWA-171 AC600 DB Wireless Adapter(rev.A1) [Realtek RTL8811AU]

You are in the wrong repo. Run:

$ sudo ./remove-driver.sh

Delete the driver folder, reboot and then go to ...

https://github.com/morrownr/8821au-20210708

That is the driver for the 8811au.

tr1p0p, 2022-01-18:
Well. I'm dumb, don't know how i didn't saw this earlier. Closing. Thanks for noticing !

closed 2022-01-18 by tr1p0p

morrownr, 2022-01-18:
We needed a little excitement foday. Keeping up with all the details is a challenge. I reverted the PR after cross checking. Let me know how that 8821au driver works... but post the message in the right repo. He he.

FWIW: Be glad you have the 8811au chipset. That 8821au driver is a much better driver. It has several capabilities that this driver does not have.