morrownr/8814au #58: The driver doesnt work on Kali Linux (Kernel 5.15)

shuopz, 2022-02-12:
The driver works perfectly on Ubuntu 21.10 (kernel 5.13). But Kali Linux misfired.

How to reproduce:

1) Install the official **Virtual Box image** for Kali Linux.

2) Follow the instructions for this driver. You wont be able to perform the third step without upgrading the system (this is required by the linux-headers-* package).  

Result: an adapter (Alfa AWUS 1900) cant scan access points. It doesnt see any of them.

morrownr, 2022-02-14:
Hi @shuopz 

> Follow the instructions for this driver. You wont be able to perform the third step without upgrading the system (this is required by the linux-headers-* package).

I do not understand. Can you reword this?

shuopz, 2022-02-14:
@morrownr Hi. Follow the steps in your driver installation instruction (README.md)

I just wanted to warn you that if an error occurs in the step 3, it is resolved by running the commands:

`sudo apt update && sudo apt upgrade`

morrownr, 2022-02-14:
> I just wanted to warn you that if an error occurs in the step 3, it is resolved by running the commands:

> sudo apt update && sudo apt upgrade

Okay. I am not sure where things went wrong here. Step 2 tells you to do what you are warning me about. Did you skip Step 2? Should I reword Step 2?

shuopz, 2022-02-14:
@morrownr No, I didn't skip step 2, I only did `sudo apt update` first. And that was enough for Ubuntu. But on Kali Linux you still need to do `sudo apt upgrade`.  Because if you don't do that, in step 3 you get an error something like "Cant find the linux-headers-* package".  At least that's what I encountered.

> Should I reword Step 2

No, you already have a note in step 2 about installing updated packages.