morrownr/8814au #60: Kali 2021.4a & AWUS1900 - not fully working

magitechIT, 2022-03-06:
Hi,
I have installed the driver for my USB Wifi adapter but when I perform the check:
`aireplay-ng --test -i wlan0 wlan1`
I got the following results:

```
19:32:26  Trying card-to-card injection...
19:32:26  Attack -0:           OK
19:32:26  Attack -1 (open):    OK
19:32:26  Attack -1 (psk):     OK
19:32:26  Attack -2/-3/-4/-6:  OK
19:32:26  Got a reply with an incorrect sequence number (wanted 0x0a00 got 0x9a9b).
19:32:26  Got a reply with an incorrect sequence number (wanted 0x0a00 got 0x9a9b).
19:32:26  Got a reply with an incorrect sequence number (wanted 0x0a00 got 0x9a9b).
19:32:26  Got a reply with an incorrect sequence number (wanted 0x0a00 got 0x9a9b).
19:32:29  Got a reply with an incorrect sequence number (wanted 0x0a00 got 0x5a9c).
19:32:34  Got a reply with an incorrect sequence number (wanted 0x0a00 got 0x8a9f).
19:32:35  Got a reply with an incorrect sequence number (wanted 0x0a00 got 0xaa9f).
19:32:38  Attack -5/-7:        Failed
```
What do I miss?

Thx

morrownr, 2022-04-06:
Hi @magitechIT 

You may not be missing anything. I can investigate when I have time but users of monitor mode often find that Mediatek based adapter provide better results:

https://github.com/morrownr/USB-WiFi

ljluestc, 2023-06-25:
```
echo "Enabling monitor mode..."
sudo airmon-ng start wlan0
echo "Testing different injection methods..."
aireplay-ng --test -9 -i wlan0 wlan1
```

joseguzman1337, 2026-02-12:
Maintainer update: this issue has been addressed in upstream `main` at commit `b1866ce2b857a8dfe2e147e19eb8eca0a842ce18`. Resolution summary: AWUS1900 hot-plug native/oot switching + runtime diagnostics now provide deterministic no-reboot remediation path. This is mitigated with code and runtime validation tooling; if you still reproduce, please share fresh diagnostics from `tools/runtime-healthcheck.sh`.