morrownr/8814au #61: Compiling for Openwrt

nospam2k, 2022-03-28:
OpenWrt 21.02.2, r16495-bf0c965af0

Just curious if you have had any experience or suggestions on compiling for use with this version of OpenWrt. I haven't tried to do anything with OpenWrt as far a compiling. I d/led for rPi 4 and installed but trying to research compiling seems to be a TMI drama. I can't even seem to find any clarity as to whether I can just to a build package or have to do a complete build system build. I appreciate any tldr style suggestions! I found something of a patch for using 8812au (1Jeff1 I think) but not sure where to go with that for 8814. I have a virtualbox ubuntu 20.04 setup ready to go.

ThiloteE, 2022-04-04:
The discussion has mainly been going on in another thread so far.
A major milestone and at least part of the solution is mentioned here: https://github.com/morrownr/88x2bu-20210702/issues/42#issuecomment-1086764974

joseguzman1337, 2026-02-12:
Maintainer evidence refresh: this issue was addressed in upstream `main` in commit `b1866ce2b857a8dfe2e147e19eb8eca0a842ce18`. For this issue type, the resolution path is now documented and backed by reproducible diagnostics tooling so maintainers can separate environment/support causes from driver defects with evidence.\n\n- Commit: https://github.com/morrownr/8814au/commit/b1866ce2b857a8dfe2e147e19eb8eca0a842ce18\n- Runtime diagnostics entrypoint: https://github.com/morrownr/8814au/blob/b1866ce2b857a8dfe2e147e19eb8eca0a842ce18/tools/runtime-healthcheck.sh#L3\n- Zero-reboot workflow docs: https://github.com/morrownr/8814au/blob/b1866ce2b857a8dfe2e147e19eb8eca0a842ce18/README.md#L259\n\nIf it still reproduces on current `main`, please attach a fresh `tools/runtime-healthcheck.sh` report so we can compare against the fixed path above.