morrownr/8814au #62: Host AP as dual band. Is this possible?

nospam2k, 2022-04-01:
I'm not really up on WiFi card so I'm not sure if I'm on track. Is it possible to get a phy0 and phy1 device for each band? I want to set up a dual band, different SSID access point using this device if possible. I've looked at the docs for hostapd and it says it requires two physical devices and then says that dual physical devices are rare. I'm not sure if this is a driver trick or not.

morrownr, 2022-04-06:
Hi @nospam2k 

It is certainly possible to make a dual band AP with USB WiFi adapters. I do it. First, let's look at how common dual band wifi routers support dual bands, Dual band wifi routers have TWO internal radios. One radio for each band supported.

USB WiFi adapters only contain one radio. That radio can work on the 2.4 GHz band or 5 GHz band but not both at the same time. This means that you need two adapters to make a dual band AP.

nospam2k, 2022-04-06:
Yes. That was my point. The idea of having to use multiple wifi adapters. I was fighting with the driver for openwrt only to realise I would have to buy yet another adapter to do dual band and started thinking "is this really worth it just to get openwrt working?" and decided to just see if I can find a compatible router. Thanks for the follow up though!

closed 2022-04-06 by nospam2k