morrownr/8814au #63: depmod: ERROR: Bad version passed 8814au.ko

Colum31, 2022-04-06:
After I have built the driver with make, I get following error:

```
sudo depmod 8814au.ko
depmod: ERROR: Bad version passed 8814au.ko
```

I am on Fedora 34. When I use the dkms install method, the module does not load too, yielding the same error.
Is there any way to fix this?

```
uname -r
5.16.18-100.fc34.x86_64
```

```
modinfo 8814au
[...]
vermagic:       5.16.18-100.fc34.x86_64 SMP preempt mod_unload 
```

All headers are installed.

morrownr, 2022-04-06:
Hi @Colum31 

> I am on Fedora 34. When I use the dkms install method, the module does not load too, yielding the same error.
Is there any way to fix this?

The most common problem with non-Debian based distros seems to be with secure mode. I'm not aware of any secure mode issues with Debian based issues. It is one of those things that just works. However, many distros do their own thing with some aspects of the operating system and I have not had the time to sort those issues out and would really need an expert with those distros to sort the issues out.

With that said, if you have secure mode turned on, you might try to turning it off, at least temporarily, to see it has something to do with this issue.

Regards

Colum31, 2022-04-07:
Hello @morrownr ,

I have tested it with two seperate machines, one with and without secure boot enabled.

It does not work on both.

morrownr, 2022-04-07:
Good day @Colum31 

I think the best way to work toward a fix in this case is for me to try to duplicate the issue. Can you provide me some help to setup an installation like yours?

- Link to the Fedora iso that you have installed.
- If the kernel is upgraded, how did you do the upgrade?

Regards

Colum31, 2022-04-09:
Hello,

I reproduced the error in a VM. 
I used GNOME boxes, and have downloaded and installed the [current Fedora 35 ISO (x86_64)](https://getfedora.org/en/workstation/download/).

sha256: 
`85d9d0c233d560e401e2ad824aa8e6d5614e8b977dfe685396bfb2eb3ba5b253  Fedora-Workstation-Live-x86_64-35-1.2.iso
`

1. Install to virtual disk.
2. `sudo dnf update`
3. reboot
4. `sudo dnf -y install git dkms kernel-devel kernel-debug-devel`
5. reboot
6. clone source
7. cd source
8. sudo make
9. depmod 8814au.ko

I forgot to mention, that I got the following warning, when building (in vm and on target system):

`Skipping BTF generation for /home/daniel/Treiber/8814au/8814au.ko due to unavailability of vmlinux`
This might be the reason, why it does not work.

Regards

morrownr, 2022-04-10:
>`Skipping BTF generation for /home/daniel/Treiber/8814au/8814au.ko due to unavailability of vmlinux`

>This might be the reason, why it does not work.

That statement is info only and should not be a problem.

Glad you spelled out the installation steps you are using. I have some concerns:

You seem to be doing a manual installation. That's fine but is there a reason that you are not using the installation script ( install-driver.sh ) per the installation guide?

I see problems with your manual installation starting with step 8:

```
sudo make
depmod 8814au.ko
```

Change those two lines to these 4 lines:

```
make clean
make
sudo make install
sudo reboot
```

Tell me what happens.

Regards

Colum31, 2022-04-11:
Hello,

the manual instructions work. I think my initial issue was secure boot related and caused by signing and installing wrong.
Thus my steps to reproduce the issue were wrong too.

I am sorry for wasting your time.
Regards

morrownr, 2022-04-12:
No problem.

closed 2022-04-13 by Colum31