morrownr/8814au #64: Compilation failed using 5.18

IsaacMvmv, 2022-05-03:
I know it is not supported, but here is the log showing what fails
[log.log](https://github.com/morrownr/8814au/files/8608818/log.log)

morrownr, 2022-05-03:
Well, that is ugly. What distro are you using?

I'll take a look as able.

Regards

N0tN, 2022-05-09:
Archlinux

morrownr, 2022-05-12:
> archlinux

I'm not an archlinux expert but I've managed to help a few users at times.

> As far as I know, secure mode isnt on

Run the following and post the output:

sudo uname -a; mokutil --sb-state; lsusb; rfkill list all; dkms status; iw dev

> Im using right headers, right kernel, and same error

Is there a way in archlinux to check which headers are being used?

I know enough to know that this is an issue that is confusing and 5.18 has not been released yet? Where did you get it and the headers?

Regards

IsaacMvmv, 2022-05-12:
The link: https://git.kernel.org/torvalds/t/linux-5.18-rc6.tar.gz

morrownr, 2022-05-12:
I decided to throw 5.18 rc6 on a test system that runs Ubuntu 22.04. I went to download the Mainline 5.18 rc6 from:

https://kernel.ubuntu.com/~kernel-ppa/mainline

And I see that the amd64 version failed self tests:

Test amd64/self-tests failed

I looked at the logs. Some stuff failed that could be a problem with what we are trying to do.

Recommend you wait on 5.18 rc7 and see what happens. Let me know what happens.

Regards

IsaacMvmv, 2022-05-13:
I have installed it using this repo: https://github.com/Frogging-Family/linux-tkg

It has a customization.cfg file which lets yu choose the scheduler, frequency, target cpu, etc

I think to build it on ubuntu you can use install.sh
It includes many patches to support winefutex, anbox....
You can choose if adding them or not

morrownr, 2022-05-14:
Hi @IsaacMvmv 

I've had a bug over the last week and am having to take it easy.

I noticed that rc4 was the last rc that passed tests as far as Ubuntu is concerned. You might look at trying rc4. I will hold off until such time as an rc that passes all tests is available as I don't have extra time to chase down issues that very well may be bugs in a not-yet-released kernel. 

Regards

morrownr, 2022-05-16:
Hi @IsaacMvmv 

I was able to test and determine that this driver needed a patch for 5.18. Please remove what you have been using and try the newly patched version. Please let me know.

Regards

IsaacMvmv, 2022-05-22:
Working now

thanks ;)

closed 2022-05-22 by IsaacMvmv

morrownr, 2022-05-22:
Thanks for the report.

Cheers