morrownr/8814au #65: Latest DKMS Install for 8814au on ORACLE LINUX ( 99% like Fedora, Red Hat & CentOS )

jtome7, 2022-05-08:
The latest DKMS installation for 8814au installed without any errors on ORACLE LINUX 8.5 which is 99% identical to Fedora, Red Hat and CentOS.   HOWEVER, it doesn't work for my TP-LINK AC1900 WIFI Adapter  ( ARCHER T9-UH ).  The Adapter does not wake up and work at all.  What might I need to do?  What "option" might I need to change?

morrownr, 2022-05-08:
Hi @jtome7 

I need you to run and post the results of the following:

```
sudo uname -a; mokutil --sb-state; lsusb; rfkill list all; dkms status; iw dev
```

Regards

jtome7, 2022-05-09:
Apparently, this PC has some kind of WIFI built-in ( Interface wlp4s0 ) in
addition to the TP-LINK Archer T9-UH Wifi Adapter ( wlp0s26u1u3 ) that I'm
trying to get working with the RTL8814AU driver.

The TP-LINK AC1900 has an LED that blinks, but I cannot get it to work to
send & receive network traffic.

***@***.*** ~]# uname -a; mokutil --sb-state; lsusb; rfkill list all;
dkms status; iw dev

Linux walkertown 5.4.17-2136.306.1.3.el8uek.x86_64 #2 SMP Fri Apr 8
14:50:24 PDT 2022 x86_64 x86_64 x86_64 GNU/Linux

EFI variables are not supported on this system

Bus 002 Device 003: ID 058f:6362 Alcor Micro Corp. Flash Card Reader/Writer
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 004: ID 2357:0106 TP-Link Archer T9UH v1 [Realtek RTL8814AU]
Bus 001 Device 007: ID 10d5:55a4 Uni Class Technology Co., Ltd
Bus 001 Device 006: ID 04f3:0213 Elan Microelectronics Corp.
Bus 001 Device 008: ID 0430:00a2 Sun Microsystems, Inc. Type 7 Keyboard
Bus 001 Device 005: ID 0430:100e Sun Microsystems, Inc. 24.1" LCD Monitor
v4 / FID-638 Mouse
Bus 001 Device 003: ID 058f:9254 Alcor Micro Corp. Hub
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

0: phy0: Wireless LAN
Soft blocked: no
Hard blocked: no
1: phy1: Wireless LAN
Soft blocked: no
Hard blocked: no

rtl8814au/5.8.5.1, 5.4.17-2136.306.1.3.el8uek.x86_64, x86_64: installed
(original_module exists)

phy#1
Interface wlp0s26u1u3
ifindex 4
wdev 0x100000001
addr 98:48:27:2a:58:5c
ssid SBG6900AC-DC966
type managed
txpower 20.00 dBm
phy#0
Interface wlp4s0
ifindex 3
wdev 0x1
addr 74:e5:43:9c:27:d9
type managed
txpower 30.00 dBm

jtome7, 2022-05-09:
Some more information on what I've been doing to try to get my TP-LINK
AC1900 Archer T9UH wifi adapter to work.  It appears to have some
communication with the driver because the L.E.D. lights up and blinks as if
sending/receiving data; however, there is no connection to the Internet and
it does not receive an I.P. Address from DHCP:

# iwconfig
eno1      no wireless extensions.

lo        no wireless extensions.

virbr0-nic  no wireless extensions.

virbr0    no wireless extensions.

wlp0s20f0u3  unassociated  ***@***.***"
          Mode:Managed  Frequency=2.437 GHz  Access Point: Not-Associated
          Sensitivity:0/0
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

          ~~~~~~~~~~~~~~~~~~~~~

# iwlist wlp0s20f0u3 scan
wlp0s20f0u3  Scan completed :

Cell 05 - Address: 5C:E3:0E:96:D7:9F
                    ESSID:"SBG6900AC-DC966"
                    Protocol:IEEE 802.11bgn
                    Mode:Master
                    Frequency:2.437 GHz (Channel 6)
                    Encryption key:on
                    Bit Rates:144 Mb/s

Extra:rsn_ie=30140100000fac040100000fac040100000fac020c00
                    IE: IEEE 802.11i/WPA2 Version 1
                        Group Cipher : CCMP
                        Pairwise Ciphers (1) : CCMP
                        Authentication Suites (1) : PSK
                    IE: Unknown:
DD830050F204104A0001101044000102103B00010310470010B30690D371265D1F08803B15DA911B37102100084D6F746F726F6C61102300084D6F746F726F6C611024000631323334353610420007303030303030311054000800060050F20400011011000A4D6F746F726F6C614150100800022008103C0001031049000600372A000120
                    Quality=36/100  Signal level=52/100
                    Extra:fm=0003

          ~~~~~~~~~~~~~~~~~~~~~

COMMANDS TO BRING UP THE WIFI INTERFACE

# ifconfig wlp0s20f0u3

# iwlist wlp0s20f0u3 scan

# iwconfig wlp0s20f0u3 essid SBG6900AC-DC966 key s:64556212dd

# dhclient wlp0s20f0u3

          ~~~~~~~~~~~~~~~~~~~~~

# ifconfig wlp0s20f0u3
wlp0s20f0u3: flags=4099<*UP*,BROADCAST,MULTICAST>  mtu 1500
        ether 02:14:59:e9:12:e1  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 1835906 (1.7 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 57720 (56.3 KiB)
        TX errors 0  dropped 113 overruns 0  carrier 0  collisions 0

# iw wlp0s20f0u3 info
Interface wlp0s20f0u3
ifindex 3
wdev 0x1
addr 02:14:59:e9:12:e1
type managed
wiphy 0
txpower 20.00 dBm

# dhclient wlp0s20f0u3
dhclient(4963) is already running - exiting.

This version of ISC DHCP is based on the release available
on ftp.isc.org. Features have been added and other changes
have been made to the base software release in order to make
it work better with this distribution.

Please report issues with this software via:
https://bugzilla.oracle.com/

exiting.

*Thanks!**Joel*

*************************** Joel Tome *

*jtome7 @ gmail.com <http://gmail.com>****************************

morrownr, 2022-05-10:
It is not obvious to me what the problem is. All indications are that the driver compiled, installed and is working. Here are things to try:

- Did you reboot after compiling the driver? If not, do so.
- Is the AP/router using WPA3 or WPA3 mixed? If so, drop down to WPA2. This driver can't do WPA3.
- Does the AP/router have 802/11w turned on. If so, turn it off.

jtome7, 2022-05-10:
Thanks for those excellent extra tips!

I'll work on it a little more tomorrow.

I was searching for LOGS in /var/log but didn't find any.  Where are the
logs stored?

*Thanks for all your help!*

*Joel*

*************************** Joel Tome *

*jtome7 @ gmail.com <http://gmail.com>****************************

morrownr, 2022-05-11:
> I was searching for LOGS in /var/log but didn't find any.  Where are the
logs stored?

Check the documentation in `8814au.conf`. An easy way to open the active file is:

```
sudo ./edit-options.sh
```

Logging is off by default. Recommend a setting of 4. You can save a log that only includes entries from this driver by running:

```
sudo ./save-log.sh
```
The log file will be called `rtw.log`.

I also have something to recommend. I looked over the other 4 driver repos that I maintain here and I came across something that is recent. Add the following to the end of the options line while you have `8812au.conf` open:

rtw_country_code=US

Change `US` to your country code if necessary, save a reboot.

Regards

jtome7, 2022-05-13:
Hello, Nick!   I have an update:

I added the option rtw_country_code=US and now my TP-Link Archer T9-UH is
mostly working.

Curiously, I have not been able to manipulate this device by commands such
as these:

ifconfig wlp0s20u6
iwlist wlp0s20u6 scan
iwconfig wlp0s20u6 essid SBG6900AC-DC966 key s:64556212dd
dhclient -4 wlp0s20u6

After executing that set of commands, the L.E.D. blinks but the wifi
adapter does not connect.

However, when I use the Gnome3 desktop GUI for networking to connect the
wireless adapter,  it works!  What commands would the GUI be running that
are any more powerful than iwconfig & dhclient?

Since there are both 2.4 GHz and 5 GHz wifi connections available, my
TP-Link wifi adapter picks up a different one any time I reboot unless I
manually select the 5 GHz:

[image: Screenshot from 2022-05-13 12-51-47.png]

By the way, this Gnome networking GUI shows "AR9462 Wireless Network
Adapter" but I have no idea what that is unless it's a built-in that I have
never used ... nor paid any attention to ... but I do see this listing in
both ifconfig & iwconfig:

*wlp3s0*    IEEE 802.11  ESSID:off/any
          Mode:Managed  Access Point: Not-Associated   Tx-Power=19 dBm
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off

So, I'm left with two things I would like resolve if possible:

-  How to make the TP-Link wifi adapter always have the same device name.
It changes the final *digit* in this name every time I reboot:  wlp0s20u*10*
Sometimes, it ends in 5, sometimes 6, sometimes 10 and so forth.

-  How to make it always connect to the 5GHz connection automatically
without my having to do manual intervention to open the Gnome GUI, select
the network, and type in the password for the router.

If you have any ideas about these two issues, I would be very grateful to
learn about them.

The speeds for these connections show up as these:

*For 5 GHz*
Bit Rate:289 Mb/s

*For 2.4 GHz*
Bit Rate:216.7 Mb/s

Here are the examples of what iwconfig reveals depending on which network
connection it gets assigned to.  It's curious to me why one says
"802.11bgn" while the other says "802.11AC":

wlp0s20u6  IEEE 802.11bgn  ESSID:"SBG6900AC-DC966"  ***@***.***
"
          Mode:Managed  Frequency:2.462 GHz  Access Point: 5C:E3:0E:96:D7:9F
          Bit Rate:216.7 Mb/s   Sensitivity:0/0
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:****-****-****-****-****-****-****-****   Security
mode:open
          Power Management:off
          Link Quality=100/100  Signal level=57/100  Noise level=0/100
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

wlp0s20u10  IEEE 802.11AC  ESSID:"SBG6900AC-DC966-5G"
 ***@***.***"
          Mode:Managed  Frequency:5.745 GHz  Access Point: 5C:E3:0E:96:D7:A0
          Bit Rate:289 Mb/s   Sensitivity:0/0
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:****-****-****-****-****-****-****-****   Security
mode:open
          Power Management:off
          Link Quality=88/100  Signal level=42/100  Noise level=0/100
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

*As always ... Thanks for taking time to help everybody!*

*Joel*

*************************** Joel Tome*

*Greensboro, NC *

*jtome7 @ gmail.com <http://gmail.com>****************************

jtome7, 2022-05-13:
PS -

Here's what my TP-Link wifi adapter ( Realtek 8814au ) looks like ... very
stylish!

[image: TP-Link-Archer-T9UH-WIFI-Adapter.jpg]

But, I forgot to mention in that last e-mail that *ping* doesn't always
work the first time I type the command, so I wonder if the adapter goes to
sleep.  Would the *power-saving mode *cause that behavior by chance?
# Power options: ( rtw_power_mgnt )
#
# 0 = Disable power saving
# 1 = Power saving on, minPS (default)
# 2 = Power saving on, maxPS (not recommended for AP mode)
#

*Thanks!**Joel*

*************************** Joel Tome *

*jtome7 @ gmail.com <http://gmail.com>****************************

morrownr, 2022-05-14:
Hi @jtome7 

I have been ill over the last week so am behind.

> I added the option rtw_country_code=US and now my TP-Link Archer T9-UH is
mostly working.

Okay. This is progress.

> I have not been able to manipulate this device by commands such
as these:

Recommend you do not use those old, depreciated command. Use `iw` and `ip`. See links below:

https://wireless.wiki.kernel.org/en/users/documentation/iw
https://phoenixnap.com/kb/linux-ip-command-examples

Note: The docs for `iw` say to download and compile but ALL modern distros that I am aware of already have it included. You can test with the following command:

```
iw dev
```

> when I use the Gnome3 desktop GUI for networking to connect the
wireless adapter,  it works!  What commands would the GUI be running that
are any more powerful than iwconfig & dhclient?

`iw` and `ip`. See my answer above.

> -  How to make the TP-Link wifi adapter always have the same device name.
It changes the final *digit* in this name every time I reboot:  wlp0s20u*10*
Sometimes, it ends in 5, sometimes 6, sometimes 10 and so forth.

I have a pretty idea what would cause this but this is really a question to ask in the support forums for the distro you are using.

> -  How to make it always connect to the 5GHz connection automatically
without my having to do manual intervention to open the Gnome GUI, select
the network, and type in the password for the router.

With the Gnome gui there is a button called `Forget connection." You will have to forget all connects that you do not want to the system to use.

There is also an issue where some routers set both the 2.4 and 5 GHz networks to the same name. That is a really bad idea. You need to set the 5 GHz to something that is not the same as 2.4 so that you can control which you connect to. Here is a little guide I have:

https://github.com/morrownr/USB-WiFi/blob/main/home/Recommended_WiFi_Router_Settings.md

> But, I forgot to mention in that last e-mail that *ping* doesn't always
work the first time I type the command, so I wonder if the adapter goes to
sleep.  Would the *power-saving mode *cause that behavior by chance?
# Power options: ( rtw_power_mgnt )

I am not aware of any problems that are caused by the default setting. Recommend that you work on some of the things I mentioned above and report back. We can then step through in more details.

With all of that said, I feel that I need to warn you that this driver is not the greatest driver. It is the best available driver from Realtek that is available but it is not that good. If you have time and want to read information about USB WiFi adapters and Linux:

https://github.com/morrownr/USB-WiFi

Regards

jtome7, 2022-05-15:
Hello, Nick!

Sorry to hear that you weren't feeling well the other day.  Sure hope
you're on the mend and much better now!

I have great news actually regarding my learning curve using the Realtek
8814au driver.  I have gotten it working reliably on my computers, so I'm
very happy because I bought four (4) of these TP-Link AC1900 Archer T9-UH
USB wifi adapters at least two years ago ... or maybe three years ago by
now ... and I have NEVER gotten them to work on Linux until now with your
drivers, and moreover with your kind help.

I had also bought a Linksys WUSB6300 USB wifi adapter back at the same time
which required Realtek 8812au driver, so now, that Linksys WUSB6300 is also
working thanks to your 8812au driver.  It's like Christmas with gifts and
good cheer all around!

So, I'm going to try not to bother you again after seeking clarification
about this question if you don't mind commenting, please:

I read that /etc/sysconfig/network-scripts has been deprecated for use with
the current NetworkManager.service.   Is that true?

If so, where does NetworkManager store the configuration & setup
information that I entered thru the Gnome3 Desktop GUI?

And more importantly for today,  I wanted to ask where to find your *"Buy
Me a Cup of Coffee"* icon so that I can contribute a bit to enhance my
THANK YOU!  Or, do you have a PayPal account?

Where are you located on this revolving planet?  I Googled "Nick Morrow" to
see if you had a developer website, and I found quite a number of folks
named "Nick Morrow" spanning the gamut of human endeavor from professional
football player to Human Rights Campaign communications director.

If you could use me to test any drivers on Oracle Linux in the future,
please let me know!

*Joel*

*************************** Joel Tome*

*Greensboro, NC *

*jtome7 @ gmail.com <http://gmail.com>****************************

morrownr, 2022-05-16:
Hi @jtome7 

> I read that /etc/sysconfig/network-scripts has been deprecated for use with
the current NetworkManager.service.   Is that true?

I don't know. A lot of my use and work is with distros that either don't use NM or have NM deactivated. I guess you might call me a casual NM user.

> If so, where does NetworkManager store the configuration & setup
information that I entered thru the Gnome3 Desktop GUI?

See above. Not much help am I?

> And more importantly for today,  I wanted to ask where to find your *"Buy
Me a Cup of Coffee"* icon so that I can contribute a bit to enhance my
THANK YOU!  Or, do you have a PayPal account?

I don't have a PayPal account but I appreciate the offer. Maybe you could pay me back as you have time by reading the documents at the Main Menu for this site:

https://github.com/morrownr/USB-WiFi

I could some eyes on the documents so as to find spelling mistakes, grammar errors, things that don't work well for non-native speakers, dead links and suggestions for improvement. I've been working on this to make the site easier to navigate but it is a lot of work and my old eyes do not always do a good job.

> Where are you located on this revolving planet?

Northwest Arkansas, USA. I keep a very low profile on the internet. This Github site is my only presence to speak of. 

Nick

jtome7, 2022-10-11:
Latest update on my progress with Realtek 8814au driver on Oracle Linux
8.5  kernel 5.4.17

5.4.17-2136.307.3.1.el8uek.x86_64

using TP-Link AC1900 Archer T9UH

( and I was able to get my Linksys AC1200 wifi adapter working as well with
your driver for Realtek 8812au )

This script sort of works for the initial startup after the driver has been
installed and shows up in lsmod

### wifi-start   -  password has been replaced with dashes

rfkill unblock all

export INF=`iwconfig | grep wlp0s20 | cut -d" " -f1`

#export INF=`ifconfig | grep wlp0s20 | cut -d: -f1`

iwconfig $INF essid SBG6900AC-DC966-5G key s:-----------

dhclient -4 $INF

Bottom line is that I still have to go thru the Gnome3 GUI to start
networking and select the network connection.  The Linux commands in the
script just above have not been enough to initialize the wifi adapter by
themselves without depending on the Gnome3 GUI.

*Thanks!**Joel*

*************************** Joel Tome *

*jtome7 @ gmail.com <http://gmail.com>****************************

joseguzman1337, 2026-02-12:
Maintainer update: this issue has been addressed in upstream `main` at commit `b1866ce2b857a8dfe2e147e19eb8eca0a842ce18`. Resolution summary: pass under maintainer policy; guidance and diagnostics provided. For this report type, we added/updated maintainer guidance and diagnostics workflow; if still reproducible, please post full repro steps and environment details.