morrownr/8814au #68: How do I fix this?

froginperil, 2022-05-23:
At this point I have managed to thoroughly confuse myself. Stuff I learned since RH7.x has gone down the crapper.

Running install-driver.sh version 20210917
Starting installation...
Copying source files to: /usr/src/rtl8814au-5.8.5.1
Copying 8814au.conf to: /etc/modprobe.d
Error! DKMS tree already contains: rtl8814au-5.8.5.1
You cannot add the same module/version combo more than once.
An error occurred. dkms add error = 3
Please report this error.

morrownr, 2022-05-23:
> Error! DKMS tree already contains: rtl8814au-5.8.5.1

This is telling you that the driver is already installed.

I suspect the driver is now not working... right?

What you probably should do is remove the currently installed driver...

$ sudo ./remove-driver.sh

Then delete the driver directory you have and reboot.

At that point, follow the installation instructions from the beginning.

My guess is that you have upgrade your kernel and headers to a version that the driver version you had cannot handle it.

Regards

morrownr, 2022-06-13:
Status?

nu11secur1ty, 2024-02-17:
./remove-driver.sh then remove all packages manually by using apt the install it again.
Reason: You have ready-loaded the module a.k.a LKM - module, and first you must stop it before you do all procedures =)
On the next try, you broke the installation because you trying to recompile the already compiled module again :D BR