morrownr/8814au #71: Problem with creation of second interface ( AWUS1900 Alfa )

hokapefr, 2022-06-07:
I've had no problem with the driver on Ubuntu 20.04 and Debian 11, as long as I use only the first interface.

But every time when I try to create a second interface, with type "station", I had the same error :

iw dev phy0 interface add xxxx type station
command failed: No such device (-19)

But this command is successfull : 

iw dev phy0 interface add xxxx type monitor

Any ideas ?

morrownr, 2022-06-12:
Hi @hokapefr 

Sorry for the delay. This has been a busy week. I can answer your question but let me post in a second reply later today..

Regards

hokapefr, 2022-06-14:
Hi,

Let me thank you very much for the time that you invested to give me such a clear answer.

I understand now much better the difference between in-kernel and out-kernel.

To the end, I have only on to say : "Live long and prosper"

Regards

nu11secur1ty, 2024-02-17:
> > I've had no problem with the driver on Ubuntu 20.04 and Debian 11, as long as I use only the first interface.
> 
> One interface is all you are going to get. You are following a guide that following Linux Wireless Standards and Realtek out-of-kernel drivers do not.
> 
> > Any ideas ?
> 
> Yes. You have two options:
> 
> 1. Get an adapter that uses in-kernel drivers:
> 
> https://github.com/morrownr/USB-WiFi
> 
> That link is to the Main Menu of this site. It contains a lot of information. Menu item 2 shows links to a lot of adapters that works with in-kernel drivers. It also gives details of the newest in-kernel driver - mt7921u. We should have new WiFi 6 usb wifi adapters available soon. If you are in a hurry, there are many WiFi 5 and WiFi 4 adapters to pick from.
> 
> 2. Go to:
> 
> https://github.com/morrownr/Monitor_Mode
> 
> Monitor_Mode.md has information and the script may be useful for automating going into monitor mode. Many people, including myself like to operate one adapter in monitor mode while still remaining connected to the internet. I do this with my laptop. The internal mt7921 card connects to the internet and then I use the script to go into monitor mode with my ALFA AWUS036ACHM. That particular ALFA adapter has wonderful range and works using the in-kernel mt7610u driver.
> 
> Regards

Yep ;)