morrownr/8814au #72: Cannot compile on OSMC - fatal error: asm/bitsperlong.h no such file or directory

xnosek00, 2022-06-12:
Hi, 

  I have problem with compilation driver, running Rasbperry pi 4, version B on OSMC distribution. OSMC is based on Debian.
   
  I have problem with compilation drivers. Maybe you can help. 

  Here are some inofrmation about hardware:

root@osmc:/usr/src/8814au# lsusb 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter

root@osmc:~# uname -a
Linux osmc 5.10.78-5-osmc #1 SMP PREEMPT Mon Feb 28 01:18:12 UTC 2022 aarch64 GNU/Linux

And when I try compile, I have this result:

root@osmc:/usr/src/8814au# ./ARM_RPI.sh
I386 support was turned off in Makefile as planned.
Raspberry Pi OS (32 bit) support was turned on in Makefile as planned.
root@osmc:/usr/src/8814au# make
make ARCH=arm CROSS_COMPILE= -C /lib/modules/5.10.78-5-osmc/build M=/usr/src/8814au  modules
make[1]: Vstupuje se do adresáře „/usr/src/rbp464-headers-5.10.78-5-osmc“
  CC [M]  /usr/src/8814au/core/rtw_cmd.o
In file included from ./include/asm-generic/int-ll64.h:11,
                 from ./arch/arm/include/uapi/asm/types.h:5,
                 from ./include/uapi/linux/types.h:5,
                 from ./include/linux/types.h:6,
                 from /usr/src/8814au/include/basic_types.h:75,
                 from /usr/src/8814au/include/drv_types.h:26,
                 from /usr/src/8814au/core/rtw_cmd.c:17:
./include/uapi/asm-generic/int-ll64.h:12:10: fatal error: asm/bitsperlong.h: Adresář nebo soubor neexistuje                                                                                 
 #include <asm/bitsperlong.h>
          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:280: /usr/src/8814au/core/rtw_cmd.o] Chyba 1
make[1]: *** [Makefile:1825: /usr/src/8814au] Chyba 2
make[1]: Opouští se adresář „/usr/src/rbp464-headers-5.10.78-5-osmc“
make: *** [Makefile:2383: modules] Chyba 2

I can't find, what is wrong.

root@osmc:/usr/src/rbp464-headers-5.10.78-5-osmc# find . -type f -name "bitsperlong.h"
./include/uapi/asm-generic/bitsperlong.h
./include/asm-generic/bitsperlong.h
./arch/arm64/include/uapi/asm/bitsperlong.h

root@osmc:/usr/src/rbp464-headers-5.10.78-5-osmc# lsb_release -a
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 10 (buster)
Release:        10
Codename:       buster

root@osmc:/usr/src/rbp464-headers-5.10.78-5-osmc# cat /etc/os-release
PRETTY_NAME="Open Source Media Center"
NAME="OSMC"
VERSION="March 2022"
VERSION_ID="2022.03-1"
ID=osmc
ID_LIKE=debian

ANSI_COLOR="1;31"
HOME_URL="https://www.osmc.tv"
SUPPORT_URL="https://www.osmc.tv"
BUG_REPORT_URL="https://www.osmc.tv"

cat /proc/device-tree/model
Raspberry Pi 4 Model B Rev 1.5

I tried ./ARM64_RPI.sh before compiling too, but the same result:

root@osmc:/usr/src/8814au# ./ARM64_RPI.sh 
I386 support was turned off in Makefile as planned.
Raspberry Pi OS (64 bit) support was turned on in Makefile as planned.
root@osmc:/usr/src/8814au# make
make ARCH=arm CROSS_COMPILE= -C /lib/modules/5.10.78-5-osmc/build M=/usr/src/8814au  modules
make[1]: Vstupuje se do adresáře „/usr/src/rbp464-headers-5.10.78-5-osmc“
  CC [M]  /usr/src/8814au/core/rtw_cmd.o
In file included from ./include/asm-generic/int-ll64.h:11,
                 from ./arch/arm/include/uapi/asm/types.h:5,
                 from ./include/uapi/linux/types.h:5,
                 from ./include/linux/types.h:6,
                 from /usr/src/8814au/include/basic_types.h:75,
                 from /usr/src/8814au/include/drv_types.h:26,
                 from /usr/src/8814au/core/rtw_cmd.c:17:
./include/uapi/asm-generic/int-ll64.h:12:10: fatal error: asm/bitsperlong.h: Adresář nebo soubor neexistuje
 #include <asm/bitsperlong.h>
          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:280: /usr/src/8814au/core/rtw_cmd.o] Chyba 1
make[1]: *** [Makefile:1825: /usr/src/8814au] Chyba 2
make[1]: Opouští se adresář „/usr/src/rbp464-headers-5.10.78-5-osmc“
make: *** [Makefile:2383: modules] Chyba 2

Do you have any idea, what is wrong and what should I try? I was googling all day, but can't find any helpfull information. I think, there is some problem with paths and kernel headers.

xnosek00, 2022-06-12:
there was some localization in the code, so there is translation: 

"Adresář nebo soubor neexistuje" -> No such file or directory.
"Chyba 1" -> Error 1

morrownr, 2022-06-12:
Hi @xnosek00 

> Bus 001 Device 004: ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter

If the above is the adapter that you seek to install a driver for, then this repo is the wrong driver because this driver is for adapters based on the rtl8814au chipset. The driver you need is:

https://github.com/morrownr/8812au-20210629

> Linux osmc 5.10.78-5-osmc #1 SMP PREEMPT Mon Feb 28 01:18:12 UTC 2022 aarch64 GNU/Linux

I am not familiar with `osmc` so would need to investigate. Can you provide a url?

> aarch64

./ARM64_RPI.sh appears to be the best option but there is a lot I don't know about this distro.

> root@osmc:/usr/src/8814au# make

You are not following the guidance in the installation instructions. There are many ways to do this but if you are not an expert, I recommend you following the instructions and work from the following directory:

~/src/8814au

FYI: In case you have not seen it, the Main Menu for this site is located as follows and has a lot of info for usb WiFi adapters and Linux:

https://github.com/morrownr/USB-WiFi

Regards

xnosek00, 2022-06-12:
Hi @morrownr ,

 thank you for answering me. You are right, I made mistake. I really have this device:

> Bus 001 Device 003: ID 0bda:8812 Realtek Semiconductor Corp. RTL8812AU 802.11a/b/g/n/ac 2T2R DB WLAN Adapter

And I made second mistake. I installed wrong package with kernel-headers. I installed 

> apt install rbp464-headers-5.10.78-5-osmc 

but I should install thisone

> apt install rbp2-headers-5.10.78-5-osmc 

So it was my fault. After installing right kernel headers: "rbp2-headers-5.10.78-5-osmc" for OSMC, I can compile drivers.

Maybe if somebody do the same mistake, this thread helps him.

Thank you for your time. You can close this issue.

morrownr, 2022-06-12:
Glad to help. I'll mark this solved and leave it up in case it helps other.

Nick

xnosek00, 2022-06-13:
Hi, 

  I thought, that I resolved the problem, but problem still persists. And I found, that some people had the same issue with compilation. 

  As you can see issue in this repository: https://github.com/fastoe/RTL8812BU_for_Raspbian/issues/10

  I thought, that problem was with bad kernel headers. So I installed **rbp2-headers...** and it looked like, it helped. Because I finished compiling the module. But when I was trying modprobe module, there were an error:
modprobe: ERROR: could not insert '8812au': Exec format error

  OSMC is debian based and it's good for multimedia, because it has preinstalled KODI. That's why I am using it. 
 
  I found, that kernel is 64 bit and userspace is 32 bit. And somebody wrote, it worked on 32 bit kernel. But I can confirm it. 

  I found, that this can help to solve some issues (https://gitlab.com/kalilinux/packages/realtek-rtl88xxau-dkms): 
  
```
$ export ARCH=arm
$ sed -i 's/^MAKE="/MAKE="ARCH=arm\ /' dkms.conf
```

or 

```
$ export ARCH=arm64
$ sed -i 's/^MAKE="/MAKE="ARCH=arm64\ /' dkms.conf
```

but I can't still compile drivers for OSMC kernel.

Steps to reproduce:

* download iso for Rasbperry pi 4: https://osmc.tv/download/  (button disk images)
* install kernel headers "apt install rbp464-headers-5.10.78-5-osmc"
* try to compile drivers

I do not have enough knowledge about linux kernel, so I don't know, where could be problem. And it looks like, it's problem with OSMC - if other distribution works.

morrownr, 2022-06-16:
Hi @xnosek00 

> Other distros has kernel headers in one directory and that's why compiling works in other distros.
> 
> I'm only guessing, maybe it helps to find a solution for OSMC distribution.

I'm going to tell you some things that you may not want to hear but that you probably need to hear.

The open nature of Linux makes it relatively easy for individuals and companies to throw together their own distros. In these distros we see maintainers changing things having to do with package management and the dev environment because they think they have a better idea. I'm all for people inventing a better mouse trap but there are problems with using these distros. Inexperienced users will find pretty quickly that there is limited support in the form of documentation and active forums. This puts pressure on people like me that are trying to help the Linux community. Rarely a day goes by that I do not get a message in one of the repos here from a user that is trying out Linux Marvelous v1 (or you can add your own distro name). I simply do not have the time to investigate all of the various popup distros and what they have done to mess things up.

My recommendation is that you use Raspberry Pi OS until such time as you have the knowledge to handle the messes that these popup distros make.

What happens in situations like yours is that trying various things in an effort to make things work often makes a mess of your installation. I think that is part of the problem here. Setting up a sd with a fresh install of RasPiOS so that you can see a good installation of this driver would be a good thing. You will have something good to use and refer to if you want to continue playing with OSMC.

Another thing to consider is if you are going to play with non-core distros, you would likely be happier if you use a USB WiFi adapter that uses an in-kernel driver. See menu item 2 at the Main Menu for this site:

https://github.com/morrownr/USB-WiFi

Regards

xnosek00, 2022-06-16:
hi @morrownr , 

  thank you for your tips. I will try combination. I will ask in OSMC forum and I will buy another wifi adapter. It's good to know, somebody tested AP's which worked on Linux.

   My last question is about wi-fi adapter COMFAST CF-WU782AC. You wrote, you owned it. Is it possible to run wifi 2,4 and 5 GHz simultaneously? Because I need both frequencies in the same time. For now, I own 2 wifi adapters for this purpose.

morrownr, 2022-06-17:
> My last question is about wi-fi adapter COMFAST CF-WU782AC. You wrote, you owned it. Is it possible to run wifi 2,4 and 5 GHz simultaneously? Because I need both frequencies in the same time. For now, I own 2 wifi adapters for this purpose.

The adapter I have is the WU785AC but internally I think both are almost identical. 

If you are using usb wifi adapters to make a dual band ap, you will need 2 adapters. The reason is that usb wifi adapters only contain one radio. The radio on dual adapters can work on either band but only one at a time. Dual band WiFi routers/aps have two radios that function separately. Tri-band have 3 radios.

It may not be what you are trying to do but maybe Main Menu item 9 could provide some helpful info:

https://github.com/morrownr/USB-WiFi

If you want to describe your end goal, maybe I can help.

xnosek00, 2022-06-18:
Ok, thank you, so I have to buy 2 usb wifi adapters. I'm trying to create wifi router with 2,4 and 5 GHz wifi AP.  It is not possible to connect external antenna to Raspberry Pi 4 without hardware customization, so the solution should be a USB wifi card with antenna.

Internal wifi works, but it have poor signal.

I didn't ask, but it looks like, that it is possible to run your COMFAST CF-WU785AC in mode access point, isn't it?

morrownr, 2022-06-18:
> I'm trying to create wifi router with 2,4 and 5 GHz wifi AP.

I use a RasPi4B as a dual band AP so I would be glad to provide tips. If you want the RasPi to be fully capable WiFi router, I recommend you take a look at using OpenWRT as the operating system as it is designed for this use and it is professional quality:

https://openwrt.org/

I use OpenWRT and can help you get started. You might initially want to learn how to install OpenWRT to an sd and learn how to use it with the internal wifi and then you can make a decision about which usb wifi adapter(s) to use with it for greater range. OpenWRT supports several chipsets for usb wifi adapters with drivers... mt7610u and mt7612u and soon mt7921u (wifi 6).

> it is possible to run your COMFAST CF-WU785AC in mode access point, isn't it?

Yes. All of the adapters located at the follow link support ap mode with the exception of adapters based on the mt7601u chipset:

https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Adapters_that_are_supported_with_Linux_in-kernel_drivers.md

My recommendation is to learn OpenWRT on your RasPi4B using the internal radio for 2.4 Ghz and when you are comfortable, look at buying one additional card that works with OpenWRT and then later maybe a second adapter. I have used and tested many adapter with a RasPi4B and with OpenWRT so I can advise you when the time is right.

Regards

xnosek00, 2022-06-19:
@morrownr thank you for you discussion, it helped me. And I finally solved problem with compilation drivers on OSMC.

It's described in this thread: https://discourse.osmc.tv/t/wifi-driver-compilation-rpi-4b-kernel-headers-error-no-such-file-or-directory/93252/3

At first, I had to installed only **rbp464-headers-5.10.78-5-osmc** and not **rbp464-headers-sanitised-5.10.78-5-osmc**.

Second, I had to prepare chrooted environment to compile drivers by installing **aarch64-toolchain-osmc**.

```
apt install aarch64-toolchain-osmc
```

This thread helped me too: https://discourse.osmc.tv/t/trying-to-install-build-essential-arm64-on-pi4/91990/3

```
chROOT="/opt/osmc-tc/aarch64-toolchain-osmc"

mkdir -p $chROOT/home
mount --bind /home $chROOT/home #module source is in /home/osmc
mkdir -p $chROOT/usr/src #kernel source
mount --bind /usr/src $chROOT/usr/src
mkdir -p $chROOT/lib/modules #module makefiles refer to /lib/modules/$KVER/build
mount --bind /lib/modules $chROOT/lib/modules

chroot /opt/osmc-tc/aarch64-toolchain-osmc/
```

After the last command, I was in chrooted environment and I could find source drivers and compile them by tiping "make".

```
cd /home/osmc/src/
git clone https://github.com/morrownr/8812au-20210629
cd 8812au-20210629
./ARM64_RPI.sh
make -j 4
```

Exit from chrooted environment and install drivers.

```
exit
make install
```

So compilation was successfull and I could install driver. I know, it is not compeletly resolved, because I must compile drivers after each new kernel is released, but for now it is enough for me, because I can test wifi card.

I hope, this discussion will be useful for others. Thanks.

closed 2022-06-19 by xnosek00