morrownr/8814au #73: The driver is Installed completely, but not working. What shoudl I check for now?

yosungho, 2022-06-15:
Hi, thank you for sharing this awesome work.
My problem is that I checked the driver was Installed well, but my wifi is not turning on.
Could you help me what I can do for next? 

Here is my environment information.
- Ubuntu 20.04
- Kernel 5.13.0-51-generic
- Device: USB wifi adaptor (EFM ipTIME A5000UA)
- The wifi adaptor uses USB 3.0.

yosungho, 2022-06-15:
Hello @morrownr 

This is what I got. 
$ lsusb
Bus 002 Device 002: ID 2109:0812 VIA Labs, Inc. VL812 Hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 006: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 001 Device 005: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 004: ID 0603:00f5 Novatek Microelectronics Corp. 
Bus 001 Device 003: ID 2109:2812 VIA Labs, Inc. VL812 Hub
Bus 001 Device 002: ID 0bda:8813 Realtek Semiconductor Corp. RTL8814AU 802.11a/b/g/n/ac Wireless Adapter
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

morrownr, 2022-06-16:
> Bus 001 Device 002: ID 0bda:8813 Realtek Semiconductor Corp. RTL8814AU 802.11a/b/g/n/ac Wireless Adapter

That confirms this is the correct driver.

Time to do some more digging. Can you run and post the results of the following:

$ sudo uname -a; mokutil --sb-state; rfkill list all; dkms status; iw dev

yosungho, 2022-06-16:
This is the results:
$ sudo uname -a; mokutil --sb-state; rfkill list all; dkms status; iw dev
Linux AD02019911 5.13.0-51-generic #58~20.04.1-Ubuntu SMP Tue Jun 14 11:29:12 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
SecureBoot enabled
rtl8814au, 5.8.5.1, 5.13.0-51-generic, x86_64: installed

morrownr, 2022-06-16:
My next recommendation is for you to go into the bios and turn off SecureBoot so as to see if that may be causing a problem.

dkms is telling us the driver is installed but no wifi interface is showing.

yosungho, 2022-06-16:
It works after disabling SecureBoot. I appreciate you.

closed 2022-06-16 by yosungho