morrownr/8814au #74: Raspberry pi question

nospam2k, 2022-06-17:
This is more a question than an issue. Sorry if there is a better place to put this.

I'm going to use this on an Rpi 3b+ as an access point, but only 2.4ghz.

What are the best settings for modprobe and hostapd.conf so to be most stable in that environment? It's just talking with IOT devices in the local network, no internet. I am having problems now with locking up, but it may have been the version of raspian. I just installed a lite version and am reinstalling everything. Thx!

morrownr, 2022-06-18:
> What are the best settings for modprobe and hostapd.conf so to be most stable in that environment?

Run the following and post the results of the uncommented options line:

$ ./edit-options.sh

It would probably work out better if you post the contents of your hostapd.conf so I can look at it.

I do have Main Menu item 9 at this location:

https://github.com/morrownr/USB-WiFi

It has settings for this adapter and many more.

Recommend you download and burn RasPiOS 64 bit 2022-04-04.

Regards

nospam2k, 2022-06-19:
I'm using both the defaults for hostap and your driver.

hostap.conf
------------
interface=wlan0
hw_mode=g
channel=7
wmm_enabled=0
macaddr_acl=0
auth_algs=1
ignore_broadcast_ssid=0
wpa=2
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP
ssid=homectl2
wpa_passphrase=********

edit_options.sh
---------------
options 8814au rtw_drv_log_level=0 rtw_led_ctrl=1 rtw_vht_enable=1 rtw_power_mgnt=1 rtw_beamform_cap=0 rtw_switch_usb_mode=0

I can't use 64bit as I'm using a Pi 3b+

Thanks for the help!

morrownr, 2022-06-19:
> I can't use 64bit as I'm using a Pi 3b+

Um... yes you can. Go to the following site and scroll down to `Raspberry Pi OS (64-bit)` and check the list of compatible systems:

https://www.raspberrypi.com/software/operating-systems/

> edit_options.sh

> options 8814au rtw_drv_log_level=0 rtw_led_ctrl=1 rtw_vht_enable=1 rtw_power_mgnt=1 rtw_beamform_cap=0 rtw_switch_usb_mode=0

This looks good based on what you are trying to do. You might drop the `rtw_beamform_cap=0` parameter.

For `hostapd.conf`, the 2.4 Ghz example in my guide provides a good example:

https://github.com/morrownr/USB-WiFi

Click on Main Menu item 9. Recommend you only use channels 1, 6 or 11. Let me know how that goes.

Regards

nospam2k, 2022-06-19:
Wow! All this time I thought my 3b+'s were 32bit! Thanks. I'm going out of town for a few weeks, but I'll pick up on it when I get back and try your suggestions. Thanks again for you help!

nospam2k, 2022-07-04:
Ok, seems to be working ok. I install 64bit ;) I didn't wholly follow the hostapd as I'm not bridging but tried to use you example AMAP. I am having an issue with DHCPCD it seems (may not be related but thought I'd mention it.) Cold boot seems to not give a ipv4 address but ipv6 and after a minute or so, I loose access through ethernet. I disabled ipV6 and found that only a warm boot (reboot) will get an ip for wlan0 even though it's set to static. Reboot comes up fine. Not sure why that is happening or if it is even related to the 8814au. I've included the conf files.

dhcpcd.conf
```
hostname
clientid
persistent
option rapid_commit
option domain_name_servers, domain_name, domain_search, host_name
option classless_static_routes
option interface_mtu
require dhcp_server_identifier
slaac private

interface eth0
static ip_address=192.168.0.251/24
static routers=192.168.0.254
static domain_name_servers=192.168.0.254

interface wlan0
static ip_address=192.168.1.254/24

denyinterfaces eth0
denyinterfaces wlan0
noipv6rs
noipv6
```

hostapd.conf
```
interface=wlan0
hw_mode=g
channel=6
country_code=US
beacon_int=100
dtim_period=2
max_num_sta=32
fragm_threshold=2346
wmm_enabled=1
macaddr_acl=0
auth_algs=1
ignore_broadcast_ssid=0
wpa=2
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP
ssid=homectl
wpa_passphrase=...
```

morrownr, 2022-07-04:
> hostapd.conf

You are missing:

driver=nl80211

hostapd defaults to `driver=hostap` which may work but I really think you need to try the above.

Would you mind sending me a link to the routed wireless access point guide that you are using. I might look at making/testing a routed guide. I might also stumble across something that can help.

nospam2k, 2022-07-05:
I think I might have found the source of the problem.

```
/var/log/daemon.log:Jul  5 09:35:44 homectl dhcpcd[435]: wlan0: starting wpa_supplicant
/var/log/daemon.log:Jul  5 09:35:44 homectl dhcpcd-run-hooks[456]: wlan0: starting wpa_supplicant
/var/log/daemon.log:Jul  5 09:35:45 homectl hostapd[436]: wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
/var/log/daemon.log:Jul  5 09:35:45 homectl dnsmasq[484]: warning: interface wlan0 does not currently exist
/var/log/daemon.log:Jul  5 09:35:45 homectl dhcpcd[347]: wlan0: connected to Access Point `'
/var/log/daemon.log:Jul  5 09:35:46 homectl dhcpcd[347]: wlan0: waiting for carrier
/var/log/daemon.log:Jul  5 09:35:46 homectl dhcpcd[347]: wlan0: carrier acquired
/var/log/daemon.log:Jul  5 09:35:46 homectl dhcpcd[347]: wlan0: probing address 192.168.1.254/24
/var/log/daemon.log:Jul  5 09:35:46 homectl dhcpcd[347]: wlan0: carrier lost
/var/log/daemon.log:Jul  5 09:35:47 homectl dhcpcd[347]: wlan0: carrier acquired
/var/log/daemon.log:Jul  5 09:35:47 homectl dhcpcd[347]: wlan0: probing address 192.168.1.254/24
/var/log/daemon.log:Jul  5 09:35:47 homectl dhcpcd[347]: wlan0: carrier lost
/var/log/daemon.log:Jul  5 09:35:47 homectl hostapd: wlan0: STA 2c:3a:e8:3b:6d:76 IEEE 802.11: associated
/var/log/daemon.log:Jul  5 09:35:47 homectl hostapd: wlan0: STA 2c:3a:e8:3b:6d:76 RADIUS: starting accounting session ED9509C21C2A11EB
/var/log/daemon.log:Jul  5 09:35:47 homectl hostapd: wlan0: STA 2c:3a:e8:3b:6d:76 WPA: pairwise key handshake completed (RSN)
/var/log/daemon.log:Jul  5 09:35:47 homectl dnsmasq-dhcp[484]: DHCP packet received on wlan0 which has no address
```

It looks like I need some kind of delay to wait for the interface to come up.

I'm not actually routing because the rPi is functioning as an interface between IOT and node-red using MQTT.

nospam2k, 2022-07-05:
Ok further testing. Cold boot doesn't get an ip and ends up locking up the rPi after a few minutes so disabling ipV6 didn't hlpe. No kbd response on the pi. If I quickly login in and reboot, I get an ip.

morrownr, 2022-07-05:
What are you seeing for wifi interfaces?

$ iw dev

nospam2k, 2022-07-05:
iw dev
phy#0
	Interface wlan0
		ifindex 3
		wdev 0x1
		addr 00:13:ef:4f:05:92
		ssid homectl
		type AP
		txpower 20.00 dBm

morrownr, 2022-07-05:
Have you turned off the RasPi3B+ internal wifi via config.txt?

The interface looks about like what I would expect so... onward.

Let me see... the RasPi3B+ only has USB2 ports. correct? 

I was rereading the thread trying to figure out what adapter you have. For now I'll assume it is based on the 8814au chipset since that is where we are. I have tested the power usage of a couple of adapters based on the 8814au chipset and this chipset can really pull some power. USB2 should be able to provide up to 500 mA per port per spec but the RasPi 3B, 3B+ and 4B USB subsystem can only supply up to 1200 mA total. My testing has shown 8814au based adapters using up to around 800 mA which is well beyond the capacity of USB2 so I've never tested in a USB2 port. Maybe there is circuitry to limit power usage if in a USB2 port? I don't know. 

I know I am wandering here but maybe we can get to the bottom of this. With all of this said, I do not recommend users use 8814au based adapters with RasPis due to power issues. Good adapters for use with RasPis are the ones based on the Mediatek chipsets as they are very good at keeping power usage low and they have in-kernel drivers... let me point you to the Main Menu for this site:

https://github.com/morrownr/USB-WiFi

Menu items 1 and 2 might be of interest.

Now, lastly, are you using a powered usb hub and what other usb devices do you have plugged in?

nospam2k, 2022-07-05:
I have disabled both Wifi and BT from config.txt.

Yes the 3b+ only had USB2

Yes, I am using 8814au

I have a USB z-wave dongle along with the 8814. I have a keyboard plugged in now for testing. No powered USB hub.

I looked at you USB-WiFi list. A lot there! Bottom line, what do you suggest for a rPi 3b+ long range 2.4ghz cheap USB wifi? As I said, I'm only dealing with IOT so no 5ghz needed. Just long range.

morrownr, 2022-07-05:
> Bottom line, what do you suggest for a rPi 3b+ long range 2.4ghz cheap USB wifi?

So, cheap, long range and single band:

https://www.amazon.com/Panda-150Mbps-Wireless-Adapter-Antennas/dp/B004AC6X0K

I have one of those. I use it currently as the 2.4 Ghz part of my dual band RasPi4B Wireless Access Point. It has pretty good range, runs 24/7 without a hiccup and is pretty cheap. It is plug and play as the driver for it is in-kernel and has been for a few years. I can give the specific settings for hostapd.conf when you are ready if you go this route.

If you look at Main Menu item 3 (I think), this adapter is rated as a Superstar, however, the rt3070 chipset went end of life last year so I expect makers to start running out at some point so this adapter could cease to be available at some point this year.

If you don't like it, tell me what you don't like and I'll try to suggest something else. Remember that N150 adapter give you around 72 Mbp/s and since this is half duplex technology, you can expect a throughput of somewhere are 32-36 Mbp/s second. Given that most IoT device pass small amounts of data around, that can probably handle a lot of them.

Regards

nospam2k, 2022-07-06:
I guess the real question (I don't really know that much about wifi chips) is whether it would be worth it for me to just modify the internal rPi 3b+ wifi and add an antenna or should I use a usb. I am always adding 8266 based IOT devices so there is a real possibility I may have up to 200 devices. They don't talk much but I'm not that familiar with client connectivity to know if there is a limit on client devices.

morrownr, 2022-07-06:
> there is a real possibility I may have up to 200 devices.

Ouch. In that case, I would look at doing both. Use the internal radio to do clients that are close and the usb adapter to handle clients that are farther away. hostapd can handle 2 adapters. One could be on channel 1 and the other on channel 11 for example. I can show you how.

nospam2k, 2022-07-06:
I'm going to give this a try:

https://www.amazon.com/HIGHROCK-Enhancer-Female-Charge-Extension/dp/B00NIGO4NM/ref=asc_df_B00NIGO4NM/?tag=hyprod-20&linkCode=df0&hvadid=309807187084&hvpos=&hvnetw=g&hvrand=9879222032472747408&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=9031974&hvtargid=pla-761647499952&psc=1

morrownr, 2022-07-07:
It is not clear to me what that is going to do for you.

Here is another single band adapter that might work for you:

https://www.amazon.com/EDUP-LOVE-Wireless-Notebook-Raspberry/dp/B09NNMJQVL/ref=cm_cr_arp_d_product_top?ie=UTF8

Not sure of the range on this one as I don't own it.

nospam2k, 2022-07-07:
It claims the cable will allow to get external power to the usb device without a hub. My wall adapter has 2 USB power ports. Seeing if the issue is the USB power. I'm ok with your first adapter. Looks like better distance from the antennas(?)

This is the device I have now:
https://www.amazon.com/gp/product/B086ZQB2M8/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

morrownr, 2022-07-07:
> It claims the cable will allow to get external power to the usb device without a hub. My wall adapter has 2 USB power ports. Seeing if the issue is the USB power. I'm ok with your first adapter. Looks like better distance from the antennas(?)

Interesting. First time I have seen something like that. Be aware that RasPis have a history of having problems with external powered usb hubs. It seems to be a problem with the backfeeding of power into the Pis. Certain hubs work and most don't. I have no idea about this device because I did not know such a device existed.

> This is the device I have now: https://www.amazon.com/gp/product/B086ZQB2M8/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

That is a good looking adapter. I wish Realtek provided better support for that chipset.

morrownr, 2022-07-11:
> So no crashes since I put that cable on.

Cool. 

> But... There does seem to be a strange random problem with the wiFi getting an ip address. Usually a reboot is enough. Once it gets an ip it's fine.

Since this thread is getting long at this point, let me see if I am still on the right path:

You are using a RasPi3B+ with RasPiOS, latest, 64 bit. 
You are using hostapd but it is not a bridged setup like in my guide.
You are using a 8814au based adapter and using 2.4 GHz.

This kinda sounds like it might be a double dhcp problem. Is the RasPi using its own network segment?
How is the main router setup and how is the RasPi connecting to it?

In my Bridged Wireless Access Point guide, I recommend OpenWRT for users wanted a routed setup.

Here is my guide:

https://github.com/morrownr/USB-WiFi/blob/main/home/AP_Mode/Bridged_Wireless_Access_Point.md

Here is a link to setting up OpenWRT on a RasPi:

https://openwrt.org/toh/raspberry_pi_foundation/raspberry_pi

Ah heck. This driver won't work on OpenWRT. Sorry about that.

As long as I am talking about OpenWRT, it drivers for almost all of the adapters for the in-kernel adapters as shown:

https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Adapters_that_are_supported_with_Linux_in-kernel_drivers.md

Since OpenWRT is designed to be as small as possible, it does not include the drivers for the usb wifi adapters in a base install but you can quickly install them. I run a mt7610u based adapter plugged into my main router via usb port. The adapter is the ALFA ACHM.

At this point I not sure if I am helping or not.

nospam2k, 2022-07-11:
@morrownr - You have been INCREDIBLY helpful. More than I have ever experienced. Seriously!

Since I'm using node-red on this pi, I don't think openwrt would be a good solution.

I think you may be on to something re the double DHCP. The main router is running dd-wrt. To clarify, I'm neither routing nor bridging. I basically just want an "external" ip (eth0) to the device to access node-red but none of wlan devices have access to the outside world.

Here is what's running on the pi minus all the bracketed stuff and kworkers. Let me know if you see anything.

```
 323 ?        Ss     0:54 avahi-daemon: running [homectl.local]
    325 ?        Ss     0:02 /usr/sbin/cron -f
    327 ?        Ss     0:01 /usr/bin/dbus-daemon --system --address=systemd: --nofork --nopidfile --systemd-activation --syslog-only
    341 ?        Ssl   69:27 node-red
    342 ?        S      0:00 avahi-daemon: chroot helper
    367 ?        Ssl    0:02 /usr/sbin/rsyslogd -n -iNONE
    384 ?        Ss     0:01 /lib/systemd/systemd-logind
    395 ?        Ss     0:04 /usr/sbin/thd --triggers /etc/triggerhappy/triggers.d/ --socket /run/thd.socket --user nobody --deviceglob /dev/input/event*
    398 ?        Ss     0:02 /sbin/wpa_supplicant -u -s -O /run/wpa_supplicant
    430 ?        SLsl   0:02 /usr/sbin/rngd -r /dev/hwrng
    467 ?        Ss     4:38 /usr/sbin/mosquitto -c /etc/mosquitto/mosquitto.conf
    468 tty1     Ss+    0:00 /sbin/agetty -o -p -- \u --noclear tty1 linux
    470 ?        Ss+    0:00 /sbin/agetty -o -p -- \u --keep-baud 115200,57600,38400,9600 ttyAMA0 vt220
    472 ?        Ssl    0:52 /usr/sbin/ntpd -p /var/run/ntpd.pid -g -u 109:114
    484 ?        S     44:15 [RTW_CMD_THREAD]
    486 ?        S      0:29 /usr/sbin/dnsmasq -x /run/dnsmasq/dnsmasq.pid -u dnsmasq -r /run/dnsmasq/resolv.conf -7 /etc/dnsmasq.d,.dpkg-dist,.dpkg-old,.dpkg-new --local-service --trust-anchor=.,20326,8,2,e06d44b80b8f1d39a95c0b0d7c65d08458e880409bbc683457104237c7f8ec8d
    487 ?        Ss     0:11 /usr/sbin/hostapd -B -P /run/hostapd.pid -B /etc/hostapd/hostapd.conf
    493 ?        Ss     0:04 wpa_supplicant -B -c/etc/wpa_supplicant/wpa_supplicant.conf -iwlan0
    494 ?        Ss     0:00 sshd: /usr/sbin/sshd -D [listener] 0 of 10-100 startups
    610 ?        Ss     0:23 /usr/sbin/dhcpcd -w -q
   6907 ?        Ss     0:00 sshd: pi [priv]
   6910 ?        Ss     0:00 /lib/systemd/systemd --user
```

nospam2k, 2022-07-11:
I thought I would add (in case it is of interest to you) https://github.com/morrownr/8814au/issues/61. I actually did get some success with openwrt and your driver by a few hacks contained in the thread you referred me to. They really are kludges but should it be of interest...

morrownr, 2022-07-11:
> To clarify, I'm neither routing nor bridging.

Well, in my mind, that statement explains the following quote:

> none of wlan devices have access to the outside world.

Unless I am badly mistaken, you will need routing or bridging or the outside is not going to happen.

> node-red

I have no idea what that is. I only have a fuzzy idea of what you are trying to do.

> I think you may be on to something re the double DHCP.

If both your router and the RasPi are both on the segment (example: 192.168.1.1 to 192.168.1.254) and they are both providing DHCP, then you will have a conflict. To deconflict you could set DHCP on the router to be limited to 192.168.1.3 to 192.168.1.99 and limit DHCP in the RasPi to 182.168.1.100 to 192.168.1.254. That way there is no overlap and no conflict. Or not have a second DHCP.

Yes, this networking stuff is more fun than a barrel of monkeys.

nospam2k, 2022-07-12:
Lol... Sorry text is so 2 dimensional. I don't want the devices having outside access. My only issue is that sometimes wlan0 won't get a static ip. Sometimes I just get no ip when the rPi is rebooted.

Normally, when working, the wlan0 comes up static as 192.168.1.254 and eth0 comes up as 192.168.0.251. So, I shouldn't be getting requests for DHCP at all IFAIK.

Here is /etc/dhcpcd.conf
```
interface eth0
static ip_address=192.168.0.251/24
static routers=192.168.0.254
static domain_name_servers=192.168.0.254

interface wlan0
static ip_address=192.168.1.254/24

denyinterfaces eth0
denyinterfaces wlan0
noipv6rs
noipv6
```

Here is /etc/wpa_supplicant/wpa_supplicant.conf
```
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
```

Do I even need wpa_supplicant using hostapd?

morrownr, 2022-07-12:
> Lol... Sorry text is so 2 dimensional. 

And then it would appear that I enter periods of deep slumber at times.

I try to help a lot of folks. Things get cloudy sometimes. I regularly have to switch between managed mode and AP mode and monitor mode and others but those are the big 3. My mind ends up being mush.

> Do I even need wpa_supplicant using hostapd?

No, just leave it like it is when you do a fresh install. That looks like what you have.

> ```
interface eth0
static ip_address=192.168.0.251/24
static routers=192.168.0.254
static domain_name_servers=192.168.0.254

interface wlan0
static ip_address=192.168.1.254/24
```

Okay, so you are setting a fixed ip address for wlan0 and eth0

After going back through this thread, I see that you have installed dnsmasq. Can you give me a link to the guide you are using? I need to more fully understand your setup. Your ip address for eth0 and the ip address for wlan0 are in different network segments so there has to be routing for them to talk to each other. dnsmasq can do this.

I have a Pi3B, a 8814au based adapter and a couple of extra sd cards so when I have time later this week, I can try to duplicate your setup which will allow me to better see what is going on.

nospam2k, 2022-07-12:
My setup for dnsmasq is just the default install. I'm only using it for passing DHCP and static ip's to the wlan0 side.

To attempt to clarify more, there is no routing between wlan0 and eth0. Node-red and Mosquitto are running as services listening on both interfaces. Node-red is a kind of web based block wiring nodejs interface. So I'm just using port 443 and 1883. All my Internet of Things devices use wlan0 to talk MQTT to mosquitto and node-red handles the logic. In a sense Mosquitto and Node-Red are the routers as any commands in or out go through them. Ex. I have an Alexa script running at Amazon. Alexa sends a request to my node-red listener on the rPi, node-red turns the request into an MQTT request that goes out through wlan0 to the appropriate device.

This is /etc/dnsmasq.conf
```
interface=wlan0
dhcp-range=192.168.1.1,192.168.1.253,255.255.255.0,24h
```

I also have dhcp-host (static assigned) lines but nothing more.

morrownr, 2022-07-12:
> In a sense Mosquitto and Node-Red are the routers as any commands in or out go through them.

Okay. That explains a lot. I need to step back and quit pretending like I know what I am doing.