morrownr/8814au #76: Linux 5.19 compilation error

IsaacMvmv, 2022-07-02:
I am the same guy who opened the "Linux 5.18 compilation error" issue, so here we go

Error:
```
DKMS make.log for rtl8814au-5.8.5.1 for kernel 5.19.0-rc4-262-tkg-cfs-llvm (x86_64)
sáb 02 jul 2022 14:56:13 CEST
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.19.0-rc4-262-tkg-cfs-llvm/build M=/var/lib/dkms/rtl8814au/5.8.5.1/build  modules
make[1]: se entra en el directorio '/usr/lib/modules/5.19.0-rc4-262-tkg-cfs-llvm/build'
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_chplan.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_recv.o
In file included from /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_ioctl_query.c:17:
In file included from /var/lib/dkms/rtl8814au/5.8.5.1/build/include/drv_types.h:78:
/var/lib/dkms/rtl8814au/5.8.5.1/build/include/rtw_recv.h:773:52: warning: variable 'buf_star' is uninitialized when used here [-Wuninitialized]
        precv_frame = rxmem_to_recvframe((unsigned char *)buf_star);
                                                          ^~~~~~~~
/var/lib/dkms/rtl8814au/5.8.5.1/build/include/rtw_recv.h:771:14: note: initialize the variable 'buf_star' to silence this warning
        u8 *buf_star;
                    ^
                     = NULL
1 warning generated.
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_sta_mgt.o
In file included from /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_chplan.c:17:
In file included from /var/lib/dkms/rtl8814au/5.8.5.1/build/include/drv_types.h:78:
/var/lib/dkms/rtl8814au/5.8.5.1/build/include/rtw_recv.h:773:52: warning: variable 'buf_star' is uninitialized when used here [-Wuninitialized]
        precv_frame = rxmem_to_recvframe((unsigned char *)buf_star);
                                                          ^~~~~~~~
/var/lib/dkms/rtl8814au/5.8.5.1/build/include/rtw_recv.h:771:14: note: initialize the variable 'buf_star' to silence this warning
        u8 *buf_star;
                    ^
                     = NULL
1 warning generated.
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_ap.o
In file included from /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_ioctl_set.c:17:
In file included from /var/lib/dkms/rtl8814au/5.8.5.1/build/include/drv_types.h:78:
/var/lib/dkms/rtl8814au/5.8.5.1/build/include/rtw_recv.h:773:52: warning: variable 'buf_star' is uninitialized when used here [-Wuninitialized]
        precv_frame = rxmem_to_recvframe((unsigned char *)buf_star);
                                                          ^~~~~~~~
/var/lib/dkms/rtl8814au/5.8.5.1/build/include/rtw_recv.h:771:14: note: initialize the variable 'buf_star' to silence this warning
        u8 *buf_star;
                    ^
                     = NULL
1 warning generated.
In file included from /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_pwrctrl.c:17:
In file included from /var/lib/dkms/rtl8814au/5.8.5.1/build/include/drv_types.h:78:
/var/lib/dkms/rtl8814au/5.8.5.1/build/include/rtw_recv.h:773:52: warning: variable 'buf_star' is uninitialized when used here [-Wuninitialized]
        precv_frame = rxmem_to_recvframe((unsigned char *)buf_star);
                                                          ^~~~~~~~
/var/lib/dkms/rtl8814au/5.8.5.1/build/include/rtw_recv.h:771:14: note: initialize the variable 'buf_star' to silence this warning
        u8 *buf_star;
                    ^
                     = NULL
1 warning generated.
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/mesh/rtw_mesh.o
In file included from /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_vht.c:17:
In file included from /var/lib/dkms/rtl8814au/5.8.5.1/build/include/drv_types.h:78:
/var/lib/dkms/rtl8814au/5.8.5.1/build/include/rtw_recv.h:773:52: warning: variable 'buf_star' is uninitialized when used here [-Wuninitialized]
[... 565 lines cut ...]
/var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/ioctl_mp.c:2684:16: warning: equality comparison with extraneous parentheses [-Wparentheses-equality]
                        if ((tmp[1] == NULL)) {
                             ~~~~~~~^~~~~~~
/var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/ioctl_mp.c:2684:16: note: remove extraneous parentheses around the comparison to silence this warning
                        if ((tmp[1] == NULL)) {
                            ~       ^      ~
/var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/ioctl_mp.c:2684:16: note: use '=' to turn this equality comparison into an assignment
                        if ((tmp[1] == NULL)) {
                                    ^~
                                    =
2 warnings generated.
In file included from /var/lib/dkms/rtl8814au/5.8.5.1/build/hal/hal_intf.c:18:
In file included from /var/lib/dkms/rtl8814au/5.8.5.1/build/include/drv_types.h:78:
/var/lib/dkms/rtl8814au/5.8.5.1/build/include/rtw_recv.h:773:52: warning: variable 'buf_star' is uninitialized when used here [-Wuninitialized]
        precv_frame = rxmem_to_recvframe((unsigned char *)buf_star);
                                                          ^~~~~~~~
/var/lib/dkms/rtl8814au/5.8.5.1/build/include/rtw_recv.h:771:14: note: initialize the variable 'buf_star' to silence this warning
        u8 *buf_star;
                    ^
                     = NULL
1 warning generated.
make[1]: *** [Makefile:1843: /var/lib/dkms/rtl8814au/5.8.5.1/build] Error 2
```

Thanks in advance

morrownr, 2022-07-03:
Let me see what I can figure out. May be a few days until I can get to it. Ping me in 4-5 days if you haven't heard anything from me.

IsaacMvmv, 2022-07-03:
Okay, np

morrownr, 2022-07-03:
Hey @IsaacMvmv 

I ended up with some unexpected extra time this afternoon. Here is where I am:

A couple of weeks ago I had an unexpected component failure in my main dev box. Component has been replaced and hardware checks out. It does take time to fully regenerate the entire dev environment. I didn't lose anything but it was time for me to move to a more modern base system... that is what takes the most time. I am now using Kubuntu 22.04. I have enough of the system in place to test and here is what I am seeing:

- I installed `kernel 5.19.0-rc4`
- I am using gcc 11.2.
- The system is an older X86_64, Intel i7 quad core

I am seeing some cosmetic issues (warnings) that I need to clean up. This is mostly due to jumping from gcc 9.3 to 11.2. I'll start working on those issues but the driver compiled and is working. I am not seeing most of the issues from the log you posted so we need to determine why the difference in what we are seeing. Can you check and post the version of gcc that you are using and anything else that you can think of that might be contributing?

Regards

morrownr, 2022-07-03:
Some testing in managed mode:

```
[  5]  15.00-16.00  sec  77.5 MBytes   650 Mbits/sec    0   1.85 MBytes       
[  5]  16.00-17.00  sec  75.0 MBytes   629 Mbits/sec    0   1.85 MBytes       
[  5]  17.00-18.00  sec  73.8 MBytes   619 Mbits/sec    0   1.85 MBytes       
[  5]  18.00-19.00  sec  76.2 MBytes   640 Mbits/sec    0   1.85 MBytes       
[  5]  19.00-20.00  sec  72.5 MBytes   608 Mbits/sec    0   1.85 MBytes       
[  5]  20.00-21.00  sec  76.2 MBytes   640 Mbits/sec    0   1.85 MBytes       
[  5]  21.00-22.00  sec  76.2 MBytes   640 Mbits/sec    0   1.85 MBytes       
[  5]  22.00-23.00  sec  75.0 MBytes   629 Mbits/sec    0   1.85 MBytes       
[  5]  23.00-24.00  sec  77.5 MBytes   650 Mbits/sec    0   1.85 MBytes       
[  5]  24.00-25.00  sec  76.2 MBytes   640 Mbits/sec    0   1.85 MBytes       
[  5]  25.00-26.00  sec  77.5 MBytes   650 Mbits/sec    0   1.85 MBytes       
```

It seems to be working well. This is an old driver that is a challenge to maintain but I am seeing something different than you are so I guess stepping through things to figure out what is different is what we will have to do.

Regards

IsaacMvmv, 2022-07-03:
Interesting, I will see what I did wrong

morrownr, 2022-07-03:
You did not necessarily do anything wrong. Something is different between our systems. The first thing to look at is what version of gcc you are using.

morrownr, 2022-07-03:
FYI: You might want to run `git pull` to update your download as I have been committing several changes today. Most are documentation changes but there are also some things that will reduce the compiler warnings you are seeing.

I am down to the point that I have very few compiler warnings left but they will require some code changes.

Any progress on your end?

Denis-AAA, 2022-10-16:
Tried all the methods described. At first it refused to work. Then I updated the Kali kernel to 5.19 (it was 5.18) and managed to start it once, saw the device, switched to monitor mode and got the HandShake of the test router. In the morning, the ability to turn on the monitor mode disappeared. Decided to reinstall everything from the beginning. But the driver refuses to install. Installing with i`make' gives the following error:
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/5.19.0-kali2-amd64/build M=/home/kali/src/8814au  modules
make[1]: вход в каталог «/usr/src/linux-headers-5.19.0-kali2-amd64»

  ERROR: Kernel configuration is invalid.
         include/generated/autoconf.h or include/config/auto.conf are missing.
         Run 'make oldconfig && make prepare' on kernel src to fix it.

make[1]: *** [/usr/src/linux-headers-5.19.0-kali2-common/Makefile:750: include/config/auto.conf] Ошибка 1
make[1]: выход из каталога «/usr/src/linux-headers-5.19.0-kali2-amd64»
make: *** [Makefile:2384: modules] Error 2

Denis-AAA, 2022-10-16:
When installing via the .\install-driver.sh method, an error occurs:

Running install-driver.sh version 20220705
5.19.0-kali2-amd64
x86_64
Starting installation...
Copying source files to: /usr/src/rtl8814au-5.8.5.1
Copying 8814au.conf to: /etc/modprobe.d
Creating symlink /var/lib/dkms/rtl8814au/5.8.5.1/source -> /usr/src/rtl8814au-5.8.5.1
Sign command: /usr/lib/linux-kbuild-5.19/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
'make' -j2 KVER=5.19.0-kali2-amd64 KSRC=/lib/modules/5.19.0-kali2-amd64/build...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.19.0-kali2-amd64 (x86_64)
Consult /var/lib/dkms/rtl8814au/5.8.5.1/build/make.log for more information.
An error occurred. dkms build error = 10
Please report this error.
You will need to run the following before reattempting installation.
$ sudo ./remove-driver.sh

Linux kali 5.19
TP-Link archer T9UH (realtek rlt 8814 ua)