morrownr/8814au #77: More than 32 clients => kernel: [  rtw_get_camid pairwise key with MACADDR id:0 no room

DennisKuhn, 2022-07-03:
Hello,
thanks a lot for providing this great driver with amazing install instructions, which worked straight away.

I seem to be hitting a hard limit of 32 (or so) clients. When another client connects I get in `/var/log/syslog`:
```
Jul  4 10:50:37 pi kernel: [  804.147792] RTW: recv eapol packet 4/4
Jul  4 10:50:37 pi kernel: [  804.149113] RTW: rtw_get_camid(wlan1) pairwise key with 10:d5:61:a7:b5:3b id:0 no room
Jul  4 10:50:37 pi kernel: [  804.149184] ------------[ cut here ]------------
Jul  4 10:50:37 pi kernel: [  804.149188] WARNING: CPU: 3 PID: 592 at /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_wlan_util.c:1204 rtw_get_camid+0x144/0x1b8 [8814au]
Jul  4 10:50:37 pi kernel: [  804.149341] Modules linked in: nf_conntrack_netlink br_netfilter bridge rfcomm cmac algif_hash aes_arm64 algif_skcipher af_alg bnep hci_uart btbcm bluetooth ecdh_generic ecc overlay 8021q garp stp llc nft_chain_nat xt_MASQUERADE nf_nat xt_addrtype nft_counter xt_conntrack nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nft_compat nf_tables nfnetlink 8814au(O) snd_soc_hdmi_codec brcmfmac brcmutil cfg80211 rfkill v3d gpu_sched bcm2835_codec(C) bcm2835_isp(C) bcm2835_v4l2(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_dma_contig videobuf2_vmalloc raspberrypi_hwmon videobuf2_memops videobuf2_v4l2 vc4 videobuf2_common i2c_brcmstb cec drm_kms_helper videodev snd_soc_core mc snd_bcm2835(C) vc_sm_cma(C) rpivid_mem snd_compress snd_pcm_dmaengine snd_pcm snd_timer snd syscopyarea sysfillrect sysimgblt fb_sys_fops gpio_fan nvmem_rmem uio_pdrv_genirq uio i2c_dev drm drm_panel_orientation_quirks backlight fuse ip_tables x_tables ipv6
Jul  4 10:50:37 pi kernel: [  804.149490] CPU: 3 PID: 592 Comm: RTW_CMD_THREAD Tainted: G        WC O      5.15.32-v8+ #1538
Jul  4 10:50:37 pi kernel: [  804.149497] Hardware name: Raspberry Pi 4 Model B Rev 1.5 (DT)
Jul  4 10:50:37 pi kernel: [  804.149502] pstate: 80000005 (Nzcv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
Jul  4 10:50:37 pi kernel: [  804.149508] pc : rtw_get_camid+0x144/0x1b8 [8814au]
Jul  4 10:50:37 pi kernel: [  804.149594] lr : rtw_get_camid+0x1ac/0x1b8 [8814au]
Jul  4 10:50:37 pi kernel: [  804.149676] sp : ffffffc00a933c90
Jul  4 10:50:37 pi kernel: [  804.149679] x29: ffffffc00a933cb0 x28: ffffff8109eb9a00 x27: ffffffc009e2a104
Jul  4 10:50:37 pi kernel: [  804.149688] x26: 0000000000000000 x25: 0000000000000000 x24: ffffff8108876460
Jul  4 10:50:37 pi kernel: [  804.149696] x23: ffffff8108876000 x22: ffffffc009e29000 x21: ffffffc00afa24e1
Jul  4 10:50:37 pi kernel: [  804.149704] x20: ffffffc009e29000 x19: ffffffc00afa24a8 x18: 0000000000000010
Jul  4 10:50:37 pi kernel: [  804.149712] x17: 64692062333a3562 x16: 3a37613a31363a35 x15: 643a303120687469
Jul  4 10:50:37 pi kernel: [  804.149719] x14: 772079656b206573 x13: 6d6f6f72206f6e20 x12: ffffffc0093a6670
Jul  4 10:50:37 pi kernel: [  804.149727] x11: 0000000000000003 x10: ffffffc00938e630 x9 : ffffffc0080ec768
Jul  4 10:50:37 pi kernel: [  804.149734] x8 : 0000000000017fe8 x7 : c0000000ffffefff x6 : ffffff81fefca908
Jul  4 10:50:37 pi kernel: [  804.149741] x5 : ffffffc1f5f8a000 x4 : 0000000000000203 x3 : 0000000000000202
Jul  4 10:50:37 pi kernel: [  804.149749] x2 : fadb0b572cca5b00 x1 : 0000000000000000 x0 : 000000000000004a
Jul  4 10:50:37 pi kernel: [  804.149757] Call trace:
Jul  4 10:50:37 pi kernel: [  804.149760]  rtw_get_camid+0x144/0x1b8 [8814au]
Jul  4 10:50:37 pi kernel: [  804.149843]  rtw_camid_alloc+0x78/0x150 [8814au]
Jul  4 10:50:37 pi kernel: [  804.149924]  set_stakey_hdl+0x84/0x380 [8814au]
Jul  4 10:50:37 pi kernel: [  804.150006]  rtw_cmd_thread+0x448/0x5f4 [8814au]
Jul  4 10:50:37 pi kernel: [  804.150087]  kthread+0x140/0x158
Jul  4 10:50:37 pi kernel: [  804.150095]  ret_from_fork+0x10/0x20
Jul  4 10:50:37 pi kernel: [  804.150102] ---[ end trace bc776dd8b1623b88 ]---
```

I am using this as AP for my Home Automation devices and sensors. I was hoping to have at least 64 clients, rather more.

I wonder if it is possible to increase the *room* ?

Cheers,
Dennis

morrownr, 2022-07-04:
Dennis,

> I wonder if it is possible to increase the room ?

Good question. It sounds like a mystery to solve. The first thing that popped into my mind was how am I going to duplicate this? I have a RPi4B et al but how can i come up with > 32 clients? 

Are you using 2.4 or 5 Ghz?

DennisKuhn, 2022-07-04:
All the clients are 2.4 GHz, my `hostapd.conf` is et to:
```
hw_mode=b
```

DennisKuhn, 2022-07-04:
I got some light bulbs, plugs, presence sensors, environment sensors, fans, ... ;-) I am happy to provide any help with the testing :-)

morrownr, 2022-07-04:
>  I am happy to provide any help with the testing :-)

It will basically have to be you.

I'm getting a picture of the situation. You have a lot of low throughput clients. I could use more information:

- Would you mind posting your hostapd.conf? 

- Would you mind posting the first options line in your 8814au.conf?  `$ sudo ./edit-options.sh`

I guess need to explain where I am going with this. I'd really like to go over your setup to make sure the issue really is coming from the driver. It isn't that I don't trust your skills but I've seen external factors blow things up before and I would also like to get a better feeling for the overall need for things like range and whether using 2 adapters might work well. Do you have much 2.4 Ghz traffic in your area?

Also, something to point out to you:

https://github.com/morrownr/USB-WiFi

That is the site Main Menu. Menu item 9 is an extensive guide to do basically what it sounds like you are doing as far as setting up an AP with a RasPi.

DennisKuhn, 2022-07-04:
hostap.d
```
interface=wlan1
country_code=NZ
ssid=Thor-Things
hw_mode=b
channel=6
macaddr_acl=0
auth_algs=2
ignore_broadcast_ssid=0

wpa=3
wpa_passphrase=MyPa$$Phr@$e
wpa_key_mgmt=WPA-PSK
wpa_pairwise=CCMP
```

8814au options:
```
options 8814au rtw_drv_log_level=3 rtw_led_ctrl=1 rtw_vht_enable=1 rtw_power_mgnt=1 rtw_switch_usb_mode=1
```

You every reason to have little faith in my skill ;-) This is the first time I've used linux in over 10 years ;-)

My channel 6 is pushing anyone else to the sides ;-) I am living in a flat, so the Wifi Card nicely covers the whole area.

DennisKuhn, 2022-07-04:
I did not choose the bridged AP, because I hope that it would make it easier to integrate "Tuya local". I am by no means as far yet. If you think testing it in bridge mode could make a difference, then I am happy to give bridge-mode a go.

morrownr, 2022-07-04:
>
> interface=wlan1
> country_code=NZ
> ssid=Thor-Things
> hw_mode=b

I don't have a good feeling about this setting. You may be pushing several things into 80211b code that hasn't been well tested in 20 years. I'd feel a lot better with the following:

hw_mode=g

> channel=6
> macaddr_acl=0
> auth_algs=2

I'll recommend:

auth_algs=1

> ignore_broadcast_ssid=0
> 
> wpa=3

I think you need to go over wpa= in the docs. I think you will find the following is the setting you want:

wpa=2

> wpa_passphrase=MyPa$$Phr@$e
> wpa_key_mgmt=WPA-PSK
> wpa_pairwise=CCMP

The following is missing:

driver=nl80211

You could add the following:

ieee80211n=1
wmm_enabled=1
ht_capab=[SHORT-GI-20]

> 8814au options:
> 
>
> options 8814au rtw_drv_log_level=3 rtw_led_ctrl=1 rtw_vht_enable=1 rtw_power_mgnt=1 rtw_switch_usb_mode=1

You are actually wasting your time going into USB3 mode if you are running 2.4 Ghz only and USB2 is more stable so recommend:

rtw_switch_usb_mode=0

> 
> You every reason to have little faith in my skill ;-) This is the first time I've used linux in over 10 years ;-)

It just takes a little time to get up to speed.

> 
> My channel 6 is pushing anyone else to the sides ;-) I am living in a flat, so the Wifi Card nicely covers the whole area.

Copy all.

morrownr, 2022-07-04:
Another line you might want to add to your hostapd.conf

max_num_sta=64

DennisKuhn, 2022-07-04:
@morrownr thanks a lot for all the options. I've tried them and it is still the same. After 30 clients it breaks somehow, the initial devices keep working, late connected ones are unstable. Initially my mobile connects fine to it, a bit slow establishing the connection but works. After I've power up/connected 30 devices, first my mobile connects "without internet" and then just fails instanstantly with "Connection failed".

morrownr, 2022-07-04:
Hey @DennisKuhn 

I understand. You can probably tell that I am hesitant to dive into the code looking for a solution. I can try and I will be in the code soon to work on a couple of what really amounts to some cosmetic/compiler issues but there are some things you need to understand about this specific driver.

This driver has been poorly maintained by Realtek. I maintain 4 other Realtek drivers. Even the driver for the 8812au chipset, which is older than this chipset, is much better maintained and far more feature rich. There have been no new releases of code for this driver since 2019. I have struggled with this driver. With help from others, we have this driver where it can actually do AP mode, as of about a year ago. This driver is very fragile in that changes in one place will lead to unintended consequences in another place. So, what I am trying to say is that I may not be able to find a solution to this issue and if I do, it will likely take a LOT of work and won't come quickly.

Suggestions: Temporarily switch to using the internal 2.4 Ghz radio to test if you see the same issue. Making the change should be as simple as changing the following in hostapd.conf:

interface=wlan0

I realize the range won't be as good but maybe this will work for testing. Depending on what you find, I may have additional suggestions.

I have many usb wifi adapters as you might imagine given the contents of this site. To be very straightforward with you, if you have ask me what usb wifi adapter to get for a project such as this, any adapter with a 8814au chipset would have been at the very bottom of the list. I maintain assessments of the drivers I maintain with item 5 in the Main Menu (I posted a link earlier).

DennisKuhn, 2022-07-05:
@morrownr thanks a lot for your input. I support your hesitation to modify the driver, I am not 100% sure that it's the driver, the driver catching the problem or just happening at the same time. Unfortunately swapping the interfaces isn't as trivial, I am currently routing wlan1 -> wlan0. I will check it out closer to the weekend, that will definitely help to isolate the issue.

I wish I would have seen your list of USB Wifi before, I will certainly use it if I can't get the card to run. But at the moment I am hassistant, as it may not even be the driver.

Thanks again for your diligence, I'll be back "shortly".

DennisKuhn, 2022-07-10:
@morrownr I have plugged the RPi directly into eth0:
- using the 8814 driver I still have the same issue at 32 devices, it counts up in get_camid to 31
- using the built in wifi I get no error but it stops at 8 clients

DennisKuhn, 2022-07-10:
@morrownr unfortunately our choice in New Zealand isn't huge:
- [D-Link DWA-X1850 Dual Band AX1800 Wi-Fi 6 USB Adapter](https://www.pbtech.co.nz/product/NETDLK61850/D-Link-DWA-X1850-Dual-Band-AX1800-Wi-Fi-6-USB-Adap)
- [TP-Link Archer T9UH Dual-Band AC1900 High Gain USB Wi-Fi Adapter](https://www.pbtech.co.nz/product/NETTPL0035/TP-Link-Archer-T9UH-Dual-Band-AC1900-High-Gain-USB) I got his one
- [TP-Link Archer T4U Plus Dual-Band AC1300 USB Wi-Fi Adapter with High-Gain Antenna](https://www.pbtech.co.nz/product/NETTPL0121/TP-Link-Archer-T4U-Plus-Dual-Band-AC1300-USB-Wi-Fi)
- [Edimax EW-7811DAC AC600 Wi-Fi Dual-Band Directional High Gain USB Adapter - 802.11 ac -](https://www.pbtech.co.nz/product/NETEDM1081/Edimax-EW-7811DAC-AC600-Wi-Fi-Dual-Band-Directiona)
- [Edimax EW-7811UAC AC600 Wi-Fi Dual-Band High Gain USB Adapter. IEEE 802.11ac and backward compatible.](https://www.pbtech.co.nz/product/NETEDM1086/Edimax-EW-7811UAC-AC600-Wi-Fi-Dual-Band-High-Gain)
- [TP-Link TL-WN822N N300 High Gain USB Wi-Fi Adapter](https://www.pbtech.co.nz/product/NETTPL0822/TP-Link-TL-WN822N-N300-High-Gain-USB-Wi-Fi-Adapter)

I am thinking about buying the Alfa AWUS036ACM Dual Band on eBay, as it seems to be the recommended one on your list. Do you think it would solve the problem?

morrownr, 2022-07-10:
> * using the 8814 driver I still have the same issue at 32 devices, it counts up in get_camid to 31
> * using the built in wifi I get no error but it stops at 8 clients

Not much help there.

morrownr, 2022-07-10:
> @morrownr unfortunately our choice in New Zealand isn't huge:

I understand but I think it is hard to get anything worth getting locally anywhere in the world these days. Without online ordering we would be out in the cold.
 
> * [D-Link DWA-X1850 Dual Band AX1800 Wi-Fi 6 USB Adapter](https://www.pbtech.co.nz/product/NETDLK61850/D-Link-DWA-X1850-Dual-Band-AX1800-Wi-Fi-6-USB-Adap)

That DWA-X1850 uses a rtl8832au chipset. Avoid it!

> * [TP-Link Archer T9UH Dual-Band AC1900 High Gain USB Wi-Fi Adapter](https://www.pbtech.co.nz/product/NETTPL0035/TP-Link-Archer-T9UH-Dual-Band-AC1900-High-Gain-USB) I got his one
> * [TP-Link Archer [...]

I have no idea and I have been thinking of a way to test. I have a Alfa AWUS036ACM. I have a lot of adapters. What I lack is clients. About the best I can do 12 or if I drag a couple of things out of retirement I might get up to 16. Unless there is a way to simulate it, I can't get to 32.

Let's take a different look at this: You only need 2.4 GHz, right? 

Maybe you can find something with a rt5370 or rt3070 chipset for a good price. Main Menu item 2 has sections for 5370 and 3070 chipsets. The Panda PAU08 probably has the longest range but something like the following might work also:

https://www.amazon.com/Wireless-Adapter-Notebook-Raspberry-WINCE6-0/dp/B09NNMJQVL

That EDUP uses a 5370 chipset. I have a 5370 adapter that looks just like the EDUP but I still don't have 32 clients to test.

Have you thought about using two usb wifi adapters, both using 2.4 GHz? hostapd can handle it.

DennisKuhn, 2022-07-11:
@morrownr unfortunately the testing is an issue. I believe only expensive special hardware exists to simulate several clients. I think one card can emulate around 7 clients, that would be 7 Wifi Cards in one box going crazy ;-)

I had a dive through the driver, I can find that the buffer size is specified through `cam_ctl->num`, but I have no clue where it is set and if it's possible to set it to a bigger value.  I do think for testing, changing the starting point from 0 to like 25 and connect devices, would yield the same effect:
[core/rtw_wlan_util.c:1191: (for (i = 25; i < cam_ctl->num; i++)](https://github.com/morrownr/8814au/blob/16476ce9150fe44fbf4c0264286dc6b195c938e3/core/rtw_wlan_util.c#L1191)

Thanks a lot for those recommendations, the price isn't a issue, it's the shipping delay :-) The Panda PAU08 seems quite promising, especially as they even marked it to be AP capable.

I am currently using the internal Wifi (8) and the Archer Wifi (32) to drive 40 clients. It works but is nowhere close to enough. So I really need to find a card that can handle at least 60 clients. Do you know if it's possible to plug two Panda PAU08 in one box?

morrownr, 2022-07-11:
If you find something that you think might work and want me to take a look and offer an opinion, I'll be glad to do so.

Another option that is not a single band adapter but it works really well as an AP and like the others I've been suggesting, it uses in-kernel drivers:

ALFA AWUS036ACHM

The list of recommended adapters has a section on this adapter and has links to various sellers around the world. It also has links to ALFA so you might email ALFA and tell them where you are and ask where dealers are that are near you. This specific adapter has very long range. I have tested it against many other adapters. Using it along with a rt5370 based adapter that will probably have shorter range might be a good solution. ALFA adapters do not compete with the cheap adapters that are out there but ALFA has a long reputation of making very good quality, long range adapters.

DennisKuhn, 2022-07-18:
I've got two of the Panda PAU08 2.4 GHz and they arrived today. Especially for the price, they work amazing. Literally plug and play with 40 clients ;-) There is even a noticeable improvement in CPU usage. I haven't tried both cards yet, as I am really happy it finally works using one Wifi instead of 3 ;-) Thanks a lot for the lengthily support. If I need more Wifi I'll definitely come back to the ones recommended above. They still may come in handy instead of the onboard one, to talk to the main router. Thanks a lot for the support getting it all running smooth and still options to go ;-)

morrownr, 2022-07-18:
So a single Panda PAU08 is allowing you to have more than 32 clients without complaint? If so, that actually does not surprise me. I have seen a lot of problems solved by using adapters that use in-kernel drivers.

> If I need more Wifi I'll definitely come back to the ones recommended above. They still may come in handy instead of the onboard one, to talk to the main router.

Ah. Let me throw a suggestion out for a good adapter to connect back to your main router:

ALFA AWUS036ACHM

Long range, AC600, in-kernel driver.

https://www.desertcart.nz/products/271070964-alfa-awus-036-achm-802-11-ac-wi-fi-range-boost-usb-adapter-alfa-long-range-dual-band-ac-1200-usb-3-0-wi-fi-adapter-w-2-x-5-d-bi-external-antennas-2-4-g-hz-300-mbps-5-g-hz-867-mbps-802-11-ac-a-b-g-n-windows-mac-os-kali-linux-supported-awus-036-acm-%E3%80%90-new-version-type-c-wi-fi-usb%E3%80%91-alfa-awus-036-ach-%E3%80%90-type-c%E3%80%91-long-range-dual-band-ac-1200-wireless-usb-wi-fi-adapter-w-2-x-5-d-bi-external-antennas-2-4-g-hz-300-mbps-5-g-hz-867-mbps-802-11-ac-a-b-g-n-alfa-network-wi-fi-camp-pro-2-universal-wi-fi-internet-range-extender-kit-for-caravan-motorhome-boat-rv

DennisKuhn, 2022-07-20:
@morrownr thanks a lot for all your information. Yip the ALFA AWUS036ACHM is my next choice.

So far, a couple of days, I am very impressed. No issues at all and small noticeable CPU usage reduction and more noticeable RAM usage reduction. I just plugged the second card in and it worked as client straight away. I think the Panda PAU08 2.4 GHz adapters are ideal for RPi and home assistant/automation. Both adapters run easily on the RPi 4 with an awesome range and no extra power needed. I still use the RPi onboard Wifi for the uplink.

Thanks a lot for your support, that's definitely a cost efficient and extendable solution ;-)

closed 2022-07-20 by DennisKuhn

dubhater, 2024-02-18:
In case other people run into this...

From ``hal/rtl8814a/rtl8814a_hal_init.c``:
```C
hal_spec->macid_num = 128;
hal_spec->sec_cam_ent_num = 64;
```
This suggests the chip itself can handle 127 clients (1 macid is reserved), and it can handle encryption/decryption for 64 (63?) of those clients. (For the rest *I think* the driver could handle encryption/decryption on the host CPU, but instead it throws the „no room” error.)

However, the driver limits those numbers more in ``os_dep/linux/os_intfs.c``:
```C
dvobj->macid_ctl.num = rtw_min(hal_spec->macid_num, MACID_NUM_SW_LIMIT);
dvobj->cam_ctl.num = rtw_min(hal_spec->sec_cam_ent_num, SEC_CAM_ENT_NUM_SW_LIMIT);
```

Those constants are defined in ``include/drv_conf.h``:
```C
#define MACID_NUM_SW_LIMIT 32
#define SEC_CAM_ENT_NUM_SW_LIMIT 32
```

So if you want more than ~32 clients, try to make those two constants 128 and 64.