morrownr/8814au #80: Monitor mode freezes network stack

sab24, 2022-08-16:
When using the kernel module from this git repo, typing the command

```sudo iw phy phy0 interface add mon0 type monitor```

makes the system unresponsive. Networking is no longer working, iw commands don't work anymore, so it seems that the NL80211 kernel interface is frozen and the process shows a D flag, making it impossible to kill.

Have there been people successfully putting the device in monitor mode? I tried with archlinux as host, and then made a VM with usb passthrough with KVM on ubuntu 22 which uses 5.15. I tried multiple USB ports.

morrownr, 2022-08-16:
Hi @sab24 

> Have there been people successfully putting the device in monitor mode?

Yes. The command you are using works with Linux Wireless Standards compliant in-kernel drivers. For a list of adapters that use in-kernel drivers, see Main Menu item 2:

https://github.com/morrownr/USB-WiFi

Realtek out-of-kernel drivers will not work with the command you used. To help, I created documentation and a script to help you get your 8814au based adapter using this driver into monitor mode. Go to the Main Menu and select item 10. The script will take you into monitor mode in a way that will work with Realtek drivers...it also works with the Mediatek in-kernel drivers.

Let me know how it goes.

Nick

sab24, 2022-08-16:
Hi Nick,

That seems to work, thanks for the help. I also like your write up about which USB devices are well supported, thinking of getting either a ALFA AWUS036ACM or ALFA AWUS036ACHM.

closed 2022-08-16 by sab24

morrownr, 2022-08-16:
@sab24 

I have both an ACM and ACHM. I think they are good quality adapters. Which you choose really depends on what you will be doing most of the time with the adapters.

If the adapters is primarily for monitor mode, I would go with the ACHM as it has range that is second to none. If the adapter will sometimes be used for monitor mode and mostly for other things and the faster speed is important, then the ACM is probably the one to go with.

If you go over to the Main Menu (link in previous msg) you will find that the repo there has `Issues` and there are a lot of discussions about adapters. You could post an issue there asking what folks recommend depending on your use case.

Good luck.