morrownr/8814au #81: Compilation error on android linux kernel version 4.14

KiGamji, 2022-08-24:
I was trying to compile this driver in kali nethunter, but got this error:
[image]
My device is Redmi Note 8 Pro, and the full kernel version is 4.14.141-g77f1f89.
Here is the make.log output:
```
DKMS make.log for rtl8814au-5.8.5.1 for kernel 4.14.141-g77f1f89 (aarch64)
Wed Aug 24 07:31:48 UTC 2022
make ARCH=arm64 CROSS_COMPILE= -C /begonia-q-oss M=/var/lib/dkms/rtl8814au/5.8.5.1/build  modules
make[1]: Entering directory '/begonia-q-oss'

  WARNING: Symbol version dump ./Module.symvers
           is missing; modules will have no dependencies and modversions.

  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_chplan.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_sta_mgt.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_ap.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/mesh/rtw_mesh.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_rson.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_tdls.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_br_ext.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_iol.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_sreset.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_btcoex_wifionly.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_btcoex.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_beamforming.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_odm.o
In function '__nat25_add_pppoe_tag',
    inlined from 'nat25_db_handle' at /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_br_ext.c:908:11:
/var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_br_ext.c:116:9: warning: 'memcpy' forming offset [40, 2051] is out of the bounds [0, 40] of object 'tag_buf' with type 'unsigned char[40]' [-Warray-bounds]
  116 |         memcpy((unsigned char *)ph->tag, tag, data_len);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_br_ext.c: In function 'nat25_db_handle':
/var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_br_ext.c:877:63: note: 'tag_buf' declared here
  877 |                                                 unsigned char tag_buf[40];
      |                                                               ^~~~~~~
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_rm.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_rm_fsm.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/rtw_rm_util.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/core/efuse/rtw_efuse.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/osdep_service.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/os_intfs.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/usb_intf.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/usb_ops_linux.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/ioctl_linux.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/xmit_linux.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/mlme_linux.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/recv_linux.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/ioctl_cfg80211.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/rtw_cfgvendor.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/wifi_regd.o
  CC [M]  /var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/rtw_android.o
[... 115 lines cut ...]
   10 |         .name = KBUILD_MODNAME,
      |                 ^~~~~~~~~~~~~~
/var/lib/dkms/rtl8814au/5.8.5.1/build/8814au.mod.c:11:10: error: 'struct module' has no member named 'init'
   11 |         .init = init_module,
      |          ^~~~
/var/lib/dkms/rtl8814au/5.8.5.1/build/8814au.mod.c:11:17: warning: excess elements in struct initializer
   11 |         .init = init_module,
      |                 ^~~~~~~~~~~
/var/lib/dkms/rtl8814au/5.8.5.1/build/8814au.mod.c:11:17: note: (near initialization for '__this_module')
/var/lib/dkms/rtl8814au/5.8.5.1/build/8814au.mod.c:15:10: error: 'struct module' has no member named 'arch'
   15 |         .arch = MODULE_ARCH_INIT,
      |          ^~~~
/var/lib/dkms/rtl8814au/5.8.5.1/build/8814au.mod.c:15:17: error: 'MODULE_ARCH_INIT' undeclared here (not in a function)
   15 |         .arch = MODULE_ARCH_INIT,
      |                 ^~~~~~~~~~~~~~~~
/var/lib/dkms/rtl8814au/5.8.5.1/build/8814au.mod.c:15:17: warning: excess elements in struct initializer
/var/lib/dkms/rtl8814au/5.8.5.1/build/8814au.mod.c:15:17: note: (near initialization for '__this_module')
/var/lib/dkms/rtl8814au/5.8.5.1/build/8814au.mod.c:8:25: error: storage size of '__this_module' isn't known
    8 | __visible struct module __this_module
      |                         ^~~~~~~~~~~~~
make[2]: *** [scripts/Makefile.modpost:129: /var/lib/dkms/rtl8814au/5.8.5.1/build/8814au.mod.o] Error 1
make[1]: *** [Makefile:1634: modules] Error 2
make[1]: Leaving directory '/begonia-q-oss'
make: *** [Makefile:2384: modules] Error 2
```

KiGamji, 2022-08-28:
@morrownr ?

morrownr, 2022-08-28:
Hi @KiGamji 

Guess I missed your earlier report. It has been a busy week.

> Am I doing something wrong or is it a kernel/driver issue?

I wouldn't say you did anything wrong. This and the rest of the Realtek out-of-kernel drivers at this site are not tested with Android. I have no problem helping you work toward getting the driver going on your rig but I currently do not have Android on anythinh except my phone so unless we can figure out how to get a version of Android that is a near match to yours on something I have around here, it will be you testing. I have a RasPi3B and RasPi4B. They are both ARM64 so if the Android you are using comes in a version of the RasPi's, point me to it.

Can you what you did for installation Steps 3 and 8?

For what it is worth, this does not look bad. We can probably sort this out.

Nick

KiGamji, 2022-08-29:
> if the Android you are using comes in a version of the RasPi's

What do you mean?

> Can you what you did for installation Steps 3 and 8?

For step 3 i downloaded begonia (redmi note 8 pro) kernel open source: begonia-q-oss.
For step 8 i ran ARM64_RPI.sh script.
I have also kernel headers for akame kernel (modded kernel for begonia) i was using it earlier, and this driver was compiling successfully after installing kernel headers for akame kernel. but now i have no ability to use it, so the compiled driver dont works anymore. but i can send you akame kernel headers .deb file.

morrownr, 2022-08-29:
> > if the Android you are using comes in a version of the RasPi's
> 
> What do you mean?

It helps me figure out what the problem is if I can test the problem here and see it with my own eyes. I am unfamiliar with the Android Linux distro you are using. If I can install it on some computer I have here, it will greatly assist me in helping you. Can you point me to the website for this distro?

> > Can you what you did for installation Steps 3 and 8?
> 
> For step 3 i downloaded begonia (redmi note 8 pro) kernel open source: begonia-q-oss.

Generally for Step 3, the required dev environment is loaded with a command similar to the following:

sudo apt install -y linux-headers-$(uname -r) build-essential bc dkms git libelf-dev

Without us knowing what the proper command is for your distro, some of the right stuff may not be loaded.

> For step 8 i ran ARM64_RPI.sh script. I have also kernel headers for akame kernel (modded kernel for begonia) i was using it earlier, and this driver was compiling successfully after installing [...]

The .deb file would not help. Does the distro maker have forums where you can ask questions? If so, you might ask about this there to see if someone can give us some hints.

KiGamji, 2022-08-29:
> Can you point me to the website for this distro?

yup, its open source: https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/begonia-q-oss

> linux-headers-$(uname -r)

but my kernel version is 4.14, its old, so i am getting error. also it have random numbers and letters at the end (thats commit id i guess).
also, android kernel is very different, not based on debian, or any other distro, so this wont work i guess.

> The .deb file would not help.

why? after installing deb file via dpkg -i, i can find my kernel version (4.14.141-g77f1f89) folder somewhere is /usr, and the driver compiles successfully with it. that folder haves some of the files from begonia-q-oss but some of them are edited. but that .deb file is for akame kernel, and akame kernel has different kernel name, so the driver dont working with stock kernel.

> Does the distro maker have forums where you can ask questions?

if you meant stock original kernel maker, then no. Xiaomi made it

morrownr, 2022-08-29:
> also, android kernel is very different, not based on debian, or any other distro, so this wont work i guess.

It can work if we are able to set the build environment up correctly. Android is different in some ways but a 4,14 kernel should not be a problem.

KiGamji, 2022-08-30:
i cant install it here for some reason. i think its too old and can be found only in kali archives
[image]

morrownr, 2022-08-30:
The headers had to have already been in place or you would not have gotten as far as you did. My opinion is that the needed changes are probably small. It is just a matter of knowing what to do. Do you know of any forums that discuss this Android that you use? The answer is there, it is just a matter of finding it.

KiGamji, 2022-08-31:
> My opinion is that the needed changes are probably small. It is just a matter of knowing what to do.

Yea, but the problem is that I don't know what to do.

> Do you know of any forums that discuss this Android that you use?

You mean't that device and that android version?

morrownr, 2022-09-01:
> Yea, but the problem is that I don't know what to do.

Me either.

> You mean't that device and that android version?

The Android that you are using.

KiGamji, 2022-09-01:
> The Android that you are using.

Yes, I know. But does that really depends on android version, and not on kernel vendor? I am using custom rom now, and its based on begonia-r-oss. (kernel for android R). i'll try to compile with this kernel source

KiGamji, 2022-12-08:
@morrownr Hey there! I learned about linux kernels, learned basics of C and made a modded kernel for my device. Turns out your driver easily compiles on
 4.14 kernel. I saw 4.9 kernels with built rtl88xxxu drivers, so your drivers works with these versions!
My suggestion is to update rtl88xxxu README's to state that 4.9 and 4.14 are also supported.
Btw I'm also enjoying with your driver on my Mint 21 (kernel 5.15) Thank you!
[image]
[image]
[image]

closed 2022-12-08 by KiGamji

morrownr, 2022-12-08:
This is cool. I'm really bust right now but will address your comments as I have time. I am cusrious as to whether you would be interest in making a document explain how to compile on Android that I could include with this and the other drivers?

KiGamji, 2022-12-09:
Hm, would be cool. Why not. But I have a trouble with 88x2bu driver. I enabled its USB WiFi in config, but i dont see it building as a module, while 8814au does. Do you have more socials so it will be more convenient to talk?

morrownr, 2022-12-09:
@KiGamji 

I will delete this message after sending.

morrownr@gmail.com

You are welcome to contact me.

Nick

cr4sh-me, 2023-01-27:
> Hm, would be cool. Why not. But I have a trouble with 88x2bu driver. I enabled its USB WiFi in config, but i dont see it building as a module, while 8814au does. Do you have more socials so it will be more convenient to talk? UPD: was a problem on my side, fixed it.

Which patch did you applied?

KiGamji, 2023-01-27:
> Which patch did you applied to get 88xxAU working?

First of all, what kernel version do patching? Can you provide a link for a git repository with it's source, if it exists?
I can help you easier, if you will contact me on other socials. Do you have Telegram or Discord?

cr4sh-me, 2023-01-27:
> > Which patch did you applied to get 88xxAU working?
> 
> First of all, for what kernel version are you trying to compile this driver? If that's a modded kernel, can you provide a link for a git repository with it's source, if it exists? I can help you easier if you will contact me on any other social. Do you have Telegram or Discord?

Yes, add me on discord and ill send you everything 
r00ber#9964