morrownr/8814au #84: SoftAP drops after a few minutes

logan001, 2022-09-05:
Adapter USB AC-68
kernel 5.19.6
after a few minutes from connecting to AP the connection drops and i can't reconnect anymore. the phone is trying to connect but fails.
last time i use it softap was on 5.18 and worked very nice.

morrownr, 2022-09-05:
Hi @logan001 

Please provide the output of the following command:

sudo uname -mr; mokutil --sb-state; lsusb; rfkill list all; dkms status; iw dev; iw reg get

Have you updated the driver since moving to kernel 5.19.6? Such as...

$ sudo ./remove-driver.sh
$ git pull
$ sudo ./install-driver.sh

Regards

morrownr, 2022-09-05:
FYI: I was able to duplicate the problem while using this driver in AP mode on a RasPi. I'll investigate further as I have time. This old driver is fragile so I could run into issues.

logan001, 2022-09-06:
i just realize that i forgot to add it to ignore list in networkmanager. after ignore it, so far (20 minutes) its still working

morrownr, 2022-09-06:
If you have the time, keep me posted on this. This driver is a challenge to maintain and it stopped when I tested on a RasPi so I need to get to the bottom of that anyway so I will be watching it for a while.

Nick

logan001, 2022-09-06:
ok. ill be testing it for the next few days and let you know if i get that bug again or not

logan001, 2022-09-08:
ill close this. i had no problems after ban the adapter in network manager

closed 2022-09-08 by logan001