morrownr/8814au #85: Ubuntu 22.04 net/ipx.h: No such file or directory

MedievalSteel, 2022-09-08:
[image]

closed 2022-09-08 by MedievalSteel

VortexAcherontic, 2023-08-28:
> Same

linux-kernel-headers installed?

ashpoolin, 2023-08-28:
Yes, linux-headers-5.15.0-79-lowlatency, although I was trying to compile 8822bu and 8812-8821AU drivers. Ubuntu 22.04 LTS afaik. I get the same error for a missing include (ipx.h) and so every source of these drivers I find fails to compile in the same way.

VortexAcherontic, 2023-08-28:
> Yes, linux-headers-5.15.0-79-lowlatency, although I was trying to compile 8822bu and 8812-8821AU drivers. Ubuntu 22.04 LTS afaik. I get the same error for a missing include (ipx.h) and so every source of these drivers I find fails to compile in the same way.

Hm, can't only speak for openSUSE Tumbleweed and Leap. Those build just fine.
The ipx.h file seems to be provided by glibc.

```
rpm -qf /usr/include/netipx/ipx.h 
glibc-devel-2.38-1.1.x86_64
```

Maybe you're just missing the Ubuntu equivalent.

morrownr, 2023-08-29:
@ashpoolin 

> linux-headers-5.15.0-79-lowlatency, 

I test as widely as I have the capability to but I do not currently test with the lowlatency kernel. I suspect this might have something to do with this problem. My bet is that the problem would go away if you revert to a regular kernel. What you could try is updating to the 5.19 HWE kernel.

I suspect we can find the problem but I am having some health problems currently which have slowed me a lot.

ashpoolin, 2023-08-29:
It's not a problem. I'm afraid I'm unhappy with this kernel--didn't mean to install it. Just wanted document my experience. Thank you, and I hope you can get well soon. Take care, Ash