morrownr/8814au #88: Error((( I need help(((

Denis-AAA, 2022-10-16:
Tried all the methods described. At first it refused to work. Then I updated the Kali kernel to 5.19 (it was 5.18) and managed to start it once, saw the device, switched to monitor mode and got the HandShake of the test router. In the morning, the ability to turn on the monitor mode disappeared. Decided to reinstall everything from the beginning. But the driver refuses to install. Installing with i`make' gives the following error:
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/5.19.0-kali2-amd64/build M=/home/kali/src/8814au modules
make[1]: вход в каталог «/usr/src/linux-headers-5.19.0-kali2-amd64»

ERROR: Kernel configuration is invalid.
include/generated/autoconf.h or include/config/auto.conf are missing.
Run 'make oldconfig && make prepare' on kernel src to fix it.

make[1]: *** [/usr/src/linux-headers-5.19.0-kali2-common/Makefile:750: include/config/auto.conf] Ошибка 1
make[1]: выход из каталога «/usr/src/linux-headers-5.19.0-kali2-amd64»
make: *** [Makefile:2384: modules] Error 2

Denis-AAA, 2022-10-16:
When installing via `.\install-driver.sh` an error occurs:

Running install-driver.sh version 20220705
5.19.0-kali2-amd64
x86_64
Starting installation...
Copying source files to: /usr/src/rtl8814au-5.8.5.1
Copying 8814au.conf to: /etc/modprobe.d
Creating symlink /var/lib/dkms/rtl8814au/5.8.5.1/source -> /usr/src/rtl8814au-5.8.5.1
Sign command: /usr/lib/linux-kbuild-5.19/scripts/sign-file
Signing key: /var/lib/dkms/mok.key
Public certificate (MOK): /var/lib/dkms/mok.pub

Building module:
Cleaning build area...
'make' -j2 KVER=5.19.0-kali2-amd64 KSRC=/lib/modules/5.19.0-kali2-amd64/build...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.19.0-kali2-amd64 (x86_64)
Consult /var/lib/dkms/rtl8814au/5.8.5.1/build/make.log for more information.
An error occurred. dkms build error = 10
Please report this error.
You will need to run the following before reattempting installation.
$ sudo ./remove-driver.sh

P.s.: 
Kali 5.19
Tp-Link archer T9UH

morrownr, 2022-10-17:
Hi @Denis-AAA 

> ARCH=arm64

> x86_64

I'm seeing some conflicting information. What kind of a CPU are you running?

Denis-AAA, 2022-10-17:
x86_64

Denis-AAA, 2022-10-17:
> Hi @Denis-AAA
> 
> > ARCH=arm64
> 
> > x86_64
> 
> I'm seeing some conflicting information. What kind of a CPU are you running?
I have x86_64. Why it also shows ARCH=arm64, I don't know

Denis-AAA, 2022-10-17:
I have x86_64. Why it also shows ARCH=arm64, I don't know

morrownr, 2022-10-17:
It is not clear to me either.

Open up the Makefile in a test editor and make sure the following lines are like you see below:

```
###################### Platform Related #######################
CONFIG_PLATFORM_I386_PC = y
CONFIG_PLATFORM_ARM_RPI = n
CONFIG_PLATFORM_ARM64_RPI = n
```

Denis-AAA, 2022-10-18:
Yes. Absolutely such

morrownr, 2022-10-18:
I'm going to be working on this driver later this week to try to get it to work with kernel 6.1. I will see what I can find out.

Denis-AAA, 2022-10-18:
OK. Thank you. I will follow the updates.

Denis-AAA, 2022-10-21:
ID 2357:0106 TP-Link Archer T9UH v1 [Realtek RTL8814AU]

morrownr, 2022-10-21:
@Denis-AAA 

Just a note to let you know I am working on this driver today. You will likely see several uploads. Hope I run into the problem.

Denis-AAA, 2022-11-06:
Unfortunately, these errors remain.

morrownr, 2022-11-07:
Hi @Denis-AAA 

Thanks for checking in. I get very busy at times.

I had to reread the thread to see what comes to mind. I still can't explain the ARM64 vs amd64 issue. The only way that should happen is if you run `./ARM64_RPI.sh` to change the Makefile to compile for ARM64.

Try the below and post the results:

From the driver directory:

$ make clean
$ make

Denis-AAA, 2022-11-07:
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.19.0-kali2-amd64/build M=/home/kali/8814au  modules
make[1]: вход в каталог «/usr/src/linux-headers-5.19.0-kali2-amd64»

  ERROR: Kernel configuration is invalid.
         include/generated/autoconf.h or include/config/auto.conf are missing.
         Run 'make oldconfig && make prepare' on kernel src to fix it.

make[1]: *** [/usr/src/linux-headers-5.19.0-kali2-common/Makefile:750: include/config/auto.conf] Ошибка 1
make[1]: выход из каталога «/usr/src/linux-headers-5.19.0-kali2-amd64»
make: *** [Makefile:2389: modules] Ошибка 2

Denis-AAA, 2022-11-07:
Same way((((

morrownr, 2022-11-07:
Okay, well, it looks like maybe the ARM64 vs amd64 issue has gone away so let's press on.

It looks to me like there is an issue with your header files. You could do a purge of the header files that are currently installed and then reinstall.

With that said, I currently do not have Kali installed on the test machine where it normally is because the last release refused to install. I have little toleration for BS so Kali is in the penalty box right now, When a new release is available, I may try again. Kali is not a particularly stable distro imho so this problem could be a bug in Kali.

This could also be an issue with how you are upgrading the kernel.

Denis-AAA, 2022-11-11:
Reinstalled Kali. Driver installed .\install-driver.sh. However, the adapter now does not see the network. Switched to monitor mode, but also did not see the network

morrownr, 2022-11-11:
What does the wireless interface look like:

$ iw dev

Denis-AAA, 2022-11-11:
phy#2
        Interface wlan0
                ifindex 5
                wdev 0x200000001
                addr 7c:c2:c6:32:aa:67
                type managed
                txpower 20.00 dBm

morrownr, 2022-11-11:
That is a good wireless interface which tells us a lot of things are correct.

Does your system see any AP's?

What does the following show?

$  iw reg get

Tell me about the AP you are trying to associate with? Dual band? Security settings? WPA3, WPA2? Both bands using the same name?

Denis-AAA, 2022-11-11:
No, he does not see a single access point in Kali. Windows sees my access point and neighbors access point.

$ iw reg get
global
country RU: DFS-UNSET
        (2400 - 2483 @ 40), (N/A, 20), (N/A)
        (5150 - 5350 @ 160), (N/A, 20), (N/A), NO-OUTDOOR
        (5650 - 5850 @ 160), (N/A, 20), (N/A), NO-OUTDOOR
        (57000 - 66000 @ 2160), (N/A, 40), (N/A), NO-OUTDOOR

I tried different countries, but nothing has changed, just does not see.
In monitor mode, too, sees nothing.

$ iwconfig

wlan0     unassociated  Nickname:"WIFI@RTL8814AU"
          Mode:Managed  Frequency=2.412 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

Sensitivity should be 0/0 ?

morrownr, 2022-11-11:
> Sensitivity should be 0/0 ?

It and some of the other settings should not be 0 if you are associated with an AP but you are not.

This is not the greatest driver. At this point, we are down to things like:

- change the USB port in use.

- use a USB2 port if a USB3 port is in use.

- do a manual scan with wpa_supplicant to see if scanning is working.

- Check the module (driver) parameters (options):

$ sudo ./edit-options.sh

You can try to change various options and see what happens. They take effect with a reboot.

morrownr, 2022-11-14:
It is good to see progress. I have a guide to using usb adapters in virtualbox in the FAQ.

Denis-AAA, 2022-11-17:
OK. Thanks I'll look into it in a few days.