morrownr/8814au #89: Unable to locate package raspberrypi-kernel-headers (raspberry pi 4b)

FahdSaadA, 2022-10-24:
hello everyone
thank you for this cool tool and effort, however i got this issue..i have tried the firmware in raspberry pi 4b the latest kernel 5.15 and it work there..however it does not at 5.10 which seems unable to relocate the kernel...
i had followed the steps and the installation failed at the last step (10)  sudo ./install-driver.sh

the kernel did not show up...by the way i did not upgrade , because i need this firmware in this version 5.10.92 for other tools are not supported by the latest 5.15 raspbian..so i wish to know if i still can use and relocate the kernel to solve this issue

the message error 

Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package raspberrypi-kernel-headers.
E: Couldn't find any package by glob 'raspberrypi-kernel-headers.'
E: Couldn't find any package by regex 'raspberrypi-kernel-headers.'
root@raspberrypi:/home/pi# ^C

morrownr, 2022-10-24:
Hi @FahdSaadA 

The problem is definitely that the wrong headers are installed.  I have not experienced this but have seen others that have. There is a thread in the RasPi forums that addresses this issue and shows a fix but I can't remember what the fix was. You might want to do search in the RasPi forums for "wrong header files" and some other variations to see what you come up with.

If you don't find the fix, you might try purging any and all header files you currently have installed and then reinstall after reboot.

Nick

morrownr, 2022-10-24:
If you do find the fix, please post it back here future users.

FahdSaadA, 2022-10-25:
thank you very much Nick. and yes it was the header mismatch. since i was using 5.10.93 and suggested one is 5.10.103
so by using the correct header, the device works fine...but yet i did not use it for monitoring..

morrownr, 2022-10-25:
Good. I'll mark this as solved. As far as monitor mode goes, I have some info that might help:

https://github.com/morrownr/USB-WiFi

Select menu item 10.

Regards

joseguzman1337, 2026-02-12:
Maintainer update: this issue has been addressed in upstream `main` at commit `b1866ce2b857a8dfe2e147e19eb8eca0a842ce18`. Resolution summary: pass under maintainer policy; guidance and diagnostics provided. For this report type, we added/updated maintainer guidance and diagnostics workflow; if still reproducible, please post full repro steps and environment details.