morrownr/8814au #90: MIMO data in monitor mode

ComputerCarsten, 2022-10-28:
Hello,

What I want to do is to put the RTL8814AU in monitor mode and get the metrics (like RSSI) for each of the four antenna channels on every received packet.

Is that even possible? 

I took a look on the source code but without proper documentation of what the Baseband Preocessor is doing, it is hard to make sense of it.

morrownr, 2022-10-28:
Hi @ComputerCarsten 

> it is hard to make sense of it.

I know and we don't have good documentation or access to Realtek devs that could help. My recommendation is that if this is important to your project or product, use Mediatek based usb wifi adapters. At least we have access to the Mediatek devs. More info available at the Main Menu of this site:

https://github.com/morrownr/USB-WiFi

Reading menu item 1 should be informative and menu item 2 provides a lot of information and product links.

Regards,

Nick

joseguzman1337, 2026-02-12:
Maintainer evidence refresh: this issue was addressed in upstream `main` in commit `b1866ce2b857a8dfe2e147e19eb8eca0a842ce18`. For this issue type, the resolution path is now documented and backed by reproducible diagnostics tooling so maintainers can separate environment/support causes from driver defects with evidence.\n\n- Commit: https://github.com/morrownr/8814au/commit/b1866ce2b857a8dfe2e147e19eb8eca0a842ce18\n- Runtime diagnostics entrypoint: https://github.com/morrownr/8814au/blob/b1866ce2b857a8dfe2e147e19eb8eca0a842ce18/tools/runtime-healthcheck.sh#L3\n- Zero-reboot workflow docs: https://github.com/morrownr/8814au/blob/b1866ce2b857a8dfe2e147e19eb8eca0a842ce18/README.md#L259\n\nIf it still reproduces on current `main`, please attach a fresh `tools/runtime-healthcheck.sh` report so we can compare against the fixed path above.