morrownr/8814au #92: Android kernel, 5.4.147, unable to compile

TQMatvey, 2022-11-19:
Hello, i would like to get some help, i am working on nethunter kernel for S21 ultra (Android 13), kernel 5.4.147, trying to compile this driver with kernel results in

`error: unknown warning option '-Wno-discarded-qualifiers'; did you mean '-Wno-ignored-qualifiers'? [-Werror,-Wunknown-warning-option]`

Doing what it asks ( Replace `-Wno-discarded-qualifiers` with `-Wno-ignored-qualifiers` ) results in many more errors

https://katb.in/akaxufixodu

here is some information about prebuilt tools i use to build kernel
[image]

TQMatvey, 2022-11-19:
after "removing extraneous parentheses around the comparison to silence this warning"
https://katb.in/tafowakiyog
more of same issue

TQMatvey, 2022-11-19:
after fixing those issues, flashing, booting my kernel, and running airmon-ng start wlan1 in kali nethunter my phone reboots immediately.....

TQMatvey, 2022-11-19:
same if i try to enable mon mode from termux

morrownr, 2022-11-19:
Hi @TQMatvey 

I'm the primary maintainer here. I wish I had time to support additional platforms such as Android but I do not. I have looked at Android and Google makes many changes from the typical setup so I know it can be a challenge. This is complicated by how Realtek's drivers are designed so  with kernel 4.19 or later, I generally recommend users look at usb adapters that use in-kernel drivers. Kernel 5.4 supports chipsets such as the mt7612u and mt7610u as far as dual band goes and then you would only need to figure out which driver to compile in and where to place the firmware... and I can show you where the firmware is located.

Main Menu here can provide help with adapters that use in-kernel drivers and with firmware:

https://github.com/morrownr/USB-WiFi

Regards

TQMatvey, 2022-11-20:
Thanks will look in, i got an mt7612u will try to get it with inkernel drivers, i guess need to enable it in kernel config, Thanks

TQMatvey, 2022-11-20:
i was unable to get MT7612U or MT7601U to work :<, tried adding them to defconfig, they got built but adapter wasnt detected :(

morrownr, 2022-11-20:
>  tried adding them to defconfig, they got built but adapter wasnt detected :(

In-kernel drivers require 2 parts. defconfig on turns on the compiling of the open source part of the driver. For the driver to work, you also need to install the firmware (the closed source part.)

If you go to the site Main Menu and select item 8, it has instructions on where to download the firmware for Mediatek drivers. It even tells you how to install it on distros that use the mainline kernel but Android is modified enough that you will probably have to research where to install the firmware files.

https://github.com/morrownr/USB-WiFi

Main Menu item 8 is about Mediatek firmware.

TQMatvey, 2022-11-20:
[image]

uhhh well after reboot it still isnt detected

TQMatvey, 2022-11-20:
[image]

here is defconfig part

TQMatvey, 2022-11-20:
[image]

also tried putting to /system/lib/firmware/mediatek

morrownr, 2022-11-20:
> [image]
> 
> uhhh well after reboot it still isnt detected

I wouldn't start creating directories. I'm pretty sure Android has a directory for firmware but it may vary depending on who is supplying the Android distro. Need to find where the firmware directory is for your distro.

With that said, a quick net search suggests perhaps:

/system/etc/firmware

...and the Mediatek drivers may have code to look in .../mediatek... iow, try both of these:

/system/etc/firmware
/system/etc/firmware/mediatek

I  know, it makes no sense but Google does what they want and usb wifi adapters are not on their radar.

TQMatvey, 2022-11-20:
ANyways, thanks for helping, couldnt get mt7612u to work, will try other stuff, have a good day!

closed 2022-11-20 by TQMatvey