morrownr/8814au #93: Compile Error on Linux Kernel 4.18.0-425.3.1.el8.x86_64  ( Oracle Linux 8.7 )

jtome7, 2022-11-24:
Compile Error on Linux Kernel 4.18.0-425.3.1.el8.x86_64  ( Oracle Linux 8.7 )

An error occurred. dkms build error = 10

/var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/ioctl_cfg80211.c:441:3: error: too few arguments to function ‘cfg80211_ch_switch_started_notify’
   cfg80211_ch_switch_started_notify(adapter->pnetdev, &chdef, 0);
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /var/lib/dkms/rtl8814au/5.8.5.1/build/include/osdep_service_linux.h:94,
                 from /var/lib/dkms/rtl8814au/5.8.5.1/build/include/osdep_service.h:57,
                 from /var/lib/dkms/rtl8814au/5.8.5.1/build/include/drv_types.h:27,
                 from /var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/ioctl_cfg80211.c:17:
./include/net/cfg80211.h:7902:6: note: declared here
 void cfg80211_ch_switch_started_notify(struct net_device *dev,
      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/ioctl_cfg80211.c: At top level:
/var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/ioctl_cfg80211.c:9981:3: error: ‘struct cfg80211_ops’ has no member named ‘mgmt_frame_register’
  .mgmt_frame_register = cfg80211_rtw_mgmt_frame_register,
   ^~~~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/ioctl_cfg80211.c:9981:25: error: initialization of ‘int (*)(struct wiphy *, struct wireless_dev *, u64)’ {aka ‘int (*)(struct wiphy *, struct wireless_dev *, long long unsigned int)’} from incompatible pointer type ‘void (*)(struct wiphy *, struct wireless_dev *, u16,  bool)’ {aka ‘void (*)(struct wiphy *, struct wireless_dev *, short unsigned int,  _Bool)’} [-Werror=incompatible-pointer-types]
  .mgmt_frame_register = cfg80211_rtw_mgmt_frame_register,
                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

/var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/ioctl_cfg80211.c:9981:25: note: (near initialization for ‘rtw_cfg80211_ops.mgmt_tx_cancel_wait’)
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:317: /var/lib/dkms/rtl8814au/5.8.5.1/build/os_dep/linux/ioctl_cfg80211.o] Error 1
make[1]: *** [Makefile:1585: _module_/var/lib/dkms/rtl8814au/5.8.5.1/build] Error 2
make[1]: Leaving directory '/usr/src/kernels/4.18.0-425.3.1.el8.x86_64'
make: *** [Makefile:2389: modules] Error 2

morrownr, 2022-11-24:
Hi @jtome7 

> Linux Kernel 4.18.0-425.3.1.el8.x86_64 ( Oracle Linux 8.7 )

Is this Oracle distro based on Red Hat (RHEL)?

jtome7, 2022-11-24:
Sorry to leave that out ...
YES, this is the Red Hat kernel shipped & supported by Oracle.

*Thanks!**Joel*

*************************** Joel Tome *

*jtome7 @ gmail.com <http://gmail.com>****************************

morrownr, 2022-11-24:
> YES, this is the Red Hat kernel shipped & supported by Oracle.

I know I am going to sound like the Grinch that stole the usb wifi adapter but I have some less than optimal news for you.

Red Hat does some things in their RHEL distros that make it very hard to support out-of-kernel drivers. The only one of the Realtek drivers here at this site that has RHEL support is the one for 88x2bu. That support is provided by a very knowledgable programmer that works with RHEL. Currently the support is as follows:

RHEL 8.4 and 8.6 (kernel 4.18.0)

I allow RHEL programmers to merge RHEL support but they have to maintain it while I maintain everything else with help from a few folks. The reason I won't touch it is the amount of time it would take me to keep RHEL support up to date is large and I have more to do than I have time for anyway.

My recommendation is to use the Oracle or RHEL supports forums to ask for suggestions as to what would best serve your needs. Once you have some ideas, you could come back here for usb adapter specific info as I maintain a Main Menu here that provides a lot of information:

https://github.com/morrownr/USB-WiFi

jtome7, 2022-11-24:
Thanks for all that great information.

I have been using your GitHub driver for the Oracle kernel 5.4.17 with
these TP-Link T9UH adapters, and I really appreciate all the work you did.
That combination works great!

I have an ACER Aspire 5 laptop that fails to boot on the Oracle kernel, so
I have to use that RHEL compatible 4.18 kernel, and I was just going to try
adding the TP-Link T9UH adapter to the laptop.   No problem ... I can use
the built-in WIFI instead.

Thanks for everything.   Happy Thanksgiving!

*Thanks!**Joel*

*************************** Joel Tome *

*jtome7 @ gmail.com <http://gmail.com>****************************

morrownr, 2022-11-25:
> I have been using your GitHub driver for the Oracle kernel 5.4.17 with
these TP-Link T9UH adapters, and I really appreciate all the work you did.
That combination works great!

That is great! I wish I could be more help in this case but there are limits to how far I can stretch myself.

jtome7, 2022-11-28:
Hello, Nick!

Do you know whether I could use my cellular data SIM card with my Linux
computer?  Could I get some kind of USB stick that supports inserting a SIM
card and expect it to work on Linux?   Maybe something like this:

https://www.amazon.com/Serounder-Wireless-Network-Adapter-Receiver/dp/B07QYN4GCX/ref=sr_1_3?keywords=4G+LTE+%2F+5G+USB+SIM+Card&qid=1669594155&s=tv&sr=1-3&ufe=app_do%3Aamzn1.fos.08f69ac3-fd3d-4b88-bca2-8997e41410bb

*Thanks!**Joel*

*************************** Joel Tome*

*Greensboro, NC *

*jtome7 @ gmail.com <http://gmail.com>****************************

morrownr, 2022-11-28:
Joel,

> Do you know whether I could use my cellular data SIM card with my Linux computer?

This is something I know very little about. You might pop over to the Main Menu and leave a message in Issues as it gets more traffic than here:

https://github.com/morrownr/USB-WiFi